letscontrolit/ESPEasy #4022: esp32 devkitc v4 P052 tx and rx not available in espeasy

jvdVlugt, 2022-04-26:
### Checklist
I tried to use P052 in ESPEasy-mega-20220328 but in the pulldown menu for GPO  <-TX  and GPO ->RX the  GPO1 (TX)  and GPO3 (RX ) are grayed out and  are not availeble for selection.
[image]

ghtester, 2022-04-26:
Check the Tools - Advanced settings and disable Serial port and set the Serial Log Level to None.
[image]

jvdVlugt, 2022-04-26:
Thanks for the quick response, I see the  GPO points  without RX and TX. I need to complete theconnection with an max385 to be able to test it.
regards
Johan

jvdVlugt, 2022-04-26:
I accidently disabled  the serial port aftre enabeling the RX TX are grayed out again so turnoff logging does not help.

jvdVlugt, 2022-04-26:
Is having the serial connection connected needed to be able to select them?

tonhuisman, 2022-04-26:
You cannot have Serial log enabled when you want to use RX/TX for a task that wants to use those pins. Not sure what went wrong, but after disabling the serial log you should be able to use GPIO1 and GPIO3 for a task. Don't forget to submit that page.

jvdVlugt, 2022-04-26:
Then turning of logging did not help.

perhaps there is something wrong in the config ??? board is esp32_devkitc_V4

[env:Duco_esp32dev]
extends                   = esp32_common
lib_deps                  = ${esp32_common.lib_deps}, ServoESP32
build_flags               = ${esp32_common.build_flags}  
                                 -DFEATURE_ARDUINO_OTA 
board                     = esp32dev
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream

jvdVlugt, 2022-04-26:
Perhaps I was not clear in what the problem is.
I need the serial connection for modbus communication but the connections are not available for selection.
and turning of the logging as ghtester suggested did not do the trick.

tonhuisman, 2022-04-26:
What ghtester was (also) suggesting is to uncheck `Enable Serial port`, that will free the RX and TX pins for other uses, like what you need.

jvdVlugt, 2022-04-26:
I need RX and TX for this modbus connection so uncheck the 'enable serial connection'is not the way to enable the ports for RX and TX.

tonhuisman, 2022-04-27:
> I need RX and TX for this modbus connection so uncheck the 'enable serial connection'is not the way to enable the ports for RX and TX.

For a plugin to use the serial connection(pins), you have to disable the serial connection _for use by ESPEasy core functions like logging_, so, *yes* that checkbox *needs* to be unchecked to make the port available for other uses.

tonhuisman, 2022-04-27:
Here's the link to the documentation, explaining exactly this: https://espeasy.readthedocs.io/en/latest/Tools/Tools.html#serial-settings

TD-er, 2022-04-27:
ESP32 does have 3 serial ports, which can be configured using just about any GPIO.
If you need the serial logging, then Serial0 is already used.
So just select another serial port (e.g. Serial1)

If GPIOs are already assigned to something else (e.g. Serial0 or I2C) then the reason for its block is also listed in the combo box you also show in your screenshot.

jvdVlugt, 2022-04-29:
Thanks I have now a working modbus connection on other I/O ports.
dont know why the original TX and RX ports are not available but this is working now for me.
Thank you very much for your suggestions.
I have now a working system for the Duco IQ unit so if anyone is interested let me know.

Regards Johan

TD-er, 2022-04-29:
> I have now a working system for the Duco IQ unit so if anyone is interested let me know.

Sounds interesting.
What do you read from where?

N.B. that would be a great use case to show on the forum.

jvdVlugt, 2022-04-29:
Duco IQ unit is a system that controlles the ventilation system and has data like CO2 values , Humidity , percent opening of valves and temperatues of the different devices in the system.
because all ,modbus registers are in the same order in groupes of 10 registers  for a devive  I can calculate the modbus number just by putting the device number in espeasy and selecting what value you want to see.
I used a esp32 becouse I needed more than the 12 tasks I expanded the ESP32 to 64 tasks and that works fine for now.

tonhuisman, 2023-09-01:
Issue is solved, it seems, so this issue can be closed.

closed 2023-09-01 by TD-er