letscontrolit/ESPEasy #4024: Used flash tool has big impact on WiFi performance

TD-er, 2022-04-28:
For a long time, users have reported WiFi issues, where others experienced none. Both running the same build.

When testing the web upload tool myself, I noticed some ESP32 unit I used for development was acting extremely slow.

- [ESPEasy 20220427 build](https://td-er.nl/ESPEasy/20220427/)
- [ESPEasy 20220428 with IDF-4.4 code merged](https://td-er.nl/ESPEasy/20220428/)

I tried flashing to an ESP32 with 4MB flash. (e.g. ESP_Easy_mega_20220428_test_A_ESP32_4M316k )
The 20220427 build does not yet use the new board definitions introduced with the IDF-4.4 pull request.
So the flash may be running at 80 MHz on those builds, unless patched by the flash tool.
N.B. The web flash tool does not patch files.

- Flashing 20220427 build via the web flasher, results in a completely unusable system as it is terribly slow and resulting in lots of timeouts.
- Flashing 20220428 build via the web flasher, results in somewhat usable system, but still a lot less responsive compared to what I'm used to on that board.
- Flashing 20220428 build via the IDE, thus using esptool.py to flash, does make the board as responsive as ever with very stable WiFi.

I also performed an OTA flash to run a linux made build, just to make sure it isn't related to the build platform.
This OTA flash does result in a very responsive system, at least when starting from a responsive system to be begin with.

So I'm wondering what has to be changed to make every node work well on every build, regardless the flashing tool.

@Jason2866 Do you have any idea?

Jason2866, 2022-04-28:
I see a performance decrease when i often do (OTA) upgrades. But for me this was not related how the where flashed. The issues always disappeared after a full flash erase and flashing again. Thats a miracle for me and i have no idea whats the reason is and how it can be solved.

TD-er, 2022-04-28:
> A wild guess, maybe the wifi calibration is "damaged" somehow over the time when doing updates. Flashing without a full erase uses old wifi calibration so this is always a good idea for new devices.

Well my test node did not get a full erase of the flash and it worked fine after flashing it via the esptool.py flash tool.
Even with a major SDK jump in code.

TD-er, 2022-04-28:
> What's the difference between OTA and Web flasher please?

Ah, I get the confusion :)
- OTA = flashing via the ESPEasy web interface, as you always did.
- Web flasher = new fancy web tool (see both links I posted) where you can flash the ESP via USB directly from the browser.

TD-er, 2022-04-28:
> Is it related to ESP32 only?

Not sure, however I was not able to flash the specific ESP8266 on my desk with the web flash tool. Might be related to the specific USB to serial chip on that one, as that specific board can also not be flashed using the ESPEasyFlasher (the one with the blue color and the "special font" (let's say I'm not a fan of that font choice...)

Jason2866, 2022-04-28:
I had this with esp8266 in the past too. I do nearly nothing with a esp8266 since a year.
All time spent to get the new espressif mcu well implemented in Tasmota.
This opened a big new rabbit hole. Since support in Platformio was lacking completly
and jumping in there opened new ones...

Jason2866, 2022-04-28:
> Not sure, however I was not able to flash the specific ESP8266 on my desk with the web flash tool. Might be related to the specific USB to serial chip on that one, as that specific board can also [...]

Have you tried the new webflasher version? Talking about rabbit holes i refactored the webserial code to support the S3, now it should recognize more USB serial adapters too.

Jason2866, 2022-04-28:
Is EspEasyFlasher still using the old Esptool.exe? Since that time there have been many fixes and enhancements done. For flashing "on the desk" esptool.py v3.3 should be used. There are compiled binaries available too.

TD-er, 2022-04-28:
I updated to the latest 8.0.1 version of the web flasher, which is -as far as I know- the latest.

And the ESPEasy Flasher tool does look into the Windows registry to find the COM ports.
Apparently there are some USB to serial chips out there that trigger some driver to get installed which does not register itself in the registry like any other serial driver does.
So it is perfectly possible that the Chrome implementation to interact with the serial port suffers from the same issue.

Jason2866, 2022-04-28:
@TD-er yep using 8.0.1 Probably it fails at the same point, just looking different