letscontrolit/ESPEasy #4025: Rules Syntax

TommoT1, 2022-04-28:
Till now the Syntax of Rules was like this:
on <trigger> do <action> endon 
When it was only within one line the endon was optional.
Now with release 20220427 it is mandatory, even on "oneliners".
Is this a bug or on purpose?

Regards Tom

TD-er, 2022-04-28:
> Is this a bug or on purpose?

I think it is a bug as I have to admit I haven't tested the one-liner rules.

TD-er, 2022-04-30:
Just looking into this issue.

I think it just looks like it is the other way around;
You can't have endon on the one-liner rule with the current code.

My 2 lines of test rules:
```
on myevent do logentry,"Oneliner test" endon

on myevent2 do logentry,"Oneliner test no endon"
```

With the logs of calling first `event,myevent2` and then `event,myevent` :
```
2236763 : Info   : HTTP: event,myevent2
2236764 : Debug  : HTTP after parseTemplate: event,myevent2
2236792 : Info   : EVENT: myevent2
2236923 : Info   : ACT  : logentry,"Oneliner test no endon"
2236926 : Debug  : Command: logentry
2236927 : Debug  : logentry,"Oneliner test no endon"
2236928 : Info   : Oneliner test no endon

2256706 : Info   : HTTP: event,myevent
2256708 : Debug  : HTTP after parseTemplate: event,myevent
2256731 : Info   : EVENT: myevent
2256735 : Info   : ACT  : logentry,"Oneliner test" endon
2256738 : Debug  : Command: logentry
2256738 : Debug  : logentry,"Oneliner test" endon
2256740 : Info   : Oneliner test
2256742 : Info   : ACT  : logentry,"Oneliner test no endon"
2256744 : Debug  : Command: logentry
2256744 : Debug  : logentry,"Oneliner test no endon"
2256745 : Info   : Oneliner test no endon
```

So it seems that with `endon` present on the one-liner rule line, it just continues to also execute the next one.

Maybe I still need to do some more testing here like swapping the lines, adding more, etc.

But right now I don't know whether I should 'patch' rules lines when reading them by either adding an `endon` if missing on a one-liner, or removing it.

TD-er, 2022-04-30:
OK, fixed it.

- Now clearly mark we're done after a oneliner, instead of continuing with the next on...do block or one-liner
- The optional `endon` is stripped from one-liner lines to make sure it is not possibly affecting the command you try to execute.

closed 2022-04-30 by TD-er

TommoT1, 2022-08-01:
Hi Gijs,
is it possible you fixed the "one-liner-issue" only for 8266 builds?
I just got three sonoff s26 switches with a 8285 inside and with the build from 16.06.2022 they show the same strange behaviour as the 8266 before your fix...

Regards
Thomas

TommoT1, 2022-08-01:
Yes I know  about the 8285, I just flashed the S26R2 devices and had to create a rule to make the switch button work again, so I wrote 

on Taster#Switch=1 do event,toggle

on toggle do
 if [Relay#State]= 0
   gpio,12,1
 else
   gpio,12,0
 endif
endon

This didn't work until I changed the first part to 

on Taster#Switch=1 do 
 event,toggle
endon

So at first glance I looked for me like the one liner problem again...
Same code (the one liner) is working flawless on an device with 8266 an latest build from 20220616

TD-er, 2022-08-01:
Can you check the rules related settings on the Tools->Advanced page?

Like the optimizations checkboxes.
Please compare those to the ones on the ESP8266 unit.

TommoT1, 2022-08-02:
Yes, that's the difference, the checkbox
"Enable Rules Cache: " was unchecked. 
Now I checked "Enable Rules Cache:"  "yes" and it is working with the oneliner!
Thanks!

TD-er, 2022-08-02:
OK, I will then look into the code to see why that is.
Using the cache should not make a difference in executing the rules, only in the speed of execution.

Do you still have a checkbox to reorder the rules to optimize for speed?
If so, please uncheck that checkbox as that's for sure causing issues.

TommoT1, 2022-08-02:
Yes I have the Checkbox"Optimize Rules Cache Event Order:" but I already read in another post from you somewhere to leave it unchecked....

Tanks again for you help!

PR #4028 (2022-04-30): [Rules] Fix issues with rules one-liners (#4025)
- Now clearly mark we're done after a oneliner, instead of continuing with the next on...do block or one-liner
- The optional endon is stripped from one-liner lines to make sure it is not possibly affecting the command you try to execute.

Fixes: #4025
commit aeca829c (2022-04-30): [Rules] Fix issues with rules one-liners (#4025)

- Now clearly mark we're done after a oneliner, instead of continuing with the next on...do block or one-liner
- The optional endon is stripped from one-liner lines to make sure it is not possibly affecting the command you try to execute.

Fixes: #4025