letscontrolit/ESPEasy #4031: max7219 (7seg) plugin leads to bootloop on newer builds...

chromoxdor, 2022-05-02:
for this to happen no display needs to be attached...it is enough to just activate the plugin
tested releases:
mega-20211224  -> works
mega-20220328 -> bootloop
mega-2022042 -> bootloop

tonhuisman, 2022-05-03:
Does this behavior change if you disable the task with the SI7021 sensor? In the partial log you shared I can't see anything related to the 7DGT [P073] plugin.
Is more log available, just after the `CUT HERE FOR EXCEPTION DECODER` line there usually is a decoded exception-log?,( when using the `monitor` feature of PlatformIO)

chromoxdor, 2022-05-03:
I also testet it on a newly flashed d1mini.
If just found out, that the plugin works until i choose  clock (any) as display output.
The log on platformIO looks the same:

```
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

>>>stack>>>

ctx: cont
sp: 3fff2c00 end: 3fff30e0 offset: 01a0
3fff2da0:  00000020 00000001 0000000d 40100ded  
3fff2db0:  0000000a 0000000d 0000000c 4027fbb8  
3fff2dc0:  00000001 00000010 7ffffffe 0000000f  
3fff2dd0:  00000001 00000000 0000000f 0000000d  
3fff2de0:  0000000c 0000000f 3fff5e8c 402203e4  
3fff2df0:  3fff2d01 3fff2d54 3fff2df8 001374a7  
3fff2e00:  0000000b 00000004 00000000 0000000d  
3fff2e10:  0000000c 3fff5e8c 3fff2f20 4022050c  
3fff2e20:  000021d8 0000043b 3fff2f20 40220614  
3fff2e30:  00000000 00000000 3fff2e90 3fff2f20  
3fff2e40:  03040607 3fff0001 00000000 40261f90  
3fff2e50:  0000000f 000000ed 00000000 401061d1  
3fff2e60:  00003710 000006e2 00000000 3fff2f20  
3fff2e70:  00000004 3fff5e8c 3fff2f20 40221c45  
3fff2e80:  00000004 0000002d 00000020 4010176b  
3fff2e90:  3fff5b00 0010001f 3fff2ed0 4027cab0  
3fff2ea0:  00000004 0000002d 3fff2ed0 4027ca91  
3fff2eb0:  402d0000 3fff0d98 00000000 3fff2f20  
3fff2ec0:  00000004 0000002d 00000000 40244c2c  
3fff2ed0:  3fff7000 000c000f 80ff2f00 4027ca91  
3fff2ee0:  3fff2f20 3fff2ff0 00137b80 00000000  
3fff2ef0:  3fff2ff0 3fff2f20 3fff2f80 401061d1  
3fff2f00:  4027ac70 00000000 00000004 001374a7  
3fff2f10:  3fff2ff0 3fff2f20 00000004 40245157  
3fff2f20:  00000000 00000000 80000002 00000000  
3fff2f30:  1e1f1c1f 801f1e1f 1f1e1f00 0000001d  
3fff2f40:  80000002 00000000 3fff1d58 80249f0d  
3fff2f50:  00000100 00000000 80100d66 00000000  
3fff2f60:  00000000 00000000 00000000 00000000  
3fff2f70:  00000000 00000000 03030000 40000000  
3fff2f80:  67756c00 61436e69 80006c6c 001374a7  
3fff2f90:  00000000 00000000 4bc6a7f0 00000000  
3fff2fa0:  00000000 bff00000 40100d66 170a3d70  
3fff2fb0:  00000000 bff00000 0000000c 00004e58  
3fff2fc0:  00000002 3fff09f8 3fff3008 4027ca91  
3fff2fd0:  00000000 3fff09f8 00000000 001374a7  
3fff2fe0:  00000000 3fff09f8 3fff1d58 4025e70a  
3fff2ff0:  00000800 285eec01 80100d00 c8b43958  
3fff3000:  000008ce 3fff3020 00000300 10000002  
3fff3010:  8013738f 3fff1cf8 00000002 402558f0  
3fff3020:  000008ce 00000000 00000000 10000002  
3fff3030:  0013738f 402606b8 3fff1cf8 4026070a  
3fff3040:  0013738f 00000000 3fff1cf8 40260858  
3fff3050:  0013734f 000004e6 3fff0900 40240558  
3fff3060:  00000002 00000000 3fff1cf8 3fff312c  
3fff3070:  3fffdad0 00000000 3fff09f8 3fff312c  
3fff3080:  3fffdad0 00000000 3fff09f8 40242871  
3fff3090:  00000000 feefeffe feefeffe feefeffe  
3fff30a0:  00000000 00000000 00000001 3fff312c  
3fff30b0:  3fffdad0 00000000 3fff30e8 4020f974  
3fff30c0:  feefeffe feefeffe feefeffe 4027e4f3  
3fff30d0:  feefeffe feefeffe 3ffe86d4 40100671  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)
[... 64 lines cut ...]
3fff2f90:  00000000 00000000 4bc6a7f0 00000000  
3fff2fa0:  00000000 bff00000 40100d66 7eb851eb  
3fff2fb0:  00000000 bff00000 0000000c 00004f08  
3fff2fc0:  00000002 3fff09f8 3fff3008 4027ca91  
3fff2fd0:  00000000 3fff09f8 00000000 00137a28  
3fff2fe0:  00000000 3fff09f8 3fff1d58 4025e70a  
3fff2ff0:  00000800 39cf5e39 80100d00 3db22d0e  
3fff3000:  000008cf 3fff3020 00000300 10000002  
3fff3010:  8013794b 3fff1cf8 00000002 402558f0  
3fff3020:  000008cf 00000000 00000000 10000002  
3fff3030:  0013794b 402606b8 3fff1cf8 4026070a  
3fff3040:  0013794b 00000000 3fff1cf8 40260858  
3fff3050:  00137936 000004e7 3fff0900 40240558  
3fff3060:  00000002 00000000 3fff1cf8 3fff312c  
3fff3070:  3fffdad0 00000000 3fff09f8 3fff312c  
3fff3080:  3fffdad0 00000000 3fff09f8 40242871  
3fff3090:  00000000 feefeffe feefeffe feefeffe  
3fff30a0:  00000000 00000000 00000001 3fff312c  
3fff30b0:  3fffdad0 00000000 3fff30e8 4020f974  
3fff30c0:  feefeffe feefeffe feefeffe 4027e4f3  
3fff30d0:  feefeffe feefeffe 3ffe86d4 40100671  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

TD-er, 2022-05-03:
There has been 1 major change on that plugin since it stopped working: https://github.com/letscontrolit/ESPEasy/commit/c897535627330d0a88c0468d089575afd753040e

TD-er, 2022-05-03:
Found the bug:

```c++
  for (uint8_t i = 5; i >= 0; i--) {
    max7219_SetDigit(event, din_pin, clk_pin, cs_pin, idx_list[i], P073_data->showbuffer[i], false);
  }
```

TD-er, 2022-05-03:
Can you try the test build generated by GitHubActions: https://github.com/letscontrolit/ESPEasy/actions/runs/2263100841

closed 2022-05-03 by TD-er

chromoxdor, 2022-05-03:
everything seems to work.... thank you!!!

thiboot, 2022-05-03:
New build resolved the bootloop problem for me but i have another problem, clock works randomly... I don't know if it's related... Sometime all 8 leds turns on for 10 seconds and stop :-/

TD-er, 2022-05-03:
@thiboot Do you also use the same max7219 display?

thiboot, 2022-05-03:
Yep, it's this one : https://www.ebay.fr/itm/255283221964
I have two of the same model, same result, on an espressif ESP32 configured with the module on the GPIO-27 (DIN), GPIO-25 (CLK), GPIO-26 (CS) :-/

chromoxdor, 2022-05-03:
I have the same same display module and just tested it with an esp32 and 24h - Blink as output.
I connected the display to the same pins than yours. So far no errors...

thiboot, 2022-05-03:
Uptime is ok, no reboots, NTP time ok, i don't have exactly the same ship as yours. Your 7d is powered with 3.3V or 5V in your case ?
[image]

chromoxdor, 2022-05-03:
mine is powered with 3,3v...
I don´t see why yours shouldn’t work...  Is something else running on the ESP32?
And still:
> What do you mean with sometimes?

thiboot, 2022-05-03:
Yep, it's very weird. Sometimes means randomly i'll check all your suggestions :-) Maybe my soldering after all 😅 Without any explanations, i rebooted again and now all leds are fully lighted 🤯

PR #4035 (2022-05-03): [P073] Correct endless loop in max7219_ShowTime
Resolves #4031