letscontrolit/ESPEasy #4036: ESP32: compile error

kuppe234, 2022-05-03:
Hello,
I can´t compile binarys for the ESP32. Everything works fine for the ESP8266. 
At the beginning of April, everything still worked for the ESP32.
I work under Linux. Here is my command line:
`platformio run --environment normal_ESP32_4M316k`
Here is the end of the compile:
```
Linking .pio/build/normal_ESP32_4M316k/ESP_Easy_mega_20220503_normal_ESP32_4M316k.elf
Retrieving maximum program size .pio/build/normal_ESP32_4M316k/ESP_Easy_mega_20220503_normal_ESP32_4M316k.elf
Checking size .pio/build/normal_ESP32_4M316k/ESP_Easy_mega_20220503_normal_ESP32_4M316k.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  15.6% (used 51020 bytes from 327680 bytes)
Flash: [======    ]  64.8% (used 1231805 bytes from 1900544 bytes)
Building .pio/build/normal_ESP32_4M316k/ESP_Easy_mega_20220503_normal_ESP32_4M316k.bin
esptool.py v3.3
Creating esp32 image...
Merged 25 ELF sections
Successfully created esp32 image.
esp32_create_combined_bin([".pio/build/normal_ESP32_4M316k/ESP_Easy_mega_20220503_normal_ESP32_4M316k.bin"], [".pio/build/normal_ESP32_4M316k/ESP_Easy_mega_20220503_normal_ESP32_4M316k.elf"])
Generating combined binary for serial flashing
    Offset | File
 -  0x1000 | /home/dhe/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/bin/bootloader_dout_40m.bin
 -  0x8000 | /home/dhe/Projekte/ESPEasy-Project/ESPEasy.git/.pio/build/normal_ESP32_4M316k/partitions.bin
 -  0xe000 | /home/dhe/.platformio/packages/framework-arduinoespressif32/tools/partitions/boot_app0.bin
 - 0x10000 | /home/dhe/Projekte/ESPEasy-Project/ESPEasy.git/.pio/build/normal_ESP32_4M316k/ESP_Easy_mega_20220503_normal_ESP32_4M316k.bin
usage: esptool [-h] [--chip {auto,esp8266,esp32}] [--port PORT] [--baud BAUD]
               [--before {default_reset,no_reset,no_reset_no_sync}]
               [--after {hard_reset,soft_reset,no_reset}] [--no-stub]
               [--trace] [--override-vddsdio [{1.8V,1.9V,OFF}]]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,version}
               ...
esptool: error: argument operation: invalid choice: 'merge_bin' (choose from 'load_ram', 'dump_mem', 'read_mem', 'write_mem', 'write_flash', 'run', 'image_info', 'make_image', 'elf2image', 'read_mac', 'chip_id', 'flash_id', 'read_flash_status', 'write_flash_status', 'read_flash', 'verify_flash', 'erase_flash', 'erase_region', 'version')
*** [.pio/build/normal_ESP32_4M316k/ESP_Easy_mega_20220503_normal_ESP32_4M316k.bin] Explicit exit, status 2
=================================================== [FAILED] Took 103.67 seconds ===================================================

Environment          Status    Duration
-------------------  --------  ------------
normal_ESP32_4M316k  FAILED    00:01:43.670
============================================== 1 failed, 0 succeeded in 00:01:43.670 ==============================================

```
I think the error is in "esp32_create_combined_bin". Can someone help?
platformio are uptodate.

Greetings, kuppe.

TD-er, 2022-05-03:
Looks like you don't have the same toolchain installed.
This `merge_bin` is part of the Python package `esptool`
Can you open a console in VS code. This way the same Python environment is activated as is used during building.
Then you must run:

```
pip install esptool==3.2
```
Or simply run:
```
pip install -r requirements.txt
```

kuppe234, 2022-05-03:
Thanks for the quick help.
When starting the compilation, some packages are installed, including esptool (3.3.0).

```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32_4M.html
PLATFORM: Espressif 32 (2.0.3) > Espressif Generic ESP32 4M Flash, ESPEasy 1810k Code/OTA, 316k FS
HARDWARE: ESP32 240MHz, 320KB RAM, 1.81MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 2.0.3 
 - tool-esptoolpy 1.30300.220323 (3.3.0) 
 - tool-mklittlefs 1.203.210628 (2.3) 
 - tool-mkspiffs 2.230.0 (2.30) 
 - toolchain-esp32ulp 1.22851.191205 (2.28.51) 
 - toolchain-xtensa-esp32 8.4.0+2021r2-patch3
```
I thought that was enough.
With additional "esptool==3.2" the compilation runs.

Greetings, kuppe.