letscontrolit/ESPEasy #4045: Add I2C address selector for P107 SI1145

TD-er, 2022-05-07:
See also [forum topic](https://www.letscontrolit.com/forum/viewtopic.php?p=58341#p58341)
The Adafruit library has a hard coded default address, but apparently the chip does support other addresses too.

tonhuisman, 2022-05-07:
There can be multiple strategies here:
1) Provide a set of predefined addresses, limited to the currently 'known' 0x60 and 0x53 (will require external tooling to change the address, when not using an I2C multiplexer).
2) Provide a range of predefined addresses, including the currently 'known' addresses, similar to the changeable address in [P124 - I2C Multi-relay](https://espeasy.readthedocs.io/en/latest/Plugin/P124.html#p124-page)
3) Allow any valid I2C address to be programmed (0x08..0x77 non-reserved range)
(could be configured using a command like:
`si1145address,<old_address>,<new_address>`
and then also stored in the device configuration).

Strategy 1 seems quite limiting, and strategy 3 will probably be somewhat confusing for users though the most flexible, or we could go for strategy 2, but then we will have to decide what range to support 🤔

tonhuisman, 2022-05-07:
Additional info in the forum thread reveals that the device at 0x53 turned out to be an LTR390, as supported from PR #3984 , as the supplier seems to have shipped the wrong board 😮.

It is also unlikely that the modified I2C address is stored in the chip (it is written to RAM), so it will require some very special initialization, switching off any other SI1145 devices when the address is changed. That would require extra hardware to be used, controllable from ESPEasy (that's possible, though).
Best solution is probably to leave this plugin as is, and use an I2C multiplexer to connect multiple sensors to a single ESPEasy device.

TD-er, 2022-05-07:
OK, given the feedback on the forum, I think this can be closed. :)

closed 2022-05-07 by TD-er