letscontrolit/ESPEasy #4047: esptool.py v3.0 cannot connect to a newly bought ESP8266 Wemos D1 Mini Pro

uwekaditz, 2022-05-07:
The download tool esptool.py (v3.0, in VS integrated) cannot connect to a newly bought ESP8266 Wemos D1 Mini Pro.
It returns with `A fatal error occurred: Failed to connect to ESP8266: Timed out waiting for packet header`

**The esptool.exe in the folder letscontrolit/ESPeasy/dist works fine.**
```
esptool v0.4.12 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
        setting board to nodemcu
        setting baudrate from 115200 to 115200
        setting port from  to COM9
        setting address from 0x00000000 to 0x00000000
        espcomm_upload_file
        espcomm_upload_mem
        setting serial port timeouts to 1000 ms
opening bootloader
resetting board
trying to connect
        flush start
        setting serial port timeouts to 1 ms
        setting serial port timeouts to 1000 ms
        flush complete
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
        read 0, requested 1
trying to connect
        flush start
        setting serial port timeouts to 1 ms
        setting serial port timeouts to 1000 ms
        flush complete
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
Uploading 855504 bytes from .\ESP_Easy_mega_20220507_custom_IR_ESP8266_4M1M.bin to flash at 0x00000000
        erasing flash
        size: 0d0dd0 address: 000000
        first_sector_index: 0
        total_sector_count: 209
        head_sector_count: 16
        adjusted_sector_count: 193
        erase_size: 0c1000
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
        setting serial port timeouts to 15000 ms
        setting serial port timeouts to 1000 ms
        espcomm_send_command: receiving 2 bytes of data
        writing flash
................................................................................ [  9% ]
.................
```

TD-er, 2022-05-07:
Hmm need to check this.
What exact version of esptool.py do you use?

uwekaditz, 2022-05-07:
I'm not sure which esptool.py is called by VS, but in c:\Users\user\.platformio\packages the latest folder for an esptool is
tool-esptoolpy@1.30000.201119.
It contains at least the version string `__version__ = "3.0"` which I also see in VS.

uwekaditz, 2022-08-11:
I have found the cause of the problem.
The function `connect()` in .platformio\packages\tool-esptoolpy\esptool\loader.py is called with the parameter `no_reset`.
Because I do not know where this parameter is set in the VS code or PlatformIO for uploading I coded it hard to `default_reset` in .platformio\packages\tool-esptoolpy\esptool\loader.py and now it works.
@TD-er: Can you provide the information where the parameter `no_reset` is set?

TD-er, 2022-08-11:
Hmm that's strange, as I think I remember to use the "nodemcu" reset option.
Will have a look as I am also quite frustrated by this...
Thanks for looking into it, will fix it ASAP now that I know where to look.

TD-er, 2022-08-11:
Do you have an example of the command line you now use?

This is what I have now using verbose upload:

```
<lambda>(["upload"], [".pio\build\normal_ESP32_4M316k\ESP_Easy_mega_20220811_normal_ESP32_4M316k.bin"])
AVAILABLE: cmsis-dap, esp-bridge, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
BeforeUpload(["upload"], [".pio\build\normal_ESP32_4M316k\ESP_Easy_mega_20220811_normal_ESP32_4M316k.bin"])
Auto-detected: COM3
"C:\Users\gijsn\.platformio\penv\Scripts\python.exe" "C:\Users\gijsn\.platformio\packages\tool-esptoolpy@src-1064c5feba8d3197dfd43b0a09aadea9\esptool.py" --chip esp32 --port "COM3" --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dout --flash_freq 40m --flash_size 4MB 0x1000 C:\Users\gijsn\.platformio\packages\framework-arduinoespressif32\tools\sdk\esp32\bin\bootloader_dout_40m.bin 0x8000 C:\GitHub\TD-er\ESPEasy\.pio\build\normal_ESP32_4M316k\partitions.bin 0xe000 C:\Users\gijsn\.platformio\packages\framework-arduinoespressif32\tools\partitions\boot_app0.bin 0x10000 .pio\build\normal_ESP32_4M316k\ESP_Easy_mega_20220811_normal_ESP32_4M316k.bin
esptool.py v3.3.1
Serial port COM3
Connecting......................................

A fatal error occurred: Failed to connect to ESP32: Wrong boot mode detected (0x13)! The chip needs to be in download mode.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
*** [upload] Error 2
```
But it still isn't working.

uwekaditz, 2022-08-11:
I tested it with the latest esptool (v4.2-dev).
In the  esptool\loader.py is a warning printed if the modes is `no_reset`.
I added the line **mode="default_reset"** and it works:

    def connect(
        self,
        mode="default_reset",
        attempts=DEFAULT_CONNECT_ATTEMPTS,
        detecting=False,
        warnings=True,
    ):
        mode="default_reset"
        """Try connecting repeatedly until successful, or giving up"""
        if warnings and mode in ["no_reset", "no_reset_no_sync"]:
            print(
                'WARNING: Pre-connection option "{}" was selected.'.format(mode),
                "Connection may fail if the chip is not in bootloader "
                "or flasher stub mode.",
            )
        print("Connecting...", end="")
        sys.stdout.flush()
        last_error = None

uwekaditz, 2022-08-30:
With the latest commit ([ESP8266] Change resetmethod to "nodemcu" to fix reset during flash) the used reset mode is  **"default_reset"** and this works.

TD-er, 2022-08-30:
Yep, that's what I've now changed in the board definitions.
At least for some ESP32 boards and on all ESP8266 boards I tested it now works again.

uwekaditz, 2022-12-28:
The reset is working fine!

closed 2022-12-28 by uwekaditz