letscontrolit/ESPEasy #4048: PCF8591 DA support

ghtester, 2022-05-08:
I would like to use the PCF8591 chip's analog output as well.
The  **Analog input - PCF8591** plugin works fine but just for inputs.
Although I can set the analog output with **wdconfig,72,64,x**  command, it's reset to zero as soon as the plugin reads an input data.
I suppose the read command disables the analog output by default and it can't be configured currently.
Could it be possible to add a status bit to the plugin's configuration which would enable the analog output please?

tonhuisman, 2022-05-08:
I'll pick this one up, but then I'll also implement the 'Analog input programming' bits, for single-ended and differential combinations of inputs 😛

tonhuisman, 2022-05-08:
The new `Hardware configuration` options look like this: (PR will probably be done in about 24h)

[image]

And available `Input mode` options:

[image]

tonhuisman, 2022-05-08:
It all went a bit quicker than I expected, so there you go. 😎 
Please test, you can download once the Github Actions build is completed.

ghtester, 2022-05-09:
Wow, thanks a lot for your superquick & great work! :-)`
Quickly tested and it looks fine, I can confirm the analog output works as expected through both **wdconfig** and **analogout** commands. :-)
The differential inputs to be tested later.

tonhuisman, 2022-05-09:
The `wdconfig` command is just a happy coincidence that it is working, as AFAICS it wasn't designed to be used for this, that's why I added the `analogout` command 😃

ghtester, 2022-05-09:
Yes I know, luckily the **wdconfig**  command works even without watchdog circuit so it's a good tool for testing i2c devices. :-)
The specific command added is of course a much better solution. 
Thanks again for your good job! :-)

uzi18, 2022-05-09:
Have some applications where wdconfig is the core of communication :)

ghtester, 2022-05-09:
FYI - I made some other tests with differential inputs on another test ESP 8266 node. As I had a strange issues with hanging & crashing node after the firmware upgrade, I decided to delete the PCF8591 plugin and create it again.
Before that I have changed the plugin's configuration from 4  single-ended inputs to  3 diff. inputs. Before the plugin deletion I have noticed the formulas were added.
[image]

When I created the plugin again and selected 3 diff. inputs in first configuration, no formulas appeared in the configuration. 
[image]

ghtester, 2022-05-09:
Thanks, I currently don't need the diff. input feature so I did not read the documentation carefully but it looks like a bad chip design... ;-)

tonhuisman, 2022-05-09:
Well, when using the dual-differential mode, only channels 0 and 1 are used, so only 2 values set 😃

ghtester, 2022-05-09:
Yeah I supposed that... So most useful is the 4 single-ended input mode anyway as there's no problem to create appropriate formulas to get the results from all other modes...

tonhuisman, 2022-05-09:
There are different reasons to use a differential input, especially when having a common ground, you may need to measure a diff without connecting either of them to GND.

PR #4050 (2022-05-31): [P007] Add Input mode selection, Enable Analog output, analogout command, updated documentation
Resolves #4048 

- Add Enable Analog output (AOUT)
- Add `analogout,<0..255>` command
- Add Input mode selection, enabling also differential input options
- Use ESPEasy core I2C functions where possible
- Fix a code-typo
- Updated documentation

TODO:
- [x] Testing (help requested) Analog output: [confirmed](https://github.com/letscontrolit/ESPEasy/issues/4048#issuecomment-1120514801) Differential input: [confirmed](https://github.com/letscontrolit/ESPEasy/issues/4048#issuecomment-1121320126)