letscontrolit/ESPEasy #4052: Level 0-3m in 4-20Ma ->Wemos Esp8266

Ricco59163, 2022-05-09:
Hello, here is my problem, I recently bought a level (absolute pressure) where I come with a 4-20Ma current loop generating a voltage of 0.2V - 3V. I would like to know how to see this measurement on the Wemos I use the Build: [ESP_Easy_mega_20220427_normal_ESP8266_4M1M Apr 27 2022]

thx for help

tonhuisman, 2022-05-09:
You should be able to connect your signal to the `A0` pin on your Wemos, then configure an 'Analog input - internal' device, enable that and you should be able to see the measured value on the Devices overview page.
The Wemos should convert the 0..3.3V input to the 0..1V supported input of the ESP8266, so that should just be fine.

Ricco59163, 2022-05-09:
ok thx , I test tomorow ,

TD-er, 2022-05-09:
What you need is something like [this](https://www.aliexpress.com/item/32973712244.html?spm=a2g0o.order_list.0.0.319418028ilHU8)
Converting a 4...20 mA signal into an analog voltage can be done using a simple resistor, as the voltage over a resistor is proportional to the current through it.
But it is more practical to use one of those boards as you can also set ranges and have some protections in place.

Ricco59163, 2022-05-10:
hi,
yes I now , i use this card 

my referencial : -> https://www.turais.de/current-to-voltage-module-for-sensing-small-currents-4-20ma/

Ricco59163, 2022-05-12:
> You should be able to connect your signal to the `A0` pin on your Wemos, then configure an 'Analog input - internal' device, enable that and you should be able to see the measured value on the [...]

don't work : in multimeter , I have 2.986V for 20ma  and the Esp don't display this

Ricco59163, 2022-05-12:
work fine thx 

4/20ma  --> 0.2/3v  --------ESP------> 74/968