letscontrolit/ESPEasy #4053: GPIOtoggle dont work

kuppe234, 2022-05-10:
Hello,

there is an issue with the command `GPIOtoggle`.
With `GPIO` I can switch on or off the output of a gpio. But when I use `GPIOtoggle` to toggle the output, I only can switch them on.
I have test it with the current version (mega_fd9557b, ESP_Easy_mega_20220510). 
On version from april 1 allways worked fine.

Greetings, kuppe

tonhuisman, 2022-05-10:
Is this on ESP8266 or ESP32? As for ESP32 there has been a major upgrade of the core libraries that could be the cause here.

kuppe234, 2022-05-10:
This is on ESP32. I have test it with normal_ESP32_4M316k, test_A_ESP32_4M316k an display_ESP32_4M316k.

TD-er, 2022-05-10:
This might be fixed with the latest espressif32/Arduino SDK as there have been some internal changes in addressing GPIO pins.

tonhuisman, 2022-05-10:
Just tried, and indeed it turns on, but never off using `gpiotoggle`.

kuppe234, 2022-05-19:
Hello,
today with the actual GIT und Platformio `gpiotoggle` is working again.
Here are my configuration:
PLATFORM: Espressif 32 (2.0.3) > Espressif Generic ESP32 4M Flash, ESPEasy 1810k Code/OTA, 316k FS
HARDWARE: ESP32 240MHz, 320KB RAM, 1.81MB Flash
PACKAGES: 
 - framework-arduinoespressif32 @ 2.0.3 
 - tool-esptoolpy @ 1.30300.0 (3.3.0) 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-esp32ulp @ 1.22851.191205 (2.28.51) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3

Greetings, kuppe

TD-er, 2022-05-19:
Thanks for the feedback!