letscontrolit/ESPEasy #4054: BME280 value is not changing when sensor is removed/broken

fredvduin, 2022-05-10:
When the sensor is damaged or removed the "old" data is still available.
And when you are using syslog for debug the syslog file is exploding.
Every second 2 error messages.

17:05:24.226 -> 1318577 : Info   : BMx280 : Unable to detect chip ID (255, failed)
17:05:24.226 -> 1318578 : Info   : BMx280 : Unable to detect chip ID (255, failed)
17:05:25.231 -> 1319577 : Info   : BMx280 : Unable to detect chip ID (255, failed)
17:05:25.231 -> 1319578 : Info   : BMx280 : Unable to detect chip ID (255, failed)

Checked on other ESP12 (NodeMCU) mega-20220427
 

```

tonhuisman, 2022-05-10:
What values would then be expected for Humidity and Pressure (when selecting -127/0/125)?

tonhuisman, 2022-05-10:
This is what I stole from the DS18b20 plugin, and swapped Ignore and NaN, to have Ignore as the, backward compatible, default:

[image]

TD-er, 2022-05-10:
> What values would then be expected for Humidity and Pressure (when selecting -127/0/125)?

Humidity and pressure: -1
Temperature: -274 ;)

But to be honest, I don't think this should be done per plugin.

tonhuisman, 2022-05-10:
Once we think of a generic solution, we can delete this code again 😉

fredvduin, 2022-05-11:
I think one  value for all the plugins e.q. -127 or -274 is ok,  NaN can gives problems with MQTT
For "Error Sate Value"   select values  [ Ignore  -127(or -274)  NaN ] do I prefere

PR #4057 (2022-06-14): [Core] Add Error State Value option, implement in [P028](BMx280)
Resolves #4054

Changes:
- Add `Error State Value` option, to optionally return a predefined value if the sensor is not read successfully.
- Replaced `PCONFIG()` values by named #defines
- Formatted source using Uncrustify
- Add ESPEasy core support for Error State Value (and also Min and Max sensor values)

TODO:
- [x] Select error-values for Humidity and Pressure ~(currently using same as Temperature)~ Selected: -1
- [x] Update documentation