letscontrolit/ESPEasy #4068: Built problem error: src/ESPEasy.ino.cpp: No such file or directory

jvdVlugt, 2022-05-18:
Since today I experience problems with building  ESPEasy 
I used EspEasy-mega-20220328 as base for my custom built
It was working fine yesterday but today I get this messages during built

xtensa-esp32-elf-g++: error: src/ESPEasy.ino.cpp: No such file or directory
xtensa-esp32-elf-g++: fatal error: no input files
compilation terminated.

I see the file appear during the first steps of the built but durjng the rest of the built it is removed.

jvdVlugt, 2022-05-19:
I tried to rebuilt several times but always the same result.

xtensa-esp32-elf-g++: error: src/ESPEasy.ino.cpp: No such file or directory
xtensa-esp32-elf-g++: fatal error: no input files
compilation terminated.
*** [.pio\build\Duco_esp32dev\src\ESPEasy.ino.cpp.o] Error 1

not sure what this means. this pops up when I start the visual studio
[image]

tonhuisman, 2022-05-19:
Are you using Visual Studio (VS) or Visual Studio Code (VSCode), as ESPEasy development is advised to do using VSCode, not VS (VS and VSCode kind of have the same relation as Java and JavaScript, it _sounds_ the same, but there are some, quite crucial, differences)

For setting up your ESPEasy development environment, please use [this documentation](https://espeasy.readthedocs.io/en/latest/Participate/PlatformIO.html#platformio-with-vs-code)
Possibly the VSCode plugin `Arduino for Visual Studio Code` has to be disabled, as per the message above. I see that the documentation needs an update...

jvdVlugt, 2022-05-19:
[image]

jvdVlugt, 2022-05-19:
I also disabled Arduino for visual studio Code but stil the built probles remain

tonhuisman, 2022-05-19:
Have you disabled your antivirus for the ESPEasy folder incl. sub-folders?

TD-er, 2022-05-19:
Can you open the command prompt from within VS code and try to build it from the command line?
This way the same Python virtual env is active as when building from the GUI.
For example: `pio run -e max_ESP32_16M8M_LittleFS`

TD-er, 2022-05-19:
I just ran it myself and the first run, I also got the same error.
The second run was successful.
[image]
Put my mousepointer at the place of the command line button

jvdVlugt, 2022-05-19:
Also the same problem even with other default env selected

tonhuisman, 2022-05-19:
I'm also seeing many build issues today (yesterday, with same version and project folder has much less issues), may be related to the latest pio 6.0.1 upgrade, but also to AV updates. So, possibly, this is a pio (platformio) issue, not related to ESPEasy, only the complexity of the configuration could be of _some_ influence here.
Deleting everything from the .pio\lib_deps folder, once the build has failed (but not every time), does seem to help somewhat.

jvdVlugt, 2022-05-19:
Deleting content from pio\lib_deps folder does not help. tried it several times

TD-er, 2022-05-19:
What PlatformIO version are you running?

jvdVlugt, 2022-05-19:
2.430.820

TD-er, 2022-05-20:
I don't think that's the version of PlatformIO.

From within VS code, using the CLI:
```
PS C:\GitHub\TD-er\ESPEasy> pio --version
PlatformIO Core, version 6.0.1
```

jvdVlugt, 2022-05-20:
ok correct
version is  PlatformIO Core, version 6.0.1

jvdVlugt, 2022-05-20:
[image]
I thought this was the version number

TD-er, 2022-05-20:
You can try to downgrade the PlatformIO version, via the command line in VScode:

```
pip install platformio<6.0.0
```
This will then uninstall 6.0.1 and install 5.2.5

After the installation is complete, you need to close VS code, and delete these folders:
[image]
Then start VS code again.
This may take quite a while then as completing the PlatformIO installation does take some time.
You may see 2 build errors when you first try to build it:
- Some error about scons not being available (not sure about the exact error)
- Missing setup() and loop() -> actually the same as if ESPEasy.cpp was already deleted during compile

So you may need to call to build it upto 3 times.
Or you can first call "clean" in your PIO env. as this will download all needed packages first.

N.B. as soon as you unfold another env in the platformio menu for the first time, it will make and delete an ESPEasy.cpp file and thus cause the current active build to fail.

tonhuisman, 2022-05-20:
I've noticed that if you first get the build for the default platformio environment (`max_ESP32_16M8M_LittleFS`) it will 'calm down' the pio actions on it, as it now has all regularly checked files available, and it is quite easy to (re)build another environment.

TD-er, 2022-05-20:
OK, that kinda makes sense.
I have changed the default because I did remove the "normal 16M ESP32" build which didn't make sense to even exist.
That build option I removed was the default.

jvdVlugt, 2022-05-20:
downgrading platformio did not work but 
pip uninstall platformio 
and 
pip install platformio 
did the trick.
thanks for your suggestions and time.

TD-er, 2022-05-20:
Which version did you now install?
I discussed with @tonhuisman and we decided to go and revert back to PlatformIO 5.2.5 for now.
On your own PC you may need to do it manually, but on the build system it will do it automatically.

As far as I know, the builds made with PlatformIO 6.0.1 are OK, but there is something fishy with building it a lot each day while developing.

qiudecai, 2022-05-22:
The same error, tried all the above methods, still not working

tonhuisman, 2022-05-22:
Have you tried to open in VSCode, from the PlatformIO menu (Click the PIO icon on the left, menu on left bottom, scroll down the menu list) `Platform Core CLI`, then in the terminal area type `pip install platformio<6` and press the <Enter> key.
After completing that command, you can close that CLI session by typing `exit` and pressing <Enter>
Now you remove the folders as mentioned above by TD-er, exit VSCode, wait a few seconds, then open VSCode again in the same ESPEasy folder, select the PIO icon after it has completed starting up, expand the environment you want to build, (that takes a short while), click the Clear option, and when that's completed, click the Build option.
Though it's a lot of steps, that should work.

chemmex, 2022-05-23:
In my environment, I can compile only once after reinstall of PIO core. Then I start getting scons39.dblite Not found after each build attempt.

tonhuisman, 2022-05-23:
I have seen that first building the default PlatformIO environment (`max_ESP32_16M8M_LittleFS`) does seem to help in stabilizing the build.
And I assume you have downgraded platformio to the 5.2.5 release, per the instructions above, that's _really_ helping a lot.

TD-er, 2022-05-23:
As described here: https://github.com/platformio/platformio-core/issues/4286#issuecomment-1134269303
Can you try to update PlatformIO to the latest dev version?

chemmex, 2022-05-23:
Well, I switched to PIO dev branch (6.0.2xx) yesterday, but scons error keeps appearing after first successful build (custom_ESP.8266_4M1M). Will try with max_ESP32_16M8M_LittleFS as default env then.
Btw, downgrading to PIO core v5.2.5 gives another bunch of errors, mostly python-related

tonhuisman, 2022-05-23:
I'd expect these issues to possibly caused by the `default` environment, not any other specific env.

jvdVlugt, 2022-05-23:
I have tried the dev version and can built now and got no failure messages during built

jvdVlugt, 2022-05-23:
too bad second time it failed again with 
xtensa-esp32-elf-g++: error: src/ESPEasy.ino.cpp: No such file or directory
xtensa-esp32-elf-g++: fatal error: no input files
compilation terminated.

TD-er, 2022-05-23:
> Restart visual studio core and first got several messages ;like loop() second and thirth time a good built again.

Just keep an eye on that.
I always have these errors when I did a clean build, or deleted all files in my .platformio folder or changed something in the platformio.ini file. (and other .ini files linked in that file)
But after a successful build, I usually don't have these errors.

Zodiac-69, 2022-05-30:
I am obviously doing something stupid...
I have been trying to build an image with P057 and the plugin posted for A0YYUW, but i can't even get the normal_ESP8266_4M1M to compile.
I get the same error - error: src/ESPEasy.ino.cpp: No such file or directory - and i have followed all the ideas on how to resolve it without any success.
Any idea on how to at least get an successful build of the basic?

tonhuisman, 2022-05-30:
The instructions [above](https://github.com/letscontrolit/ESPEasy/issues/4068#issuecomment-1132558851) _should_ get you going, it helped me. The default hasn't been reset to 5.2.5 yet, but I think we're actually waiting for the _real_ fix, hopefully being corrected in a next 6.0.x release of PlatformIO, but that doesn't seem to have settled yet.

Zodiac-69, 2022-05-31:
Hi tonhuisman

I followed the instructions above the past few days without success.
I have now done the  - pip install platformio<6.0.0 and deleted the folder numerous times and then run build more than three times without success.
Finally after the 5th time i managed to have a successful build on the "normal_ESP8266_4M1M
I shall now try an do a custom build.

Z69

uzi18, 2022-05-31:
> I tried to rebuilt several times but always the same result.
>
> xtensa-esp32-elf-g++: error: src/ESPEasy.ino.cpp: No such file or directory xtensa-esp32-elf-g++: fatal error: no input files [...]

@tonhuisman @TD-er did you experience the same problem before?
some extensions conflicts with other and triggers IntelliSense ... during build time - this touch some files and you got some strange errors

TD-er, 2022-06-24:
I don't think this is still a problem anymore since PlatformIO 6.0.2
The missing `src/ESPEasy.ino.cpp` may still be a problem every now and then, especially when you completely cleaned the build folders and building it for the first time.

But I guess that's not something that will be fixed until all .ino files have been converted to .h/.cpp files.
So I will now close it.
If there are reasons why it should not be closed, please let me know.

closed 2022-06-24 by TD-er

tonhuisman, 2022-06-24:
Build often fails if there is some (PlatformIO) process running in the background, indicated by the notification icon in the right-bottom of the VSCode screen while showing the dot indicator:
[image]

TD-er, 2022-06-24:
That notification indicator isn't conclusive.

When clicking on it, I get the notifications I didn't dismiss:
[image]