letscontrolit/ESPEasy #4069: MQTT enabled will increase load to 100% and reboots the ESP

jvdVlugt, 2022-05-19:
I have a custom built ESP32 where I have 64 tasks.( 38 used)
The built is based on ESPEasy-mega-20220328
When MQTT is disabled the load is approx 28 % and the ESP works fine without problems
When MQTT is enabled the load change to 100% and stays there. The ESP restart then several times per day and finaly it freezes and need to be powered down to get it working again.
Not sure if it means something but ping response is also unstable of the ESP even sometimes timeouts  And that is independing on the location where the ESP is used.
I tried several locations and 2 separate ESP32.   the Wifi signal is strong (-59 dBm)

jvdVlugt, 2022-05-20:
I used the latest version and what I see is that the MQTT connection is dropped when the load is to high is that what the latest change is??

TD-er, 2022-05-20:
Nope
The fix I did was that it used to try and reconnect almost immediately and on a failed attempt immediately retried.
Each sending out some logs (if any log level enabled) filling up the memory and causing a lot of delays in processing these logs.
The connect attempt itself may also take some resources.
All-in-all this would lead to a serious high load and even make Wifi reconnect more difficult if that's needed.

So what I did was that there is some sort of rate limit on how quickly a new MQTT reconnect will be attempted and this interval is increasing to some max. when a connection failed (e.g. when the MQTT broker was offline there is no need to retry immediately)

jvdVlugt, 2022-05-20:
The esp32 show MQTT connection almost all the time as a red cross.
[image]

also pinging  the system is still bad.
[image]

also MQTT is working to domoticz but miss some information I have a simulation increasing regeraly in esp but in domoticz the  value jumps up not regeraly

jvdVlugt, 2022-05-20:
I get this messages when I use the monitor, is this a problem?
E (173450) ledc: ledc_get_duty(501): LEDC is not initialized
E (173455) ledc: ledc_get_duty(501): LEDC is not initialized
E (173461) ledc: ledc_get_duty(501): LEDC is not initialized
E (173466) ledc: ledc_get_duty(501): LEDC is not initialized

TD-er, 2022-05-21:
That last part is fixed in the latest code as there was an issue with PWM on ESP32 IDF4.4
You can take one of the latest GitHub Actions builds to see.
This may cause the high load too, as you can see from the timestamps those logs are sent at a very high rate (nr is msec from boot)
You can also disable the activity LED, as this PWM is used mainly on network activity, which does occur very often when connected to a MQTT broker. Disabling the activity LED is probably the easiest check to see if that helps. (Hardware tab: "Wifi Status LED" )

jvdVlugt, 2022-05-22:
I disabled the activity led and got no more messages  "ledc: ledc_get_duty(501): LEDC is not initialized"
but still the red cross remain for MQTT connection.
load is sligtly improved 
Load: | 99.13% (LC=185)

not shure if it is important but this is in  on main page 'pygit2_not_installed'

TD-er, 2022-05-22:
> not shure if it is important but this is in on main page 'pygit2_not_installed'

That's only to generate a proper description of the current git branch or tag when building.
For example, where I'm working on right now:

<html>
<body>

System Info |  
-- | --
Unit Number: | 0
Git Build: | feature/ADC_multipoint_calib_45351fe

</body>
</html>

tonhuisman, 2022-05-22:
> this is in on main page 'pygit2_not_installed'

You can install pygit2 by opening a PlatformIO Core CLI (from the PIO menus) and issuing the command `pip install pygit2`. Once completed you can again close that CLI with the `exit` command.

TD-er, 2022-05-22:
Better just do this
```
pip install -r ./requirements.txt
```
That will install all required libs for the stated version.
So if you need to go for Platformio<6 you need to change that in the txt file first.

jvdVlugt, 2022-05-23:
I Don't see this in advanced page

TD-er, 2022-05-23:
Are you running a "test" build? There the timing stats may have been excluded to make it all fit.

This is only enabled in "Custom", "normal" or "max" builds.

What tasks/plugins do you have enabled?
Anything in the rules?

jvdVlugt, 2022-05-23:
Yes I am running a built with only some plugins and yes I have some thing in rules but without the rules it also have the same result.
what flag can I add to have these timing results?

these pluins I use  P152 is my own trial
#define USES_P052
#define USES_P152   // Duco IQ unit   
#define USES_P004
 #define USES_P033

TD-er, 2022-05-23:
```c++
#define USES_TIMING_STATS
```

See the define_plugin_sets.h file in the CustomBuild folder

TD-er, 2022-05-23:
Oh and since you're using a non-released plugin, do you have a link to that source code?
It could be that something in that plugin may cause these issues.

jvdVlugt, 2022-05-23:
[_P152_Duco.txt](https://github.com/letscontrolit/ESPEasy/files/8756320/_P152_Duco.txt)
renamed the file from ino  to txt

jvdVlugt, 2022-05-23:
still timing not available  not even when #define USES_TIMING_STATS is set

jvdVlugt, 2022-05-23:
will try #define WEBSERVER_TIMINGSTATS

TD-er, 2022-05-23:
Ah yep, it needs the web page too.

jvdVlugt, 2022-05-26:
Here are the timing stats
[image]

TD-er, 2022-05-26:
Hmm a time span of 3 seconds isn't really indicative.  (remember, the stats are cleared each time the page is refreshed)
Only thing it shows is that each read from the Duco unit does take 100 msec and it looks like it is read nearly every second?
100 msec blocking is not that much of a deal usually, but every second it becomes an issue very quickly as you will miss traffic sent to the node via WiFi.

jvdVlugt, 2022-05-26:
forgot to enable MQTT
[image]

jvdVlugt, 2022-05-26:
In monitor 
[image]

TD-er, 2022-05-26:
Can you type in the command line `dns` to see what is the set DNS setting?

TD-er, 2022-05-26:
It also looks like you have enabled "Eco" mode.
Please show the other settings too, like the WiFi related settings on the tools->Advanced page (at the bottom)

TD-er, 2022-05-26:
Another point is that your system seems to load a file (no idea which one) about 170x a sec. That's also something that we should look at.

tonhuisman, 2022-05-26:
> I have a custom built ESP32 where I have 64 tasks.( 38 used)

Quoted from the original message, above

TD-er, 2022-05-26:
Ah OK, forgot about that.
Still there is quite a bit that should be looked into then as this is for sure calling some code way too often which may very well explain the reconnect issues you're seeing.
Also not sure if 38 tasks of the same plugin are the way to go.
Why do you need so many tasks of the same plugin?
I think that should be looked into first before we continue as I think this all can be done quite a bit simpler.

TD-er, 2022-05-26:
I looked through your code again and not much is really obviously wrong.

There is one thing that may be a very easy fix to reduce the used CPU cycles, but that will by no means fix your issues.
It is just some minor thing I noticed.

You're using this define:
```c++
#define P152_NR_OUTPUT_VALUES   getValueCountFromSensorType(static_cast<Sensor_VType>(PCONFIG(P152_SENSOR_TYPE_INDEX))) 
```
Every time this is called, it will call the function `getValueCountFromSensorType`.
This is not really a very "expensive" function to call, but how it is being called is not really efficient.

You're calling it in a for loop, like this (more than once):

```c++
for (uint8_t i = 0; i < P152_NR_OUTPUT_VALUES; ++i) {
```
Thus on every loop, it calls this function, even though it is not changing.

So a very easy fix can be something like this:
```c++
const uint8_t nrValues = P152_NR_OUTPUT_VALUES;
for (uint8_t i = 0; i < nrValues; ++i) {
```

Usually this isn't much of a problem, but when looking for high CPU usage, it is worth changing.

So back to the real problem here.
What are you trying to achieve here?
How often do these Duco units really have to be called? 
Is it possible that you're calling them all at the same time, thus if you have like 30 of these tasks, all set to be called every N seconds, then every N seconds these calls to PLUGIN_READ are scheduled one after eacht other and thus keep the unit busy for 30x the average time PLUGIN_READ needs.
At 100 msec on average, this means it is running for 30x 0.1 = 3 seconds straight.
If N (the interval) is then set to 3 seconds, then there is not much left to do.

If the interval of these doesn't really matter a lot, then you could try to pick a few nrs which are relative primes (e.g. 11, 17, 23 sec) and divide those intervals over all 30-ish tasks.

TD-er, 2022-05-26:
Oh and one more thing.... I need to look at how the serial port (the software object) is handling multiple instances of the same port with the same pins.
It may very well be another cause for these kinds of issues as it normally isn't shared like this.

TD-er, 2022-05-26:
Another idea...
Given that the call to PLUGIN_READ (done every <Interval> nr of seconds) may trigger some cadance of high CPU load, you may want to call them in a staged manner.
For example using the rules, you can call `taskrun` of every task in a loop.

Something like this:

```
On System#Boot do    //When the ESP boots, do
  looptimerset_ms,1,500 // Timer used to "run" every Duco task
endon

On Rules#Timer=2 do
  // %eventvalue2% is the loop counter 
  let,1,%eventvalue2% // Store in a variable so we can perform calculations
  let,1,[int#1]%38  // Modulo the nr of tasks we need (38 here)
  let,1,[int#1]+1   // Add the task nr of the first Duco task
  taskrun,[int#1]  // We now run the task modulo the loop counter
endon
```

You can set the `Device[].TimerOptional` to true in your code too.

jvdVlugt, 2022-05-26:
I added the code above in rules and reboot the esp
the result for timing is the same
where do i put Device[].TimerOptional?

TD-er, 2022-05-26:
Just as an example the part of the P001 code to show you where to put it.

```c++
case PLUGIN_DEVICE_ADD:
    {
      Device[++deviceCount].Number           = PLUGIN_ID_001;
      Device[deviceCount].Type               = DEVICE_TYPE_SINGLE;
      Device[deviceCount].VType              = Sensor_VType::SENSOR_TYPE_SWITCH;
      Device[deviceCount].Ports              = 0;
      Device[deviceCount].PullUpOption       = true;
      Device[deviceCount].InverseLogicOption = true;
      Device[deviceCount].FormulaOption      = false;
      Device[deviceCount].ValueCount         = 1;
      Device[deviceCount].SendDataOption     = true;
      Device[deviceCount].TimerOption        = true;
      Device[deviceCount].TimerOptional      = true;
      Device[deviceCount].GlobalSyncOption   = true;
```

N.B. the 38 and the first task index were a guess I made, make sure to check those.

Please show the timing stats over at least 2 minutes running these rules.

jvdVlugt, 2022-05-27:
I am not sure if the  the 1700 for  the fifty per second is realy strange
I tested with a dummy device  is has a reading of 42 on  the fifty per seconds reading.
I added  9 more dummy devices and that result in a reading of  about 480 on the fifty per second reading.
Because I have 38 readings for the duco device the result  38 times 42 is about   1600 readings

I found that because I have an esp in the same network that has the same name and unit_number caused the mqtt problem..
Now the load is 13.5 % with 38 duco devices and 10 dummy devices.

TD-er, 2022-05-27:
Well the conclusion that 1700 must be really strange is that I had forgotten you're running a build that allows for > 32 tasks.
In my mind 32*50 < 1700, so that's why I got confused.

What have you now done to get the load at 13%? Using the rules I suggested?

jvdVlugt, 2022-05-27:
I used the rules you send without any improvement so I removed them again.
It seem that the ESP takes care of this I see on an LED on the convertor that the ESP askes about every second one call. (also seen in the timing stats). 
And Device[deviceCount].TimerOptional  = true; caused the ESP to be unstable. Every edit I tried caused the esp to rebooted.
So finally  I re-used the original version. But found that the ESP had the same name and number as another one in the network,
After change the name and number the load dropped  and now I see a load  14 %.

I don’t know what made the system stable but now it looks fine. So hopefully the ESP keeps running as it does now.

I really appreciate your  time and  suggestions, it helps me a lot to get the system running.

TD-er, 2022-05-27:
Maybe you do use the unit nr in the MQTT client ID?
And if the broker sees 2 nodes connecting with the same client ID, it may disconnect one of them (or both)?
Just an idea, not sure if that is what is happening.

TD-er, 2022-06-24:
I think this issue can now be closed.
If not, please let me know