letscontrolit/ESPEasy #4071: VS Compile error with Adafruit GFX Library @ ^1.11.1

uwekaditz, 2022-05-19:
With the latest changes (using Adafruit GFX Library @ ^1.11.1) my VS environment cannot compile anymore!
I get following error message:
```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp8266_4M1M_board.html
PLATFORM: Espressif 8266 (2.6.3) > Espressif Generic ESP8266 ESPEasy 4M Flash 1M FS
HARDWARE: ESP8266 80MHz, 80KB RAM, 1015.93KB Flash
PACKAGES:
 - framework-arduinoespressif8266 2.7.4+sha.2843a5a 
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Converting ESPEasy.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing adafruit/Adafruit GFX Library @ ^1.11.1
Warning! Could not find the package with 'adafruit/Adafruit GFX Library @ ^1.11.1' requirements for your system 'windows_amd64'

```

tonhuisman, 2022-05-19:
Platformio has been upgraded to 6.0.1, I've removed all pio cache folders and .pio\lib_deps in the project folder, that helped

uwekaditz, 2022-05-19:
It's strange but it works.
I moved `c:\Users\user\.platformio\.cache\` to `projekt\.cache\` and
`projekt\.pio\libdeps\ `to `projekt\libdeps\`
VS created two new folders
`c:\Users\user\.platformio\.cache\` and
`ESPEasyPrivat\.pio\libdeps\ `
but it could update the libraries and compile.

TD-er, 2022-05-19:
Sometimes Github also timeouts when fetching libraries.
I've also seen it on GitHub Actions builds, where a build may be cancelled after 1h (or slightly less than 60m) where the download of a library takes forever.
Maybe the library download may fail and get into some limbo state preventing it to download again as it should be there in the cache...

uwekaditz, 2022-12-28:
No more problems updating the libraries.

closed 2022-12-28 by uwekaditz