letscontrolit/ESPEasy #4073: Usage of more than 10 Nextion Command lines with more than 50 chars

DrMucki, 2022-05-21:
I use [ESP_Easy_mega_20211105_display_ESP8266_4M1M Nov 5 2021] in combination with a Nextion Display.
For each value i want to move to Nextion Display i use one line. eg 

page0.tPool.txt="[temp#tPool]"

there are only 10 lines with a maximum of 50 characters.

I have to send 10 values and want to change the state of 9 Buttons (show different pictures). But with only 10 lines it does't seem to be possible. Or do i get something wrong here.

Thank you for your nice work by the way!!!!!!

TD-er, 2022-05-22:
OK, we have done similar things with other plugins, like the OLED Framed, where at least the length per line was more of an issue than the nr. of lines.
However, when changing this, two things are rather important:
- Forwards compatibility of settings
- Backwards compatibility of settings.

The first one seems obvious and is rather easy to implement too.
But the second one is more difficult as it may make it impossible to revert back to a previous build once the settings are stored in the "new" format.
I don't know how much space is left in the settings of that plugin (all tasks can have upto 1k of settings), but if that's plenty, then we could see if simply adding more lines with the same format is possible.
However, I do prefer to have string settings like these stored more effectively, to allow for example more lines in the same amount of storage, when only short lines are used.

So we have to have a look at it (probably @tonhuisman will beat me to it  ;) )

DrMucki, 2022-05-22:
When the characters per line are extended for using long lines, where for example more than one change, separated by ; can be written would also extend the possibilities.

Example of what I mean….
page0.tPool.txt="[temp#tPool]" ; page1.Test.txt="[temp#outside]"; page2.test3.txt="[temp#solar]"

Thank you 🙏

tonhuisman, 2022-05-22:
The current 10 line of 50 characters setup consume about half of the available storage, so while increasing either dimension would improve stuff a bit, it will still be a limited improvement. (Similar issues exist for the TFT and eInk displays, but can be circumvented by sending content from rules).
This implies, AFAICS, that the content/commands you want to send _can_ be sent from rules, and is most likely the most used way of filling in the content on the display.
When using the sensor events to fill in the exact data, you get updates as soon as they are available, instead of having to wait for the update interval of the display:
```
on BME#Temperature do
  nextion,page0.tPool.txt="%eventvalue1%" // Assuming the taskname is Nextion
endon
```
NB: You might want to update to the latest ESPEasy release to get better Rules processing performance.

----------------------------------------------
NB2: I have 0 (zero) experience with Nextion displays, but I'd like to suggest to try this as a possible alternative solution:
- Can a second (and optionally 3rd) instance of the plugin be configured, with similar hardware configuration, but 1 twist: The `RX` pin should only be configured for 1 plugin, so touch events (etc.) are handled only once. (Configuration labels seem to me a bit confusing...) I think it requires that a serial port on the ESP is used, and not I2C serial, (I've been looking at the config using and ESP32, so I have 3 H/W serial ports available).

As only 1 task update interval is executed at a time, the commands should be sent _after_ each other, not in parallel, as that would obviously not work as intended.

DrMucki, 2022-05-22:
Rules rulez! That was exactly what I was looking for. Thank you. I put my sensor readings there and it works like Charme. The rest will do also i think.

thank you

TD-er, 2022-05-22:
OK, so this only needs update in the documentation?

DrMucki, 2022-05-22:
Yes. A documentation would be nice. But the help is incredible here! Thank you again

PR #4280 (2022-10-14): [P075] Increase nr. of lines for commands to send to Nextion
Resolves #4073 

As requested, increases the number of lines from 10 to 20. The line-length can't be increased without breaking compatibility.

NB: When saving > 10 lines with commands, and reverting to an older release of ESPEasy, the command-lines numbered from 11 and higher will be lost if the plugin settings are saved.
NB2: When available memory is low, saving settings can be problematic, temporarily disabling other plugins may help to avoid crashes.
- Documentation is updated.
- Some minor code improvements applied.

TODO:
- [ ] [... cut]
closed 2022-10-14 by TD-er