letscontrolit/ESPEasy #4074: Unable to resolve configuration with compilerPath

jvdVlugt, 2022-05-22:
got several messages
Unable to resolve configuration with compilerPath "C:/Users/johan/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc.exe".  Using "cl.exe" instead.

problem is that the folder is not  toolchain-xtensa32 but toolchain-xtensa-**esp32**
[image]

TD-er, 2022-05-22:
I also have it in this folder:
```
C:\Users\gijsn\.platformio\packages\toolchain-xtensa-esp32
```

By the way, I may have seen similar build issues on my laptop today.
Tried to build it for about 10 times and never succeeded, so I might want to check to see what I need to do there to fix it.

That's an older installation, as I almost never compile on that laptop since it takes sooooo long... to compile.

TD-er, 2022-11-14:
This has been fixed a while ago.

closed 2022-11-14 by TD-er