letscontrolit/ESPEasy #4078: Sending Values from Nextion Display to ESPEasy not possible?

DrMucki, 2022-05-28:
Hi out there,

Within the Sexton Display Plugin it is possible to send 2 variables to use in ESPEasy one is idx, the other one is value.
For button presses a convenient way is to use the PIPE s or the PIPE u format to send these values. I prefer the PIPE u format used as described in the documentation

print "|u,i50,n,s-3" // Send idx 50 and value -3 to ESPEasy
printh 0a            // Required print termination character

I wonder if it is possible to send a variable as "s" means as value. 
Within in Nextion-Display a have a GUI, where I can setup a temperature using buttons. The temperature is a "number" field, when clicking a button, the touch press event for this counts up the number.

n0.val=no.val+1 where "val" is the value of the number"n0". 

What I want is to use the whole variable called n0.val  to get into the "value" of the ESPEasy (here the s variable)

what I tried so far is within the Touch Press Event I used the following code
n0.val=n0.val+1
print"|u,i506,n,s"
print n0.val ( to but the variable in there).  
print 0a

but this and trying to enter it within the first print action did not meet the goal...
Any suggestions or is it a Request?

Thank you, and have a nice weekend

Marc

tonhuisman, 2022-05-28:
Have you read to original issue-template? It suggests to search the [forum](https://www.letscontrolit.com/forum) and AFAICS, there is quite some info on how to use the Nextion displays...

DrMucki, 2022-05-30:
Here is the solution.....

The print|u with the "" is sending strinigs over the serial port, when you use print xy.val a value is sent, the hex values for text is different from an integer value.
So we have to convert the variable into a text (String) before we sent it after the print pipe expression.
covx is used for conversions between numeric and strings

so the following code does the job Any variable stored in va0.val can be sent and is received as value within ESPEasy

va0.val=va0.val-1
if(va0.val>40)
{
  va0.val=40
}
covx va0.val,t0.txt,0,0
print "|u,i230,n,s"
print t0.txt
printh 0a

Sorry for messing up the issues !

tonhuisman, 2023-09-01:
This is moved to the forum, and can be closed here.

closed 2023-09-01 by TD-er