letscontrolit/ESPEasy #4079: Wont Start or Wont Wake up under battery usage.

crazybanane, 2022-05-30:
ESP 32 didnt start or wake up from DeepSleep under Battery usage.
Under Config Sleep Mode i set 30 Seconds Wake 300 seconds Sleep.
I connected a Li-Ion Battery with a 1700-3302 LDO and an Capacitor.
I read out via Devices list an Analog signal from the battery via an Voltage Divider on IO34 what works great if i delete Deep Sleep and start the Esp via USB connection.

I now work on ESP_Easy_mega_20220427_normal_ESP32_4M316k
My boards are AZ-Delivery markes boards in MH-ET live Mini style.

### Steps already tried...
- [many times ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [same build but custom variant  ] Tested previous/other build (mention which one already tested)
- [3 nodes same problem ] Tested on other node to make sure hardware isn't defective.
- [cant tell ] Verified if the problem is limited to a single plugin/controller

I tested similar with esp8266 some time ago but i know for deepsleep wake up there was a pin that i have to Pull up or Pull down that the controller will start again. is there something on ESP32 boards? Or is it a other Problem?

tonhuisman, 2022-05-30:
Does it work as intended when running from USB power? If so then, most likely, the serial chip is pulling pins in a wrong state because that is powered from 5V, and when battery powered, the board only gets 3.3V. Might need a hardware mod to fix that.

crazybanane, 2022-05-30:
Yeah it works nice with USB power but didn't know the problem on hardware. Is there a pin like gpio16 from 8266 to start or wake up?

TD-er, 2022-05-30:
Nope, but there is the EN pin on the ESP which acts as a reset or power good pin.
Typically most boards have some kind of R/C pair connected to this pin to delay power on for the ESP to give the power supply some time to get a stable voltage.
But the USB to serial adapter may also toggle this pin along with GPIO-0 to enter flash mode for programming it.
If this USB to serial chip is not in a well defined state, as it is not powered on, it may either hold the EN pin low and/or the GPIO-0 pin.
Either will prevent the ESP from booting, or even worse, as the state of such a chip is undefined, it may also be "sometimes" working.

tonhuisman, 2022-05-30:
This [forum thread](https://www.letscontrolit.com/forum/viewtopic.php?t=8627) is about a Wemos D1 Mini Pro (ESP8266), but describes similar issues and has a few suggestions.

sobolkz, 2022-05-31:
> is there something on ESP32 boards?

Hi, I'm using Wemos D32 Pro, no such problems. As I know this boards schematic - USB 5V was directly connected to battery charging chip AND to 3.3 LDO through Shottky diode. But battery out was connected to this LDO by MOSFET transistor
[image]
Try to check you board's scheme. For example if D2 at my scheme was dead by short connection it'll be affected to battery powering.
Also

> but there is the EN pin on the ESP which acts as a reset or power good pin.

EN should be externally pulled UP
[image]

TD-er, 2022-05-31:
> EN should be externally pulled UP

Yep, so if it is pulled down, the ESP will not boot.

crazybanane, 2022-05-31:
Hey,
so i tried the thing to Pull Up externaly Via LDO and an 10k resistor on RST (EN) nothing works.
You called that there is also the IO0 Pin at what state it has to be pulled up or down?
I got some software from Silicon Labs where you can switch the Outputs for the CP2104 can i solve the problem with edit the parameter?

Here is the Layout what hopefully Az-D copied:
[image]
 Dont look for the Pink line its not from me.

And here ist the software for Port conrtol cp2104:
[image]

Do you find a way if i is possible to switch there something? only other option you can choose is PUSH-PULL or Latch...

crazybanane, 2022-05-31:
So i tried something else:

1.10k pull up on rst EN
2. 10k pull up on gpio0
3. 10k pull down on gpio0
4. 10k pull up on rst and gpio0
nothing changes.

crazybanane, 2022-05-31:
I found a way, even when it is dirty.
By 2 of them the function will work as follow.
Connected battery directly to VCC
The other one still wont boot without USB

TD-er, 2022-05-31:
Looks like there is a diode inbetween the 5V pin and the USB to serial chip.
This is meant to protect the PC, but here it does prevent your USB to serial chip from starting.

The GPIO-0 pin should be pulled up to boot the ESP.
What is the minimal voltage the USB to serial chip can still operate on?
If this is less than 3.3V, then you could add a diode between the 3V3 pin and the power pin of the USB to serial chip.

sobolkz, 2022-06-01:
> I connected a Li-Ion Battery with a 1700-3302 LDO and an Capacitor.

> I found a way, even when it is dirty.
> By 2 of them the function will work as follow.
> Connected battery directly to VCC
> The other one still wont boot without USB

Hmm, may be you LDO could'n work at ~4V from battery? For example it's internal drop out more than 1V and it's out doesn't enought for getting power at boot for ESP32 with stable 3.3V?

uzi18, 2022-06-02:
@sobolkz from the datasheet: _The MCP1700 is capable of delivering 250 mA with only 178 mV (355mV max) of input to output voltage differential (VOUT = 2.8V), VIN is 6V max_

tonhuisman, 2022-06-02:
> delivering 250 mA

That seems quite poor for an ESP32...

TD-er, 2022-06-02:
> > That seems quite poor for an ESP32...
> 
> Yes, as I remember - 0.5A minimum power source by Espressif ESP32 wroom datasheet.

And according everybody else who replies to threads discussing the minimum requirements :)

tonhuisman, 2023-09-09:
Does this need any follow-up? If not, it should be closed.

closed 2023-09-09 by TD-er