letscontrolit/ESPEasy #4080: [p004] ds18b20 doesn´t work with esp32

chromoxdor, 2022-05-31:
the address field stays empty...
double checked it with an esp8266

TD-er, 2022-05-31:
Which pin do you use on ESP32?
Did a build from a few months ago work? (before IDF 4.4 was merged)
For example one of end 2021 just to be sure.

chromoxdor, 2022-05-31:
no luck with 20211224...

chromoxdor, 2022-05-31:
> Please also check this table: [Best pins to use on ESP32](https://espeasy.readthedocs.io/en/latest/Reference/GPIO.html#best-pins-to-use-on-esp32) 34 should not even be possible to select in the Dallas plugin as it is input only. So that's a bug for sure.
> 
> pin 13 should be fine, but please also check the [Dallas documentation](https://espeasy.readthedocs.io/en/latest/Plugin/P004.html#more-pictures) to check if you added a pull-up resistor too.

pull up is set and it works as i said with an esp8266... 
You are right with gpio34.. missed that

chromoxdor, 2022-05-31:
dis a clean install with 20210114 and tried pin 25...this works.. i´ll do now further systematical testing and will report this here..

TD-er, 2022-05-31:
Some boards have GPIO 13 & 16 used for I2C.
So if you have some I2C device already mounted to the board, it might affect the rather timing critical bitbanging needed for the Dallas sensors.

TD-er, 2022-05-31:
Thanks for the update.

chromoxdor, 2022-05-31:
Mhmmm... something still smells funny...
i tried gpio 13,14,26.. these work fine... but since i needed different gpios i tried 32 and 33 they do not work but they should, right?

chromoxdor, 2022-05-31:
update: 
27 does work 
25 doesn´t

TD-er, 2022-05-31:
Sounds like the change to IDF 4.4 might still have broken something, so please test with the latest GitHub Actions builds too as there has been an issue with GPIO handling on ESP32, which is now fixed.

chromoxdor, 2022-05-31:
I tried... it didn´t work.. and than i flashed every release (and the latest action build) since 20211105.. with flash erases in between.. and it worked every time(i only tried pin 33) 😭 ... 
I triple checked the wiring but it should be highly unlikely since its always the same pins that do not work.

And once again i flashed two of the latest acionbuilds and there was no ds18b20 discoverable. 
I don`t get it and i must admit, that since i almost spend the whole day with this my testing has not become more reliable.
🤪

Can somebody confirm for me that there is an issue with the aforementioned pins, at least in the latest action build (https://github.com/letscontrolit/ESPEasy/actions/runs/2414526874)?

chromoxdor, 2022-05-31:
I just did a little sanity check for myself because i needed it and i wanted to rule out a hardware issue:
[image]

It seems ota vs serial flashing makes a difference..but i couldn’t  determine how

TD-er, 2022-05-31:
If you go from < IDF 4.4 to IDF 4.4, then I can imagine you may need to do a serial flash.
They also changed a lot in the bootloader and there should be a compatibility mode for new builds on old bootloader (you don't update the bootloader when flashing OTA)

There is one more difference, but this is less likely to be an issue.
There are 2 partitions for the 'sketch'.
When flashing via serial, always the 1st partition is used.
However when doing an OTA update, you will write to the 'other' partition and then the ESP will be set to boot 'the other' partition.
Thus via OTA updates, you will constantly switch between partitions.
What if for some reason there is something wrong with your flash somewhere in the area allocated for this 2nd partition? Then you may get very hard to reproduce issues as it seems unclear to why one build may work and another one fails.

chromoxdor, 2022-06-01:
Oh man... i thought i could use another gpio but then cascading multiple sensors do not work.
So i serial flashed one of my esp32 to 20220427 and there on every pin with every combination of ds18b20 (i have 4) it works fine! Since i need the ds18b20 and the newest builds because of other changes can you have a look into this @TD-er ...

TD-er, 2022-06-01:
The idea is that you should use all your Dallas sensors on the same GPIO pin.
I have tested it upto 12 sensors on a breadboard and at least on ESP8266 this always worked fine (well not always, but since the latest changes I made about 12 - 18 months ago it is working fine)
Have to admit, I have not tested it myself on ESP32.

So I don't know why it shouldn't work on ESP32 apart from the strange issue I have seen regarding GPIO handling in the 20220427 build, which is fixed in the latest GitHubActions builds.

chromoxdor, 2022-06-01:
I use them all on the same pin.
The thing is, the work fine in 20220427 now.
The action builds after that seem to cause the issues...

TD-er, 2022-06-01:
OK, that's good to know, so that might then be related to another issue we're having with the new PlatformIO.

chromoxdor, 2022-06-01:
[image]

TD-er, 2022-06-01:
Ah now I see why it was possible to select an input only pin as pin.
There is an option to use a separate pin for RX and TX (used by some Shelly units to isolate the IO from mains voltage)
So only when using a 2nd pin, you can use an input-only pin for this.
But in all other cases you should use a pin that can handle input and output.

I should add an error message when an input-only pin is selected and not a 2nd pin.

chromoxdor, 2022-06-01:
Apparently there is one ds18b20 that works on pin 33 and 32
I used the same setup und just flashed 20220503 (wanted to try an earlier action build)
**It seems, that instead of not working at all espeasy simply became picky :)** 
Thats what probably caused alle the confusion when i tried yesterday... 

[image]

TD-er, 2022-06-01:
Hmm there are many reports out there about some (fake!!!) sensors not willing to work together with > 3 in a string. Not just with ESPEasy, but with any library or "competing" firmware.
I never managed to reproduce this, even though I bought many -known to be fake- Dallas sensors to make sure I could either fix this or at least reproduce this.
So I did buy from Ali, DealExtreme and Banggood solely based on "impossible" price and shadyness but never were so 'lucky' to get units that actually didn't want to work together with more than 3 on a string.

Can you swap out units to see if other combinations (upto 3) work well together?
And maybe we can arrange some exchange of sensors so I have known-to-be ultra-fake units to test.
You may have a limited edition fakes in your hand :)
I do have quite a lot of units that do work fine with > 3 on a string.

chromoxdor, 2022-06-01:
I just tested 4 of them
They can´t be that much of a fake :)
[image]

sanyafifa, 2022-07-11:
Hi all
on the new firmware (20220616) does not work for me on GPIO 25. Any ideas?
[image]

TD-er, 2022-07-11:
Which build?
The last build should work again. (not sure about GPIO-25, have to check for that specific pin)

sanyafifa, 2022-07-11:
> Which build?
ESP_Easy_mega_20220616_normal_ESP32_4M316k Jun 16 2022

chromoxdor, 2022-07-18:
> Which build? The last build should work again. (not sure about GPIO-25, have to check for that specific pin)

Unfortunately the issue is still there. Multiple ds18b20s do not work with newer builds as  you can see in the following screenshots:
[image]
[image]
 (i tried multiple different gpios)

chromoxdor, 2022-07-18:
with or without the external one?

TD-er, 2022-07-18:
I think you still need an external one, but first try to use without external one.
The reason I think you still need one is that the internal one is used on the ESP side, not near the Dallas sensors.
So the wires will pick up noise.
The best position is to have the pull-up close to the (last) Dallas sensor.

TD-er, 2022-07-29:
This has now been fixed.
See: https://github.com/letscontrolit/ESPEasy/pull/4164

closed 2022-07-29 by TD-er