letscontrolit/ESPEasy #4083: DS18b20 - Value doesn't change when sensor disconnected, Error State Value can't be used in Rules

stwa2020, 2022-06-08:
D1 Mini Build: [ESP_Easy_mega_20220427_test_B_ESP8266_4M1M Apr 27 2022

I've two DS18b20 each Sensor is connected with 4K6 Ohm Pullup resistor, 
1x GPIO4
1x GPIO2
Both are working and measure the temperature,

I've created a rule to increase a counter when the Error State Value: 125 is reported but this counter doesn't increase.
I've tested the counter using temperature >=25 and it works, the counter increase, this is the rule for it.

```
On Temp1#Int_Sensor1 do
 if [Temp1#Int_Sensor1]>=25
  TaskValueSet 5,1,[Temperature_check#Check1]+1
 endif
endon
```

When I use now the rule to increase the counter when the Error State Value: 125 is reported, it's not inreasing

```
On Temp1#Int_Sensor1 do
 if [Temp1#Int_Sensor1]>=120
  TaskValueSet 5,1,[Temperature_check#Check1]+1
 endif
endon
```

Also when the sensor is disconnected in the GUI is still show the last meassured value and not 125, in my case it's 25 and 125 when the error is reported in the log but it always jump back to the 25 value.

I would like to create a rule which increase a failure counter, when the value is normal read it reset the failure counter to 0 and when the failure counter is >15 it should turn on a gpio and publish and error via MQTT but this is not possible when the rules are not working for the Error State Value and when the Sensor has old values reported when it's disconnected.

 
Thank you for your feedback,

Stefan

tonhuisman, 2022-06-08:
For increasing that value it may be needed to use an intermediate variable for the addition:
```
On Temp1#Int_Sensor1 do
  If %eventvalue1%>=120
    Let,1,[Temperature_check#Check1]+1
    TaskValueSet,Temperature_check,Check1,%v1%
  Endif
Endon
```
I've taken the liberty to apply a few best practices for rules: Use `%eventvalueN%` where possible, as that holds the value _at the time of the event_ and: Use `taskname`/`valuename` for `TaskValueSet` instead of  the numeric equivalents, that will make your script more portable and readable.

For the Error State Value I've been working on a generic approach, via PR [#4057](https://github.com/letscontrolit/ESPEasy/pull/4057), though that is currently only implemented for the BMx280 plugin (in that PR), the DS18b20 is also one of the 'usual suspects' to get this improvement.

NB: To reduce the number of GPIO pins used, multiple DS18b20 sensors can be connected on the same GPIO pin, even when used in separate tasks, just select the correct identifier from the drop-down list 😉.

tonhuisman, 2022-06-08:
Ah, 💡 at closer inspection, when an error state is set, no event is generated, nor is the value sent out to any configured Controllers. We might need to make that configurable.

stwa2020, 2022-06-08:
thank you, your Rules example look nicer than mine :-)
The reason why I use two GPIO is the different cable length and I need maybe a 3k3 resistore on one of the GPIOs.
The whole setup with the D1 Mini include PWM to 0-10V, 2x MOSFET for 10V, Neopixel, 1x Relay and the 2x DS18b20 are used to drop the power for PWM to 0-10V and both MOSFET when a high Temp is reached or when the DS18b20 is not working. But I have to use now an external device for the temperature control.

stwa2020, 2022-06-10:
one additional strange behaviour I've found out,

with this rule below the Check1 counter will not increase when one DS18b20 Sensor is connected.
```
On Temp1#Int_Sensor1 do
  If %eventvalue1%>=120
    Let,1,[Temperature_check#Check1]+1
    TaskValueSet,Temperature_check,Check1,%v1%
  Endif
Endon
```

When I add a second DS18b20 Sensor the Check1 counter will increase when a read error happen.

uzi18, 2022-06-10:
Please add a screenshot of your Temp1 configuration also you can add some info logs when an error occurs.

TD-er, 2022-06-10:
On the tools->Advanced page, please uncheck this optimization to allow the order of events in the rules to be changed.:

[image]

I will disable this optimization by default in a next build.

tonhuisman, 2022-06-10:
I'd be interested to know if the check count is actually updated if the Devices page is refreshed (by clicking the Devices tab again), According to the log @stwa2020 showed the counter *is* updated on error, so it may just be a screen-update 'issue', not related to ESPEasy, other than it requires javascript to run on the page, or else the values can't be updated.
What browser are you using, and is javascript execution allowed?

stwa2020, 2022-06-10:
I did some tests but I need to verify them again to be sure no mistake was done, all the tests have been done with Optimized Rules Cache Event Order: disabled. Temp2 is disabled and no sensors are connected.
When a reboot is done the values are set to 0.
To create errors I've disconnected the data line for each sensor individually.

3x Sensors, forcing errors, increased the Check1-3 counters.

[image]

After a reboot was done,
[image]

[image]

Now with one sensor connected, the Check1 counter didn't increased.
[image]

[image]

I did the same tests with two sensors connected and the Check1 & 2 counters increased but not with the correct values,

[image]

I've checked the log for ERROR! and 125 and it looks like the 125 are not counted:

_Two Sensors connected
**8x Failures, 4x Error! and 4x 125, the task triggered counter shows 4**
18593: DS   : Temperature: Error! (28-aa-7b-ec-19-13-02-0a [DS18B20])
30635: DS   : Temperature: 125 (28-aa-7b-ec-19-13-02-0a [DS18B20])
32604: DS   : Temperature: 125 (28-aa-7b-ec-19-13-02-0a [DS18B20])
34604: DS   : Temperature: 125 (28-aa-7b-ec-19-13-02-0a [DS18B20])
38616: DS   : Temperature: 125 (28-aa-7b-ec-19-13-02-0a [DS18B20])
98711: DS   : Temperature: Error! (28-aa-7b-ec-19-13-02-0a [DS18B20])
116749: DS   : Temperature: Error! (28-aa-7b-ec-19-13-02-0a [DS18B20])
132768: DS   : Temperature: Error! (28-aa-7b-ec-19-13-02-0a [DS18B20])

**50x Error! this increased the counter to 50**
12600: DS   : Temperature: Error! (28-ad-2a-59-2f-14-01-44 [DS18B20])
16600: DS   : Temperature: Error! (28-ad-2a-59-2f-14-01-44 [DS18B20])
18592: DS   : Temperature: Error! (28-ad-2a-59-2f-14-01-44 [DS18B20])
	...
297088: DS   : Temperature: Error! (28-ad-2a-59-2f-14-01-44 [DS18B20])
301095: DS   : Temperature: Error! (28-ad-2a-59-2f-14-01-44 [DS18B20])_
 
I guess I need to run the test again on the weekend with 1x sensor connected and check which errors I get in the log and do the same with 2x sensors connected and check if Error! or 125 increase the Check1 & 2 counters.

TD-er, 2022-06-11:
As Ton suggested, you may want to add a system info task and set that at 1sec interval.
This way the Javascript does update the values every second on the devices page.

I may have to check the source of the Dallas plugin to see what may happen if several instances are enabled on different pins.
Also can you show the task config regarding the pins, to see if you set the 2nd optional pin? At least it does show it on the devices page. I think I may have internally set it to both if the optional pin is not set, but I think it should not show it like this on the devices page.

stwa2020, 2022-06-11:
I did a few tests, after each reboot the counters are set to 0, RULES1 and the Check1 & 2 & 3 Values are counted with RULE2.
I've attached TEST.pdf with screenshots and all LOGS saved from ESP logging between the tests.

The intervall for the Sensors was set to 1s

Let me know if I should make other tests. 

Attachements:
[TESTS.pdf](https://github.com/letscontrolit/ESPEasy/files/8884364/TESTS.pdf)
[LOG_all_three_sensors_disconnected_after_reboot.txt](https://github.com/letscontrolit/ESPEasy/files/8884356/LOG_all_three_sensors_disconnected_after_reboot.txt)
[LOG_three_sensor.txt](https://github.com/letscontrolit/ESPEasy/files/8884357/LOG_three_sensor.txt)
[LOG_two_sensor_address_swap.txt](https://github.com/letscontrolit/ESPEasy/files/8884358/LOG_two_sensor_address_swap.txt)
[LOG_two_sensor.txt](https://github.com/letscontrolit/ESPEasy/files/8884359/LOG_two_sensor.txt)
[LOG_one_sensor_with_disconnect_later.txt](https://github.com/letscontrolit/ESPEasy/files/8884360/LOG_one_sensor_with_disconnect_later.txt)
[RULE2.txt](https://github.com/letscontrolit/ESPEasy/files/8884361/RULE2.txt)
[RULE1.txt](https://github.com/letscontrolit/ESPEasy/files/8884362/RULE1.txt)

stwa2020, 2022-06-11:
FIXED, I took a new D1 Mini wrote the empty 4MB to it, installed ESP_Easy_mega_20210503_normal_ESP8266_4M1M configured everything. It works the Sensors are not creating any sample read errors even with 1s intervall. Updated the firmware to ESP_Easy_mega_20220427_normal_ESP8266_4M1M Apr 27 2022, still the same no errors. When I disconnect one sensor it create 125 and increase my CheckX counters. I will do the same actions tomorrow with the old D1 Mini and try it again.

stwa2020, 2022-06-13:
The ESP was partly broken, the sensors are working now after I've replaced the ESP with a new one and started again from scratch,

Please let me know if I should open a new Topic for this behaviour below,
The other point I've found in version ESP_Easy_mega_20220427_normal_ESP8266_4M1M Apr 27 2022 compared with ESP_Easy_mega_20210503_normal_ESP8266_4M1M, is it's not possible to set the Generic - MQTT Import values with taskvalueset it create the message "NOT_A_DUMMY_TASK". 

This create the following situation,
There is a rule when the temperature reach 50°C it should turn of other GPIOs and publish the information to the MQTT Broker and set the  Generic - MQTT IMPORT Values on the ESP to 0 and on the MQTT Broker to 0 with publish, this is not possible because on the ESP it create "NOT_A_DUMMY_TASK" when the taskvalueset command for Generic - MQTT Import Value is used.

Also when the MQTT Broker is stopped it's not possible to set the values on the ESP to 0 by using taskvalueset with MQTT#Disconnected.
Is there a way to set the Generic - MQTT Import values to 0 (or any other value) without a connection to the MQTT Broker ?

tonhuisman, 2022-06-13:
> it's not possible to set the Generic - MQTT Import values with taskvalueset

That's actually a feature 😉 
In Tools/Advanced there is a setting that lets you override this new behavior (I don't have a recent ESPEasy where I am now, so can't give the exact text for the setting, but it does include TaskValueSet).

stwa2020, 2022-06-13:
thx, found it: Allow TaskValueSet on all plugins:

UriUriK, 2023-04-02:
Same problem as starter thread. However, %eventvalue% does not return an error code, which is set to "-127". The plugin returns the previous temperature of the disabled sensor, and this is critical for controlling an electric boiler. Frmware :[esp_easy_mega_20230306_normal_esp8266_1m mar 6 2023]

UriUriK, 2023-04-02:
Same problem as starter thread. However, eventvalue does not return an error code, which is set to "-127". The plugin returns the previous temperature of the disabled sensor, and this is critical for controlling an electric boiler. Firmware [esp_easy_mega_20230306_normal_esp8266_1m mar 6 2023].

UriUriK, 2023-04-02:
I am adding. Connected 4 sensors on gpio14. If I turn off all the sensors at once, then the situation is as described earlier. Now I tried to turn off only the sensor that determines the temperature of the boiler, I got "-127". Then I checked the following, if there is at least 1 of 4 sensors, then I get "-127", otherwise the situation is the same as in the previous post.

UriUriK, 2023-04-02:
I checked the situation with a single sensor. When the latter is disabled, the plugin returns the last measured value, not the error code.

UriUriK, 2023-04-02:
I found a temporary solution for those who want to use the firmware to work with the boiler, until the problem is fixed. at least 2 sensors must be connected, one of which must be built into the control module and monitor the temperature of the relay or triac.