letscontrolit/ESPEasy #4086: [feauture request] rules highlighting

chromoxdor, 2022-06-11:
**The basic rules highlighting project is for now finished on my side.**
It consists of three files:
codemirror.js    -> the main program
codemirror.css -> styling 
espeasy.js         -> language specific file
anyword-hint.js -> hint program with wordlist

**implementing is straightforward:**

webserver.cpp:

These lines need to be added to the head...

```
"<link rel='stylesheet' href='codemirror.css'\>"
"<script src='codemirror.js'></script>"
"<script src='espeasy.js'></script>"
"<script src='anyword-hint.js' id='anyword'></script>"
```
rules.cpp:

one line needs to be added after the textfield...

`addHtml(F("<script>initCM();</script>"));`

**I would do a pull request but i wanted to discuss some questions before:**
1. Can we host the files in the static section on this repository with the other files?
2. Should the codemirror.css like the esp.css have an option on prioritizing a local file? (for personal styling)
3. Since the css includes a dark mode can we use the esp.css from my repository as the standard css in espeasy? 

[image]

TD-er, 2022-06-11:
I think for this we surely need to add a field somewhere to define the host where static files can be found.
I usually use the GitHub CDN servers, but this has the problem that you're always one version behind.
Also for development of these kind of features, it would be nice to have a local CDN to serve the files (or even another ESPEasy node)

chromoxdor, 2022-06-11:
That is an extremely good idea! 
But apart from this is there something that keeps us from implementing right now with files hosted on the github cdn?
And what do you think about making my "automode"esp.css  as a standard?

TD-er, 2022-06-11:
> And what do you think about making my "automode"esp.css as a standard?

That depends on the size of the file.
I've spent the last few days -again- a lot of time in reducing the bin size to make things fit again.

chromoxdor, 2022-06-11:
minified 7k

TD-er, 2022-06-11:
> minified 7k

And what was the one we have now?

chromoxdor, 2022-06-11:
5k

But what does it have to do with the bin size since it is served externally

TD-er, 2022-06-11:
On "normal" builds, it is included in the bin.

chromoxdor, 2022-06-11:
ohh.. i did not know. but it was not alway like that right?

TD-er, 2022-06-11:
Nope, it used to be included in all builds, but for builds with `LIMIT_BUILD_SIZE` set, it tries to load the file from an external CDN

chromoxdor, 2022-06-11:
OK...i understand. i am pretty sure, that i could make it smaller, but not by 2k. that´s al lot.

TD-er, 2022-06-11:
Well, let's first see where it might cause issues.

chromoxdor, 2022-06-11:
What do you mean?

TD-er, 2022-06-11:
We should start implementing it and if it may run into issues, we can maybe split the CSS into 2 parts so one part is excluded when we don't want to add the "dark mode" to save some space for example.
Or we can make the CSS always be loaded from an external source for those builds that will fail as a result of this change.
Another option can be to make the "dark mode" part always external, etc. Lots and lots of options.

Grovkillen, 2022-06-11:
Don't put too much effort into this because it's already part of the GUI
engine.

chromoxdor, 2022-06-11:
> Lots and lots of options.
You are right. Which makes the automode so convenient  :) 

And what about the highlighting? You want to wait until another hosting option or should we implement it now?

tonhuisman, 2022-06-11:
Best way to start is by creating that PR, then we will see where we are, size-wise 🏋️

chromoxdor, 2022-06-11:
creating a PR is definitely not one of my strong suits...
[image]

Grovkillen, 2022-06-11:
This is some showcase for the new GUI (now still missing the final API
standardization for saving settings etc)

https://grovkillen.dribbble.com/

closed 2022-09-22 by TD-er