letscontrolit/ESPEasy #4093: MQTT command syntax changed

chemmex, 2022-06-14:
Base config: C005 openHAB controller, device subscribe to the command topic (%sysname%/cmd) to receive commands from the broker. 

Command syntax: "event,cmd=1,2,3". In the rules, respectively, "On cmd do ..."
Until very now this syntax worked flawlessly, eventvalues were parsed as 1, 2 and 3

Using latest codebase (as of today), self-built binary. The above command is recognized by device, but eventvalues are treated as "12,3", with first comma cut out. 
Sending the same command (event,cmd=1,2,3) via http gives positive result.

Eventually, I found that with the latest build the command could be sent using "event,cmd,1,2,3" syntax. Looks great, but incompatible with installed devices.

chemmex, 2022-06-14:
Not sure about the date, my previous build was of 27 May, but I suspect I might not sync the code since beginning of May. Exactly when Platformio updated to 6.x.x

TD-er, 2022-06-14:
OK, will have a look at it tomorrow, as I was already planing to go to bed. (even before 2am... I'm improving my bed-time ritual ;) )

TD-er, 2022-06-15:
@chemmex Can you test this PR: https://github.com/letscontrolit/ESPEasy/pull/4094

chemmex, 2022-06-15:
This definitely works as it should. Thanks!

TD-er, 2022-06-15:
Great!
Thanks for reporting and testing.

PR #4094 (2022-06-15): [MQTT command] Fix unintended change of MQTT command syntax (#4093)
Problem was that the command was split on a comma and not the = char.

Fixes: #4093
closed 2022-06-15 by TD-er