letscontrolit/ESPEasy #4096: BME280 Sleep Mode

roschme, 2022-06-15:
Hello diligent guys,

is there a possibility for the "endusers" of ESPEasy to use the sleep mode of the BME280 (maybe via Rule, or together with the ESPEasy triggered deep sleep of a wemos D1)?
I take a look in the Plugin028 as well as in the PluginStructs but honestly i've no chance to understand this complexity ... 
 
Thank you very much
Bob

TD-er, 2022-06-15:
Do you have an example from the BME280 datasheet about what sleep mode you mean?

roschme, 2022-06-15:
I'm thinking of the ctrl_meas 0xF4 Register e.g.: BMP280_CONTROL_SETTING = 0x00 

https://www.bosch-sensortec.com/media/boschsensortec/downloads/datasheets/bst-bme280-ds002.pdf -> on Page 15, the jump via Mode[1:0]=00

TD-er, 2022-06-15:
This is called right after all values are read from the sensor:

https://github.com/letscontrolit/ESPEasy/blob/d3cb633eca17c93a7def42066259c4e593e615fa/src/src/PluginStructs/P028_data_struct.cpp#L128-L129

So it looks like the sensor is put to sleep right after it is done taking a reading.

In ESPEasy, the readings are triggered by calling `PLUGIN_READ`, which is called periodically with "Interval" nr of seconds (which is set per task as one of the last settings per task)
Another way to call this `PLUGIN_READ` is by calling the command `TaskRun` from rules.
However the BME280 has a filter set which needs the sensor to perform several readings to get the readings close to the actual values.
So you may need to call `TaskRun` a few times and then put the ESP into deep sleep.
Since the reading from the BME was already done, the sensor should be put to sleep before the ESP starts deep sleep.

roschme, 2022-06-16:
Hello TD-er,

thank you very much for your quick response and the great infos! Initially i was wondering about the current consumption of my breadboard circuit wich is above ~60 µA with the BME connected an less than ~10 µA without the BME (Wemos D1, BME280, 3 V, without LED's, LDO('s), USB Driver and ESPEasy in Deepsleep). Maybe its an issue with the BME breakout board -> i'll check it.

TD-er, 2022-06-16:
Don't forget most boards with I2C chips do include pull-up resistors.
Meaning if the ESP pulls down the pins on the I2C bus during sleep, it will still draw current through these pull-up resistors.

Also, I wonder how you're measuring this stand-by current.
The only way you can accurately measure these low currents is with a PowerProfiler II kit (or similar device)
A multimeter simply cannot reliably measure such currents.
You have to dynamically switch the shunt resistor based on the load, then amplify the voltage over this shunt and measure it.
There are pre-amp like circuits which can be used along with an oscilloscope (or a multimeter), but it does take knowledge and experience to use these and draw correct conclusions based on measurements done with it.
The PowerProfiler II kit (not the older version, as the version II can deliver upto 1A) is rather essential for these kind of low current analysis and also it doesn't require a lot of knowledge to get started with it.  (also it is rather affordable)

chemmex, 2022-06-17:
@TD-er could this be the cause of BME280 not responding to Taskrun command? If you may remember, it does not read from the sensor, only when polled by controller or re-enabled

TD-er, 2022-06-17:
Hmm, have to check the code.
As far as I remember, this specific plugin does start a read cycle on a call to PLUGIN_READ and then it does all reading steps time based. If a reading is complete (1.somewhat seconds later), it reschedules itself to call the PLUGIN_READ since that function is the only one to actually deliver a value to controllers and send out events.

So I guess what you're seeing is that it does not immediatly have a new value ready.

Example of **not working** in the rules:
```
TaskRun,1 // run the 1st task assuming it has a BMx280
LogEntry,"BME values: [bme#T]C [bme#H]% [bme#P] mbar"
```

Instead you should do something like this:
```
TaskRun,1 // run the 1st task assuming it has a BMx280
[...]

on bme do
  LogEntry,"BME values: [bme#T]C [bme#H]% [bme#P] mbar"
endon
```
Or even better, use the `%eventvalue1%` ... `%eventvalue3%` for the logentry.

chemmex, 2022-06-17:
No, that's not the case. TaskRun,BME doesn't generate the BME event

TD-er, 2022-06-17:
Any idea since when this no longer works?

chemmex, 2022-06-17:
To be honest, I've not seen this working before. When I tried reading the values with TaskRun command once last year, id did not work already

TD-er, 2022-06-17:
OK, then it is not broken by the last edit Ton did. :)

But this plugin does things slightly different from other plugins, so I will look into this as this way how it is done here is how I plan on doing on other plugins too which now contain blocking code.
For example reading from Modbus devices now is rather blocking (depending on the device) as you need to wait for an answer before continuing.
Like the SenseAir S8 (not the S11) takes 200 msec for a reply. 
And Eastron and AccuEnergy power meters also take quite long to reply severely limiting the number of devices you could address on a single unit.

chemmex, 2022-06-17:
Well, I'm not really sure about chronology of the changes, but all other weather sensors I tried respond to TaskRun properly ;)

TD-er, 2022-06-17:
OK, thanks, that will limit the search for this issue :)

N.B. next time you're observing something like this, I would not mind to see it reported in an issue as it would then already have been fixed in yesterdays build :)

chemmex, 2022-06-17:
@roschme I would check if your BME280 module has onboard LDO and level shifter, this apart from the pullup resistors also can drain some current from the VCC

TD-er, 2022-06-17:
Hmm it looks like the sensor not sleeping seems to be related to the same issue as reported by @chemmex

chemmex, 2022-06-17:
@TD-er well there is #3879 I mentioned this problem first

roschme, 2022-06-17:
> @roschme I would check if your BME280 module has onboard LDO and level shifter, this apart from the pullup resistors also can drain some current from the VCC

Hi chemmex, i've desoldered the LDO (like: https://github.com/TomMajor/SmartHome/tree/master/Info/Ruhestrom)

roschme, 2022-06-17:
[image]

level shifter could be the problem as you mentioned

TD-er, 2022-06-24:
I did fix the issue mentioned by @chemmex and maybe this also did affect the power consuption of the BME280 sensor.

@roschme can you check with one of the last (successful) builds from GitHub Actions to see if you notice any changes in power consumption?
For example this one: https://github.com/letscontrolit/ESPEasy/actions/runs/2550549920

tonhuisman, 2023-09-09:
This doesn't seem relevant anymore, so can probably be closed.

closed 2023-09-09 by TD-er