letscontrolit/ESPEasy #4111: Is there a RGB 7 segment possible

fly74, 2022-06-23:
Like in https://werner.rothschopf.net/202005_arduino_neopixel_display_en.htm described.

I want to make a 7 segment 8 digit display with the color depending on values.

tonhuisman, 2022-06-23:
This is already possible to do (mind you, I didn't say easy 😄) using the basic [NeoPixel plugin](https://espeasy.readthedocs.io/en/latest/Plugin/P038.html#p038-page), by using the `NeoPixelLine` command.
That would best be implemented by creating a rule or rules that can 'draw' each character on a virtual 'digit', where each digit starts at a certain (fixed) offset from the previous one. Of course it can be done in C++ code, but that would require settings for all those different display configurations and will probably add quite a bit of code, but I think that would fit in the `Neopixel` builds.

fly74, 2022-06-23:
What do we need in hardware, WS2812B 7segment modules?

tonhuisman, 2022-06-23:
Well, you have now closed this, but it is a nice idea for a plugin or extension to a plugin 😉 
It's not that I have no ESPEasy projects to work on (😛), but all these display/output related things I'm interested in.

fly74, 2022-06-23:
@tonhuisman Let's play with it when I have a fitting hardware :)

fly74, 2022-06-24:
I think I print it and we test it.

fly74, 2022-07-02:
Hi @tonhuisman I think we need an option max brightness like in the neopixelmatrix in NeoPixel (BusFX) [TESTING]. For example the "<html>
<body>

nfx,rainbow[,speed[,fadetime]] | Run a rainbow effect color loop effect using speed and fadetime timings (in milliseconds).
-- | --

</body>
</html>"

uses full brightness wich takes a lot of power with 300 leds and I brick an ESP :) without separate power supply, but with low brightness it would be possible. The option should be strong without overuling.

tonhuisman, 2022-07-02:
I'll see if that's doable, not entirely sure if it's technically possible though.

tonhuisman, 2022-07-02:
@fly74 can you please create a new issue for this request, so it can be tracked separately from this, not directly related, issue?
I am able to add this feature (already working on it 😃).

fly74, 2022-07-02:
@tonhuisman He it is https://github.com/letscontrolit/ESPEasy/issues/4121

tonhuisman, 2024-08-25:
@fly74 I have added the requested NeoPixel (7-segment) display in a new PR, please join in testing 😄

fly74, 2024-08-25:
But we need to solder something to have a test. :)

tonhuisman, 2024-08-25:
I have 2 strings of these: https://aliexpress.com/item/1005006350041884.html (15 cm wires)

fly74, 2024-08-26:
I'll use this density [Link](https://www.amazon.de/BTF-LIGHTING-adressierbar-Vollfarbiger-wasserdichte-DIY-Projekte/dp/B088KJPXVB) and I want to print 15 cm high digits. That means that a vertical segment is nearly 7 cm high what need 10 led per segment at 1,44led/cm density. Horizontal maybe 2 leds less. 4 vertical and 3 horiz. segments take 64 led for a digit + 2x4 led for dots makes 4 digit 268. As I wrote this, this is to much :). Enough would be 100led/m [Link](https://www.amazon.de/BTF-LIGHTING-adressierbar-Vollfarbiger-wasserdichte-DIY-Projekte/dp/B088FJLRFP) 60/m is too less. 

Let's discuss any further.

The config should be easier. Maybe select count of digits, leds per horizontal and vertical segments for small digits e.g.
Wiring should be fix, maybe first led upper/left around the clock, middle last...

tonhuisman, 2024-08-26:
IMHO, the number of LEDs per segment you're using is quite high, knowing that these NeoPixels can be 'quite bright', to use an understatement 😄 
If I want to increase the number of LEDs per segment, I will have to do a breaking change to the settings (no conversion, as we're still in beta, hadn't expect to use > 7 LEDs per segment) and will have to do some UI changes, as the current limit is 5 pixels per segment.
Might be better to use 100 or 72 LEDs/m strips, or better yet, get the wired strings I linked to (probably also available from Amazon) where you can determine exactly the LED-distances, and have some more freedom in placing the LEDs. Or as TD-er suggested to me, to design a pcb digit board with the desired size (10x15 cm euro-card?) and number of LEDs, and have that produced by JCLPCB, including the smd-mounted LEDs. That would look awesomely professional 😎 

As shown in the screenshots I made, the current maximum of LEDs that can be configured in a single display is 816, but I don't have a power supply available that can handle that... 🤔 

Mounting a diffuser in front of each segment or digit will help even out the light per segment, and prevent users being blinded when they look at the display from a really close distance.

TD-er, 2024-08-26:
Diffusor is quite easy, just 3D print a thin layer white filament in front of the LEDs and have the 'segment' surrounded with a white border.
Then add a 3D printed black border around it to block the light from spreading.

fly74, 2024-08-26:
I know like it work, I've don some diffusors before 😎
Like:
[image]

fly74, 2024-08-26:
Something like that [Link](https://www.printables.com/de/model/937971-7-segment-rgb-led-clock-using-neopixels) but not transparent and each digit separately to build a 8 digit e.g.

TD-er, 2024-08-26:
Or maybe just the 'segments' in a square or "U" shape and place them in a way similar to "infinity grid".
This way you can print a lot larger than the 180x180 mm of a Prusa mini :)

fly74, 2024-08-26:
> Or maybe just the 'segments' in a square or "U" shape and place them in a way similar to "infinity grid". This way you can print a lot larger than the 180x180 mm of a Prusa mini :)

That's the plan.

tonhuisman, 2024-08-26:
About increasing the number of pixels per segment:
There is a technical limit of 64 pixels per digit. Each digit-pixel map is stored in an `uint64_t` type and that's the largest unsigned numerical that's available (bits are read from that by shifting a bit from right to left across that long word). To change that limit will require a major rewrite of the used library, and large parts of the plugin code.
The pixels in the decimal point, and 6 pixels for overlapping corners when enabled, _also_ count into that 64 pixels.
The offsets before and after the digit(group) do _not_ count into that 64 pixels.

If we can stay within that 64 pixels per digit limit, then it's _doable_.

fly74, 2024-08-26:
64 was to much as recognize by myself. So if assume one segment is max. 10cm high and led density 0,6led/cm make 6 led per segment x7=42 + dot leds should be enough for the biggest.

Here a sample of a 7-seg.svg
[image]

I think I have to do the connectors and it would be fine, maybe @TD-er  's o and u shape.

fly74, 2024-08-26:
@tonhuisman can we use the same fonts as in 7-seg. plugin?

tonhuisman, 2024-08-26:
Well, currently I have settings-space for up to 7 pixels per segment (3 bits...), it just needs some UI changes so the preview will be ok. I'll do that after I push a bunch of other changes and fixes I did today 😉

tonhuisman, 2024-08-26:
> @tonhuisman can we use the same fonts as in 7-seg. plugin?

It uses the exact same fonts, even most of the commands, incl. `7dbin` and `7dtext` are the same, except that we can have max. 16 digits, instead of 8 😉

TD-er, 2024-08-26:
First few layers white PETG and then change filament to black PETG for the 'walls'
Then make some frame to connect them and have the wires output through that frame.
Those "connectors" can be only connecting the "triangle" parts as you can extend the "walls" to fill up the height difference of those "connectors".

I think it can be quite professional looking like that and since the segments can be printed diagonally you can have segments of roughly 1.4x 180 mm in length :)

fly74, 2024-08-26:
@TD-er I think we start with a full segment 15cm, and I will connect the segments like this:

[image]
[image]

and from the back the inner shape in black with the taped leds inside.

TD-er, 2024-08-26:
I would make the link a bit different as now it is harder to print with the flat side down.
You can't print those large surfaces with only 2 layers thickness on top.
Also if you start at the build plate you can even make the diffusion layer more 'compressed'

TD-er, 2024-08-26:
Ehh how big is your nozzle?
With 0.4mm nozzle I typically use 0.2 mm layer thickness and the first layer is often compressed to effectively 0.1 mm.

fly74, 2024-08-26:
Sorry I mean 2x0.2 as my diffusor above.

fly74, 2024-08-27:
Tried a testprint of 2 segments with transition, look good. Have to find some led to check the diffusor.

[image]
[image]

TD-er, 2024-08-27:
I think you need to add a black surrounding to it or else it will shine through the side of the segments.
You could make the "walls" a bit smaller so the white first layers are "overhanging" and then print the black "walls" only which you can then slide over it.
Or even make the connections between the segments for the cabling in this black.

fly74, 2024-08-27:
No, the whole should be white from outside. There will be a black shape which fits in from behind, holding the led and seal off the side light.

TD-er, 2024-08-27:
White on the inside will help the diffusion.

fly74, 2024-08-27:
Not bad without inner shape.

3 led for the big segment and 2 led for the smaller one is enough. Needed 6 layers 0.2 for 1.2mm diffusor.

[image]
[image]

fly74, 2024-08-27:
[image]

That's the inner shape, so enough from 3d printing.

fly74, 2024-08-27:
@tonhuisman What would the wiring be? I need a 4 digit with 2 dots as a clock t first :) "88:88" but the dots must be separately lit up to show temperatures as 18.7° e.g.

TD-er, 2024-08-28:
Just from a traditional 7-segment design, I would expect each "digit" to have like 8 segments of which the dot is one.
It does not really make sense to have the last digit with a dot, so maybe this one could be used as a ":" ?
Then you have like 3 dots right above each other.

I think from a wiring perspective it would make the most sense if you have the digit "rotated" and the "DP" next to the F/A segment
[image]

TD-er, 2024-08-28:
> Not bad without inner shape.
> 
> 3 led for the big segment and 2 led for the smaller one is enough. Needed 6 layers 0.2 for 1.2mm diffusor.
> 
> ![IMG_9164](https://private-user-images.githubusercontent.com/38754941/361950327-0b099b57-cf24-4c0c-9f4c-152d1259e0f5.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQ4MjQ3ODksIm5iZiI6MTcyNDgyNDQ4OSwicGF0aCI6Ii8zODc1NDk0MS8zNjE5NTAzMjctMGIwOTliNTctY2YyNC00YzBjLTlmNGMtMTUyZDEyNTllMGY1LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODI4VDA1NTQ0OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY4MGZlMjUyM2Q1ZjIwZmNiMjBlODIzNDdiYzZkNmUxMjFlM2YwOTAzZjU2Y2M2OWUyZjkwMDgwNWZkMzI4ZGImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.TkWVq3OVBtWFG7ECqBKitHiJt7Tjgqy-r3Hc898xBrI) ![IMG_9166](https://private-user-images.githubusercontent.com/38754941/361950334-73e94518-0557-4001-ab2b-595bd8e0b8f1.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQ4MjQ3ODksIm5iZiI6MTcyNDgyNDQ4OSwicGF0aCI6Ii8zODc1NDk0MS8zNjE5NTAzMzQtNzNlOTQ1MTgtMDU1Ny00MDAxLWFiMmItNTk1YmQ4ZTBiOGYxLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODI4VDA1NTQ0OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE1NGU4ZTAyZDc2M2NkZTZmYzU3MWQ4MTY3ODlhMTdmMzM1ZDI4ZmRjYjc4YzYxZDA2OTEzYzRmZWY0ODM4NjImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.VCEqkBZ55B1XEyziyo8LB7wQqrG1IbbLJO5OR4pGsZ8)

Maybe also check in full sunlight during the day?

TD-er, 2024-08-28:
Oh and just a cautionary advice...
White filament is abrasive to the nozzle and the PTFE tube inside the print head. (contains titanium oxide to make it white)
So you may want to print some other color inbetween after some larger white prints, to clean the print head.

fly74, 2024-08-28:
But there is no option to have different led counts for horizontal and vertical segments. As wrote in the sample design "3 led for the big segment and 2 led for the smaller one is enough". 3 don't fit in the small segments and 2 not enough in the big segments. And the dot maybe should be the last led because it's transition to the next digit, otherwise I think of placing the ESP in the middle under the colon and so this would be a starting point :/

tonhuisman, 2024-08-28:
> But there is no option to have different led counts for horizontal and vertical segments.

So you haven't used the UI yet? 😸 
You can independently configure the number of pixels for: horizontal (1..7) and vertical (1..7) segments, dot segment (1..7) , extra pixels (1..12) and offset-before (1..15)

fly74, 2024-08-28:
Was busy with 3d prints. Now I have a look into. :)

tonhuisman, 2024-08-28:
> have the digit "rotated" and the "DP" next to the F/A segment
> And maybe "F" and "G" swapped?

Technically it's possible to have the 8 segments (7 + dp) in _any_ order, and currently the g-segment can be split in a left and right sub-segment (but will always be lit as 1 segment).
I've implemented these wiring variants:
- start-segment a (default)
- start-segment b
- dp-segment between b and c-segments (default)
- dp-segment as last segment
- g-segment between e and f segments as a single segment (not split) (default)
- g-right-segment between b and c segments, g-left-segment between e and f segments

And then there's also the Segment Corners overlap option, where each horizontal segment gets an extra pixel at each end, that's lit up for either segment that 'touches' it, so f.e. if either the a _or_ b segment light up, that corner pixel in the right-top will be lit 🥂 

The pixel number-plan is now generated clock-wise, but I'm considering to add a counter-clock-wise option, though that's quite some code to add, as the pixel orders within each segment must then also be reversed 🤔

fly74, 2024-08-28:
@tonhuisman I used from https://github.com/letscontrolit/ESPEasy/actions/runs/10596288332 the ESP_Easy_mega_20240828_neopixel_ESP8266_4M1M but no neo(7seg) selectable, is it ESP32 only?

fly74, 2024-08-28:
off but still on ?

[image]

tonhuisman, 2024-08-28:
Check the 'Decimal dot pixels' field... 😃, that checkbox just enables the pixel numbering for the dot-pixels (segment h) to be assigned _after_ all other segments (a..g).
Settings above that separator line between 'Pixels in group' and 'Pixel-offset before' determine the 'shape' of the digits, below that line influences the numbering plan. Maybe I should replace that line with a label 'Numbering options:' 🤔

tonhuisman, 2024-09-01:
@fly74 I've fixed a couple of bugs, some quite nasty like the digit segments being projected in reverse order 😨, so a fresh build is cooking in [this GH Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/10655019321) 😃

fly74, 2024-09-01:
Sorry to Iceland at the moment [link](https://www.youtube.com/live/YAQzsB9ev9Q?feature=shared)

fly74, 2024-09-04:
@tonhuisman back again. Would this be a clock config ?:

[image]

And can be more pixel used for the dots?

tonhuisman, 2024-09-04:
A few remarks:
- Group 1 doesn't have any decimal dots? I'd keep all segments exactly equal
- Group 1 has Decimal dot last segment set, that'll cause a wiring nightmare once you set the Decimal dot pixels to 1
- Group 2 only has 2 pixels for height segments? Probably a config error?
- Group 2 has Starting pixel at Left-Top? I'd keep all digits exactly equal
- On this size configuration, having more than 1 dot pixel might be a bit overdone, for really large digits, like having 5 or more pixels per segment, having 3, 4, 5 or 7 dot pixels (that seems to group nicest) might be needed to make the dot 'present' enough. There is no separate brightness setting possible for individual segments or dots, though you can set a separate color per group or digit (via commands), or light the extra pixels (before and after the group) in a specific color (also via commands)

By keeping all digits exactly equal, they are interchangeable as modules, and you can even add extra digits later, if needed. And the colon segment I'd also treat like a module, so you can easily add more if needed.
You'll probably get confused quickly when trying to assemble differently wired digits, I would for sure, KISS is the motto here 😄 

Another reason to keep all modules equal is that, like P073, this plugin also supports scrolling text from right to left. And the underlying library by default 'reduces' dots by adding them to the digit to the left of the dot, though I can overrule that behavior, I have adopted that default for now (may fine-tune that later), so not having dot pixels on a part of your display would then look strange...

fly74, 2024-09-05:
So I'm getting confused a little. I will design the digits with small width, this means that the vertical segments are larger than the horizontal. I thought this is possible. The Display only needs the colon to display times like "13:59" and temperature like "12 °C".
The ESP will be in the middle under the colon, but I think the start of the wiring must be thru other segments to the starting point.

[image]

tonhuisman, 2024-09-05:
For Group 2 you still have only 2 pixels for Height, so that will look different from the left 2 digits (Group 1), where you have 3 pixels for Height, not sure if you should want that.
If you want to display temperatures with 1 decimal, the dot will be a (functionally) required element, but you don't have that anymore.

fly74, 2024-09-05:
Damn my fault, should be the same as group 1 :)

fly74, 2024-09-05:
> For temperatures, I don't think the C is required, but when having no decimals that's possible. You'll only get in trouble when temperature drops below -9

You're right, I use it already with ° only, but the last digit could do it.

tonhuisman, 2024-09-05:
Did a minor improvement to the plugin: disable automatically using the decimal dot for displaying periods if no decimal dot pixels have been configured (entire display). GH Actions run is [here](https://github.com/letscontrolit/ESPEasy/actions/runs/10726797621)

fly74, 2024-09-09:
@tonhuisman I think we need different brightness settings for horizonal and vertical segment to adjust the look perfectly.

tonhuisman, 2024-09-09:
> @tonhuisman I think we need different brightness settings for horizonal and vertical segment to adjust the look perfectly.

Which segments are too bright now? I'd expect the 3-pixel segments to be brighter than the 2-pixel segments, that wouldn't surprise me... or maybe it's related to the orientation-angle of the pixels that they look different?
Another solution could be to adjust the diffusor thickness for the too-bright segments.

For any of the neopixel plugins, the brightness setting is applicable to the entire strip, not to individual (groups of) pixels, though you _can_ play with the color values per pixel (that what the brightness value does, recalculate each color), but I haven't (yet...) implemented setting separate colors per segment, so that's not possible right now.

fly74, 2024-09-09:
It was just to be generic for other designs. My design is not finalized right now. I'm testing diffusor and sizes at the moment.

TD-er, 2024-09-09:
Are diffusers 3D printed?
Maybe you can experiment with printing them 45 degree rotated on the print bed, so the infill is in straight lines.
And then see if horizontal segments should be printed in the other direction compared to vertical segments to get the same infill pattern in all segments.

fly74, 2024-09-09:
All is 3d printed. The first I've recognized is the led must be at least 20mm distance from the diffuser to get not light spots from the led.

[4 more replies]

PR #5113 (2024-09-28): [P165] Add plugin Display - NeoPixel (7-segment)
Resolves #4111
Resolves #4894

Features:
- [P165] Add Display - NeoPixel (7-segment) plugin, for building (large) NeoPixel based 7-segment displays (like score-boards)
  - 1..4 groups
  - 1..4 digits per group
  - segment height 1..7 pixels (configured per group of digits)
  - segment width 1..7 pixels.
  - decimal point 0..7 pixels.
  - extra pixels after a group of digits 0..12 pixels (to f.e. use as a colon).
  - offset pixels before a group, 0..15 pixels, that can also be used from commands
  - Can show:
    - Manual content (text, user defi [... cut]