letscontrolit/ESPEasy #4112: [Enhancement]: extend substring functionality to allow variable length string operations

chemmex, 2022-06-25:
Current substring usage syntax allows operations only with fixed string length, i.e start and end position limits must be predetermined, otherwise an error is thrown. This makes impossible dealing with values or system variables of unknown length.
I suggest adding sort of wildcard syntax to allow handling of "tail" segment of the string, such as {substring:0:>:%eventvalue%}. Here I used '>' to indicate the end of the string, so the whole eventvalue of any length will be processed

TD-er, 2022-06-25:
I can also make it to match the Arduino implementation of subString.
If you use an end position which does not occur in the string, it will make it the string length.

chemmex, 2022-06-25:
I have tested whatever could be tested and found that in fact subString works the Arduino way, so I can at least log the string to the last character.
What is missing, though, is some more real string functionality: all current functions except ord work with numbers only. For my particular task I would be happy to have string comparison, if that's not possible - then converting all characters to numbers would be sufficient

TD-er, 2022-06-25:
I plan on adding more string operations, but I think I will add some prefix to the operator to make sure the rules parsing will not be too complex.

For example `$=`, `$!=` and `$>` etc.
But not really sure what a compare should do?
- Compare length?
- Compare alphabetical order?

But what is "alphabetical order" ?
- case sensitive?
- What about non-letters like # etc.?
- What goes first, numbers or letters?

These should then also support string variables to store as you probably want to compare some new string against a previous one from an earlier event.

chemmex, 2022-06-25:
From my point of view, string equality and length calculation are most demanded. Not sure about alphabetical order use case, do you mean some way of sorting?

TD-er, 2022-06-25:
Yep, in order to determine which string is "less than" or "greater than", you need to know in what order things should be.
Not like I was thinking of actually providing a sorting algorithm to be used in rules, but for the compares with strings to have a meaning, you need to know what order they would have if you would sort them.

tonhuisman, 2022-12-27:
This may be (partially) solved by #4423

chemmex, 2022-12-27:
Exactly, thanks

closed 2022-12-27 by chemmex