letscontrolit/ESPEasy #4114: osc support

xinglicai, 2022-06-27:
Hello, I am engaged in the interactive multimedia industry. I am using espeasy, but most of the interactive software in our industry supports the OSC protocol. I hope to add the OSC protocol support to the protocol. Thank you!

TD-er, 2022-06-27:
Do you have links describing the protocol?

xinglicai, 2022-06-27:
This is the library link of Arduino, https://github.com/hideakitai/ArduinoOSC Thank you for developing such a good software as espeasy, which is really the gospel of our art interaction industry.

xinglicai, 2022-06-27:
It is to add OSC protocol support to the controller. It is a special UDP protocol with address. If you can add support, it will be great.

TD-er, 2022-06-27:
Just quickly browsed through the library you posted.
I find it a bit strange to see there is a special version for Ethernet and one for WiFi.
Have to look into it why that may be.

There seems also be a client and a server instance.
I assume the ESPEasy support should be a client, right?
Do you also have a link to a piece of software running on Windows/Linux implementing the broker side?

TD-er, 2022-06-27:
Oh and to get an idea of how it is actually being used, can you give some examples of what data should be sent and how it should be formatted?

xinglicai, 2022-06-27:
1. human infrared sensor. When a person walks by, the interactive software receives the address /data, the message content is 1, and the port is 7000. The interactive software plays video. When there are multiple human infrared sensors, multiple videos can be triggered
2. rotary encoder: when the encoder is rotated, the encoder value passes through the address /data, and the message content is: encoder value, port 7000. The interactive software content is synchronized with the rotation to achieve interesting interactive effects
3. the screen of the wind sensor is dandelion. The wind value is transmitted to the interactive software through the OSC. The dandelion in the screen can fly away. The stronger the wind, the faster it flies
In short, as long as it is a state change and a real-time numerical change, it can be converted into a trigger condition for interaction, so as to create interesting interaction effects.

xinglicai, 2022-06-27:
The OSC has a Windows version debugging client. the name is osctool,I will send it later.

xinglicai, 2022-06-27:
osctools link:
https://pan.baidu.com/s/14nbnEgWceLMuQcbcLUvHcQ 
password:z414

TD-er, 2022-06-27:
Don't you have a link to the original tool download?

xinglicai, 2022-06-27:
https://opensoundcontrol.stanford.edu

TD-er, 2022-06-27:
I'm not seeing a download page on there.
only a (very extensive) list of implementations.

TD-er, 2022-06-27:
I only found this one: [OSC Tools](https://www.cristianvogel.com/neverenginelabs/product/osc-tools)
But that's a paid tool.

xinglicai, 2022-06-27:
Give me a email, I will send to you.

TD-er, 2022-06-27:
My point is that it does look like it isn't free software and I don't want to encourage users to copy non-free software.

xinglicai, 2022-06-27:
Sorry, there is no open source OSC debugging tool. This is an OSC debugging tool developed by a friend of mine. In addition, you can also use interactive tools such as openFrameworks, Processing, Max, Unity, Touch Designer to debug. Because of the relationship from China, I Uploading is inconvenient. Now upload the file to the Google network disk, please check it.
https://drive.google.com/file/d/18H_Qx7HXRw8Q-kIUMQaMY04L8TauEGIq/view?usp=sharing

TD-er, 2022-06-27:
I will see if the debugging tool is needed.
If it is needed, we need to find a way to obtain a license, either by buying it or if it is being made available for free for OpenSource development (no idea what terms apply to the license)
But I will not use pirated software.