letscontrolit/ESPEasy #4119: ESP-CAM

fly74, 2022-07-01:
Do we have a plugin for the ESP32-CAM combination to get maybe an actual jpeg from the camera as jpeg?

tonhuisman, 2022-07-01:
You have commented in #3272 , have you also tried one of the suggested projects to get an image from the camera?

The listed projects there are mostly based on an Espressif demo application, using the Espressif libraries, but I don't think these functions are available in ESPEasy, that's based on the Arduino libraries. Adding all that to ESPEasy would add quite significantly to the bin size, not leaving much room for other functionality.

I suggest to continue the conversation/discussion in the other issue, as that already has quite useful information (and close this one, as it is in fact a duplicate request).

fly74, 2022-07-01:
I think it makes no sense because the camera quality is too poor as I read.

closed 2022-07-01 by fly74