letscontrolit/ESPEasy #4121: NeoPixel (BusFX) - option max brightness

fly74, 2022-07-02:
@tonhuisman we need an option max brightness like in the neopixelmatrix in NeoPixel (BusFX) [TESTING]. :)

tonhuisman, 2022-07-02:
Please test and validate, once the Actions build is finished (I've tested using the NeoPixel build myself, but would like your confirmation).

fly74, 2022-07-02:
Sure, when it's ready. :)

fly74, 2022-07-02:
Maybe we can make a little calculator in the settings. Each WB2812B draws 60mA at full brightness. So can we show what power is needed when for example you have 300 leds with brightness 40% (value 102/255). Full power 300*.060mA=18A. 18A*5V=90W. brighness 102 = 36W. When you know what I mean. Otherwise we could have the powersupply as value and the leds counts gives the max brighness which can be selected.

tonhuisman, 2022-07-02:
That is helpful, but would be more appropriate to put in the documentation, and not spend quite precious .bin space on it. I can safely assume most users know how to use a calculator 😉

I'll see if I can write that down, and include it for all NeoPixel plugins. (a paragraph 'NeoPixel Power requirements').

PR #4120 (2022-07-09): [P128] Add Max brightness setting
Resolves: #4121 

- Add 'Max brightness' setting to protect the hardware and power supply (and user eyes)
- Update documentation

TODO:
- [x] Testing
- [x] Link to new issue
- [x] Validation by requester
- [x] Add power requirement calculation examples to documentation (added in PR #3990 )
closed 2022-07-09 by TD-er