letscontrolit/ESPEasy #4122: Where to activate EXT_RTC?

crazybanane, 2022-07-02:
Hello in Custom.H there ist the line:

`#define USE_EXT_RTC                // Support for external RTC clock modules like PCF8563/PCF8523/DS3231/DS1307 `

But when i Build and Upload theres no setting or function where i can set I2C ports or theres no information where i have to connect the DS3231.

So how to use the function?

And is there a way to send the time to other nodes?

crazybanane, 2022-07-02:
Actual things i changed:
SDA pin to 21 SCL pin to 22, set on Advanced settings to Ext_Rtc DS3231. Under log i got: 487001: ExtRTC: Cannot get time from external time source.....
DS3221 is set to actual Date and Time what i do wrong

Board is MH-ET Live miniKit esp32
Latest Build version

sobolkz, 2022-07-02:
> Actual things i changed: SDA pin to 21 SCL pin to 22, set on Advanced settings to Ext_Rtc DS3231. Under log i got: 487001: ExtRTC: Cannot get time from external time source..... DS3221 is set to actual Date and Time what i do wrong
> 
> Board is MH-ET Live miniKit esp32 Latest Build version

All working at my Wemos D32 pro. First of all - module should be listed correctly in I2C scanner. After it you can add  External RTC settings in Tools-Advanced (somethere at the top of page). RTC will be activated after reboot.

crazybanane, 2022-07-02:
OK i did it how you described it.
[image]
[image]
[image]

[image]
After this i rebooted looked into LOG and see Error.....
[image]
So whats wrong here.?

crazybanane, 2022-07-02:
Oh WOW it works.......
FAQ was very unclear i thought i have to set the ds3231 manualy with another ESP or Arduino........

EasyEsp requieres 1 time online connection with ntp enabled then it will work. Pleae ad to Wiki. Thanks

tonhuisman, 2022-07-02:
I'll update the documentation, there's a documentation update PR still open.
Oh, the Wiki is no longer the main documentation, we're on [ReadTheDocs](https://espeasy.readthedocs.io/en/latest/) these days. Easier to maintain, as it is edited and generated from the source repository.

PR #4097 (2022-07-14): Documentation updates 06 2022
Updates for some documentation pages, adjusted where needed, either by updated/changed software or outdated information.

- Add a missing screenshot for Tools/Advanced (Rules options)
- Update extension list for VSCode
- Remove Atom as a development environment, as it is no longer maintained
- Improve styling (contrast), 'changed', add RETIRED plugin category
- Update documentation page on writing documentation  (non-recursive 😛)
- Procedure for configuring a RTC chip

Resolves #4122
closed 2022-07-02 by crazybanane