letscontrolit/ESPEasy #4130: Serial Proxy plugin - value empty, does not work with Rules

ghtester, 2022-07-13:
I have an issue with Serial Proxy plugin - the value is not propagated to Value in Devices list.
The firmware is Custom, Vagrant compiled - 20220708, ESP 8266_4M1M.

[image]

The testfile, sent to ESP node through serial port, looks like:

/ZPA4ZE103.v30_012
F.F(000000)
0.0(C102591)
C.1(03294067)
0.1(118079)
1.8.1(0020434#kWh)
1.8.2(0000000*kWh)
1.8.3(0000000*kWh)
1.8.4(0000000*kWh)
1.8.0(0020435*kWh)
2.8.1(0000000#kWh)
2.8.2(0000000*kWh)
2.8.3(0000000*kWh)
2.8.4(0000000*kWh)
2.8.0(0000000*kWh)
0.9.2(17-06-01 08:58)
C.7.0(0019)
0.3.3(00250.000*i\kWh)
0.C.1(21-02-17 18:32)
0.2.0(0101)
0.2.1(PRE_ZE103_DE_30)
0.2.4(5555)
C.8.1(091359:07#h:min)
C.8.2(000000:02*h:min)
C.8.3(000000:01*h:min)
C.8.4(000000:01*h:min)
C.8.0(091359:11*h:min)
C.9.1(000000:02#h:min)
C.9.2(000000:00*h:min)
C.9.3(000000:00*h:min)
C.9.4(000000:00*h:min)
C.9.0(000000:02*h:min)
C.50(091359:21*h:min)
12.4(233.0*V)
11.4(000.757*A)
13.4(0.613*cos)
11.6(013.796*A)
1.6.0(03.080*kW)
0.6.0(013497:22*h:min)

 Despite the Serial Proxy Events are logged, in Rules there's no action generated.

382873631: EVENT: SerialProxy#Input=`C.1(03294067)`
382873684: EVENT: SerialProxy#Input=`0.1(118079)`
382873764: EVENT: SerialProxy#Input=`1.8.1(0020434#kWh)`
382873845: EVENT: SerialProxy#Input=`1.8.2(0000000*kWh)`
382873924: EVENT: SerialProxy#Input=`1.8.3(0000000*kWh)`
382874004: EVENT: SerialProxy#Input=`1.8.4(0000000*kWh)`
382874087: EVENT: SerialProxy#Input=`1.8.0(0020435*kWh)`
382874184: EVENT: SerialProxy#Input=`2.8.1(0000000#kWh)`
382874264: EVENT: SerialProxy#Input=`2.8.2(0000000*kWh)`
382874345: EVENT: SerialProxy#Input=`2.8.3(0000000*kWh)`
382874424: EVENT: SerialProxy#Input=`2.8.4(0000000*kWh)`

On SerialProxy#Input do
  Publish  %sysname%/status,OK!
  taskvalueset,8,1,0
Endon

Another issue is that logging of Serial Proxy events depends on serial port speed - at higher speeds only some events are logged, at lower speed all events are logged from the start but after several recorded events the logging is stopped completely (and continue after navigating away from the Tools - Log webpage and open it again). Tested in Mozilla & Edge web browsers and the issue is the same.

BTW. the Rules Sets can't be saved here from Edge - clicking the Save button does nothing. From Mozila it works fine.

tonhuisman, 2022-07-13:
Documentation (though not explicitly this example): [Rules matching events using wildcards](https://espeasy.readthedocs.io/en/latest/Rules/Rules.html#matching-events-using-wildcard) and [similarly here too](https://espeasy.readthedocs.io/en/latest/Rules/Rules.html#event-name-eventname-or-eventpar)
Explicitly this: `Event values can now also be strings, just make sure to use the wildcard when matching the event name in the rules.`
And there is an explicit Note in the examples, a bit lower on the page.

The values can currently only contain numeric values, so that's why it's not shown in the devices overview.

ghtester, 2022-07-14:
OK, thanks again for explanation and useful links (shame on me I did not find it), now understood... So only the issues with logging and saving Rule Set from MS Edge remains but it's another subject => this thread may be closed.

closed 2022-07-14 by ghtester