letscontrolit/ESPEasy #4133: minimal_core_302_ESP8266_1M_OTA_FHEM_HA missing the OTA

fmuntean, 2022-07-14:
I was trying to build a 1MB firmware and found that the OTA is actually disabled in these builds.
minimal_core_302_ESP8266_1M_OTA_FHEM_HA has the Firmware update button in the UI however that page does not return anything:
URI: /update
Method: GET
Arguments: 0

fmuntean, 2022-07-14:
also just tested: ESP_Easy_mega_20220616_normal_ESP8266_1M_VCC.bin and has the same issue.

TD-er, 2022-07-14:
It does check for the available free space on the "sketch" partition/area 
If this is not enough for the "2-step OTA" bin, then it will not allow for an OTA update as it would then destroy the file system area

fmuntean, 2022-07-14:
Hmm,
This is what is see in the status page:

Sketch Size: | 850 kB (24 kB free)
Max. OTA Sketch Size: | 599 kB (614384 bytes)
OTA possible: | false
OTA 2-step Needed: | true

what is the 1MB firmware that we can use with OTA support  from the last release ?

TD-er, 2022-07-14:
There isn't any for quite some time.
Maybe if you disable all controllers, it might just fit.

But I guess we must remove almost all web config stuff from the build to make it work for OTA again. So then you may have to copy a working settings file from another node to make it work on such a node.

But if the modules you're using have the flash chip accessible, I strongly suggest to replace those flash chips with 4M versions.
It really isn't hard to solder, as long as it is accessible.

TD-er, 2022-07-14:
Nope, 64k is too small.
SPIFFS does need 2 blocks (8k) free to work properly.
This leaves roughly 45k for 2 settings files and when writing you need to have all pages of a block to be marked as "unused" in order to erase the block again to write to it.
With 128k SPIFFS it is already stretching it, but 64k is really too small.
Maybe 96k might work, but then you probably cant changes settings often or else the file system will be too fragmented to save stuff and thus corrupt the settings file.

Trust me, it would be less time for me to travel over to your place, desolder the flash chips and solder new ones on it, compared to getting the code to work with such small file systems.
Maybe even travelling to China to pick up the flash chips myself and then drive over to your place.

fmuntean, 2022-07-14:
I have expected this. The flash replacement is the easy part. The hard part is to retrieve and place the modules back from the hard to reach locations.
I guess I will have to start that process if I want to upgrade their firmware.
Looks that the newer the core for the ESP8266 the more flash it requires. 
I have built 1MB OTA custom ESPEASY firmware up to the last year with no issues but the newer the platform the bigger the size with the same parameters.

closed 2022-07-14 by fmuntean

TD-er, 2022-07-15:
Yep.
I noticed Espressif also has ESP32-C3 modules which are "pin compatible" with the ESP-12F modules, so they could in theory be a replacement.
Support for the C3 platform is on my todo list anyway, but I wonder how many people will actually replace those modules.
I can imagine that if you have designed some PCB and enclosure, then it would make sense to switch to such modules for new builds, but would you replace the ESP modules if it were supported by ESPEasy?