letscontrolit/ESPEasy #4134: Let & Taskvalueset leading zero issue

ghtester, 2022-07-14:
I have a problem with string to float variable conversion.
In a rule I need to convert some strings to variables
The sources looks like:
000.018
013.796
0020438 

Unfortunately the **Let** or **Taskvalueset** commands convert the input string with leading zero(es) to zero (except there's just one leading zero followed by decimal point).
I am not sure if it's a bug or a feature but so far I am not able to find a simple solution to solve it as the number of zeroes is not always the same (and I would like to avoid using many **IF**s in the rule). Could anybody direct me the right way plese?

TD-er, 2022-07-14:
It is a feature.
I can change the parsing function to get a numerical with multiple leading zeroes.
However I'm not sure what side effects this may have.

Sort of trim function which does strip leading zeroes.

ghtester, 2022-07-14:
OK, thanks a lot, I'll give it a try.
I believe the current transformation is not correct and the proposed code update should not break anything but I understand we need to be careful.

ghtester, 2022-07-15:
I have updated the source code as outlined, recompiled, upgraded the node and quickly tested.
The good news is that now the leading zeroes seems to be eliminated properly so the Let and Taskvalueset commands transform correctly the values with more leading zeroes.
The bad news is that my rule stopped working because it looks now I can't get the substring which starts at position 0. 

For instance:
On SerialProxy#Input=* do
Let,1,{substring:0:1:%eventvalue%}    
now returns the Unknown token error:

2855147: EVENT: SerialProxy#Input=`1.8.1(0020439#kWh)`
2855154: ACT  : Let,1,{substring:0:1:1.8.1(0020439#kWh)}
2855159: Calculate: Unknown token
2855164: ACT  : Let,2,8
2855170: ACT  : Let,3,1

TD-er, 2022-07-15:
OK, I guess the code should also "peek" for the next char to only remove zeroes when there is a next zero.

Change the added code into this:

```c++
  while ((firstDec + 1) < bufLength && tBuf.charAt(firstDec) == '0' && tBuf.charAt(firstDec + 1) == '0') {
    ++firstDec;
  }
```

TD-er, 2022-07-15:
Ah I see, the check should take away leading zeroes, until the "next" char is a `.`
So more like this:

```c++
  while ((firstDec + 1) < bufLength && tBuf.charAt(firstDec) == '0' && tBuf.charAt(firstDec + 1) != '.') {
    ++firstDec;
  }
```

ghtester, 2022-07-16:
Thanks a lot again, recompiled, updated node, quickly tested and now everything looks OK! :-)

TD-er, 2022-07-16:
Please let me know how that works out.
I will then make it optional (with a flag on the Advanced page) to enable it.
Maybe the function should also accept floating point values starting with a `.` (and also `+.` or `-.`)

ghtester, 2022-07-16:
So far everything works fine on both nodes.

> Maybe the function should also accept floating point values starting with a `.` (and also `+.` or `-.`)
I can confirm that works too.
From my perspective this transformation is much better than previous and I would vote for making it as a new default.
Hopefully it should not break anything.

TD-er, 2022-07-19:
Please test this PR: https://github.com/letscontrolit/ESPEasy/pull/4146
Especially with matching times in rules.

I think this may also fix some unnoticed issues where `0123` may have been parsed as an octal value.
So this may have caused issues when matching times in rules with `08` or `09` in the time notation. (either in the minutes or hours)

ghtester, 2022-07-21:
Thanks a lot, the PR seems to be working fine on my test node.

PR #4146 (2022-07-23): [Rules] Strip leading zeroes parsing numericals (#4134)
This should strip any leading zeroes when extracting a numerical string part from a larger string.

Fixes: #4134
commit 082d52f2 (2022-07-23): [Rules] Strip leading zeroes (#4134)

Fixes: #4134
closed 2022-07-23 by TD-er