letscontrolit/ESPEasy #4149: MLX90614 not working after ESPEasy_mega-20200310

whatsupskip, 2022-07-19:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

I found the same behaviour as raised in this forum discussion: https://www.letscontrolit.com/forum/viewtopic.php?t=8209
Changing the lower I2C clock didn't change anything.

### Expected behavior

Current versions should show the "correct" temperature.

### Actual behavior

The only temperature reported was 1037.

### Steps to reproduce

1. Just using the default ESP_Easy_mega_20220616_normal_ESP8266_4M1M.bin
2. Power cycling didn't solve the problem.
3. It did work correctly in ESP_Easy_mega-20200310_normal_ESP8266_4M1M.bin

### System configuration

Hardware: Wemos D1 mini clone on USB power.

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

TD-er, 2022-07-20:
The title suggests it has been broken for over 2 years.
Was that just the previous build you used, or did you backtrack versions until it worked again?

whatsupskip, 2022-07-21:
Thanks for your help with this.

I have now just tested ESP_Easy_mega_20211224_normal_ESP8266_4M1M.bin and realised I did have the EMONCMS controller configured, but not enabled. Also I had a "-" character in the name.
After deleting the EMONcms controller, enabling Force Slow I2C speed and changing the "-" character to "_". It reported the correct temperature.

I now have it working correctly with ESP_Easy_mega_20220616_normal_ESP8266_4M1M, Force Slow I2C speed enabled and no controller configured or enabled. Without the Force Slow I2C speed enabled it does show the incorrect temperature (1037.5). A couple of times with Force Slow I2C speed enabled I did see the 1037.5 flash, but that could be a browser caching issue.

closed 2022-07-21 by whatsupskip

TD-er, 2022-07-21:
Maybe this sensor needs some tweaking of the "clock stretching" setting on the tools->Advanced page?