letscontrolit/ESPEasy #4159: ESP32S2 AP mode not working

fmuntean, 2022-07-24:
building and flashing the normal_esp32s2_4M316k firmware.
I am connecting to it using serial and sending the command to set the ap mode but  SSID does not show on my laptop or the phone.

see logs:
RESET: Resetting factory defaults... using default settings
RESET: Warm boot, reset count: 0
RESET: formatting...
RESET: formatting done...
186095 : Info   : SaveToFile: free stack: 7092
186113 : Info   : FILE : Saved config.dat offset: 8192 size: 820
186113 : Info   : SaveToFile: free stack after: 7092
186115 : Info   : SaveToFile: free stack: 7124
186161 : Info   : FILE : Saved config.dat offset: 0 size: 3004
186161 : Info   : SaveToFile: free stack after: 7124
186163 : Info   : SaveToFile: free stack: 6932
186227 : Info   : FILE : Saved security.dat offset: 0 size: 593
186227 : Info   : SaveToFile: free stack after: 6932
186230 : Info   : SaveToFile: free stack: 6676
186238 : Info   : FILE : Saved security.dat offset: 1024 size: 6
186239 : Info   : SaveToFile: free stack after: 6676
RESET: Successful, rebooting. (you might need to press the reset button if you've just flashed the firmware)
1872ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40026e29
=> 0x40026e29: esp_restart_noos_dig at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c:44 (discriminator 1)
SPIWP:0xee
mode:DOUT, clock div:1
load:0x3ffe6100,len:0x7c
load:0x4004c000,len:0x7fc
=> 0x4004c000: ?? ??:0
load:0x40050000,len:0x25c4
=> 0x40050000: ?? ??:0
SHA-256 comparison failed:
Calculated: 83ae2367affa2e8df1f79f3f57e25911a7fcc70f9804967996c7e2a6669a3cc3
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x4004c15c
=> 0x4004c15c: ?? ??:0
�U1163 : Info   : 

INIT : Booting version: HEAD_e2c98cf (ESP32 SDK v4.4.2)
1163 : Info   : INIT : Free RAM:176640
1165 : Info   : INIT : Soft Reboot #1 Last Action before Reboot: Intended Reboot: ResetFactory Last systime: 187 - Restart Reason: CPU0: Software reset digital core
1166 : Info   : FS   : Mounting...
1174 : Info   : FS   : Mount successful, used 141815 bytes of 290156
1191 : Info   : CRC  : SecuritySettings CRC   ...OK
1195 : Info   : INIT : I2C
1197 : Info   : INIT : SPI not enabled
1197 : Info   : Time set to 187.000 Time adjusted by 185803.00 msec. Wander: 0.000 msec/second Source: RTC at boot
1198 : Info   : Current Time Zone: STD time start: 1970-10-25 03:00:00 offset: 0 min
1200 : Info   : Local time: 1970-01-01 00:03:07
1201 : Info   : WiFi : Start network scan all channels
1244 : Info   : WiFi : STA Started
6362 : Info   : WiFi : Scan finished, found: 73
6368 : Error  : WIFI : No valid wifi settings
6469 : Info   : WIFI : Set WiFi to AP+STA
6474 : Info   : WiFi : AP Started
7048 : Info   : WiFi : AP Stopped
7049 : Info   : WiFi : AP Started
7052 : Info   : WIFI : AP Mode ssid will be ESP-Easy with address 192.168.4.1
7054 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
7157 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 7056 ms
7170 : Error  : WiFi : Scan not allowed, unprocessed WiFi events:  disconn
7270 : Info   : WIFI : Set WiFi to OFF
7488 : Info   : WiFi : Start network scan all channels
7497 : Info   : WiFi : STA Started
11815 : Info   : WiFi : Scan finished, found: 3
11917 : Info   : WIFI : Set WiFi to AP+STA
12494 : Info   : WIFI : AP Mode ssid will be ESP-Easy with address 192.168.4.1
12596 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 11 s
12610 : Error  : WiFi : Scan not allowed, unprocessed WiFi events:  disconn
12709 : Info   : WIFI : Set WiFi to OFF
12929 : Info   : WIFI : Arduino wifi status: WL_NO_SHIELD 255 ESPeasy internal wifi status: DISCONNECTED
13031 : Info   : WIFI : Set WiFi to AP
13042 : Info   : WiFi : AP Started
13614 : Info   : WiFi : AP Stopped
13615 : Info   : WiFi : AP Started
13618 : Info   : WIFI : AP Mode ssid will be ESP-Easy with address 192.168.4.1
13650 : Info   : INIT : Free RAM:120296
13706 : Info   : INFO : Plugins: 47 [Normal] (ESP32 SDK v4.4.2)
13711 : Info   : Webserver: start
13716 : Info   : OTA  : Arduino OTA enabled on port 3232
15061 : Info   : WD   : Uptime 0 ConnectFailures 0 FreeMem 104856 WiFiStatus WL_NO_SHIELD 255 ESPeasy internal wifi status: DISCONNECTED
45054 : Info   : WD   : Uptime 1 ConnectFailures 0 FreeMem 105068 WiFiStatus WL_NO_SHIELD 255 ESPeasy internal wifi status: DISCONNECTED
73292 : Info   : WiFi : Start network scan all channels
73294 : Info   : WiFi : STA Started
77813 : Info   : WiFi : Scan finished, found: 25
77917 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 1 m 17 s
77930 : Error  : WiFi : Scan not allowed, unprocessed WiFi events:  disconn
78030 : Info   : WIFI : Set WiFi to OFF
78250 : Info   : WIFI : Arduino wifi status: WL_NO_SHIELD 255 ESPeasy internal wifi status: DISCONNECTED
78253 : Info   : WiFi : Best AP candidate: Linksys 74:31:70:B6:AE:D2 Ch:9 (-48dBm) WPA2/PSK
78254 : Info   : WIFI : Connecting Linksys 74:31:70:B6:AE:D2 Ch:9 (-48dBm) WPA2/PSK attempt #1
78354 : Info   : WIFI : Set WiFi to STA
78365 : Info   : WiFi : STA Started
78586 : Info   : WD   : Uptime 1 ConnectFailures 0 FreeMem 105084 WiFiStatus WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
80532 : Info   : WiFi : STA Stopped
80644 : Info   : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2273 ms
80665 : Error  : WiFi : Scan not allowed, unprocessed WiFi events:  disconn
80765 : Info   : WIFI : Set WiFi to OFF
80985 : Info   : WIFI : Arduino wifi status: WL_NO_SHIELD 255 ESPeasy internal wifi status: DISCONNECTED
80987 : Info   : WiFi : Best AP candidate: Linksys 74:31:70:B6:AE:D2 Ch:9 (-48dBm) WPA2/PSK
80988 : Info   : WIFI : Connecting Linksys 74:31:70:B6:AE:D2 Ch:9 (-48dBm) WPA2/PSK attempt #2
81088 : Info   : WIFI : Set WiFi to STA
81098 : Info   : WiFi : STA Started
98360 : Info   : WIFI : Set WiFi to AP+STA
98363 : Info   : WiFi : AP Started
98935 : Info   : WiFi : AP Stopped
98936 : Info   : WiFi : AP Started
98939 : Info   : WIFI : AP Mode ssid will be ESP-Easy with address 192.168.4.1
105071 : Info   : WD   : Uptime 2 ConnectFailures 0 FreeMem 102384 WiFiStatus WL_DISCONNECTED 6 ESPeasy internal wifi status: 
DISCONNECTED
135076 : Info   : WD   : Uptime 2 ConnectFailures 0 FreeMem 102376 WiFiStatus WL_DISCONNECTED 6 ESPeasy internal wifi status: 
DISCONNECTED
165064 : Info   : WD   : Uptime 3 ConnectFailures 0 FreeMem 102376 WiFiStatus WL_DISCONNECTED 6 ESPeasy internal wifi status: 
DISCONNECTED
195070 : Info   : WD   : Uptime 3 ConnectFailures 0 FreeMem 102376 WiFiStatus WL_DISCONNECTED 6 ESPeasy internal wifi status: 
DISCONNECTED
225069 : Info   : WD   : Uptime 4 ConnectFailures 0 FreeMem 102376 WiFiStatus WL_DISCONNECTED 6 ESPeasy internal wifi status: 
DISCONNECTED
255066 : Info   : WD   : Uptime 4 ConnectFailures 0 FreeMem 102384 WiFiStatus WL_DISCONNECTED 6 ESPeasy internal wifi status: 
DISCONNECTED
285058 : Info   : WD   : Uptime 5 ConnectFailures 0 FreeMem 102384 WiFiStatus WL_DISCONNECTED 6 ESPeasy internal wifi status: 
DISCONNECTED
>wifiapmode

OK
315060 : Info   : WD   : Uptime 5 ConnectFailures 0 FreeMem 102384 WiFiStatus WL_DISCONNECTED 6 ESPeasy internal wifi status: 
DISCONNECTED
345056 : Info   : WD   : Uptime 6 ConnectFailures 0 FreeMem 107352 WiFiStatus WL_DISCONNECTED 6 ESPeasy internal wifi status: 
DISCONNECTED
375054 : Info   : WD   : Uptime 6 ConnectFailures 0 FreeMem 107352 WiFiStatus WL_DISCONNECTED 6 ESPeasy internal wifi status: 
DISCONNECTED
399041 : Info   : WIFI : Set WiFi to STA
399043 : Info   : WiFi : AP Stopped
405061 : Info   : WD   : Uptime 7 ConnectFailures 0 FreeMem 110076 WiFiStatus WL_DISCONNECTED 6 ESPeasy internal wifi status: 
DISCONNECTED
435062 : Info   : WD   : Uptime 7 ConnectFailures 0 FreeMem 112176 WiFiStatus WL_DISCONNECTED 6 ESPeasy internal wifi status: 
DISCONNECTED
465059 : Info   : WD   : Uptime 8 ConnectFailures 0 FreeMem 112176 WiFiStatus WL_DISCONNECTED 6 ESPeasy internal wifi status: 
DISCONNECTED

fmuntean, 2022-07-24:
also when settings the ssid and ssidkey is not connecting to the AP.
Another custom build I uploaded previously was connecting to the AP.

TD-er, 2022-07-24:
Is it related (or the same?) to this issue: https://github.com/letscontrolit/ESPEasy/issues/4118 ?

fmuntean, 2022-07-26:
does not seems to be related as on the build where the WiFi works the AP mode also works.

Jason2866, 2022-07-28:
Can you try to compile with this framework?
```
platform = https://github.com/Jason2866/platform-espressif32.git
```
replacing the current used platform for the esp32 series in platformio setup

TD-er, 2022-07-28:
@Jason2866 
What did you change/patch?

Jason2866, 2022-07-28:
@TD-er  Using latest IDF Arduino (Arduino only minor changes) to build the libs.
IDF has changed / updated the closed source wifi libs.

TD-er, 2022-07-28:
Ah yep, glad they are as predictable as always when it comes to changelogs.
No surprises there.

But I will for sure test it here too, as I also had some serious incompatibility issues with ESP-NOW with the last few ESP32 SDK versions.

fmuntean, 2022-07-28:
> platform = https://github.com/Jason2866/platform-espressif32.git

building this now. will let you know the outcome.

fmuntean, 2022-07-28:
The build worked and wifi seems to work so far.
for reference here is the config I used:
[env:mfd]
extends                   = esp32s2_common
board                     = esp32s2
lib_deps                  = ${esp32s2_common.lib_deps}
                            ServoESP32
platform = https://github.com/Jason2866/platform-espressif32.git

However the build is now rejected due to size:

Wrote 0x13cec0 bytes to file D:\git\ESPEasy\.pio\build\mfd/ESP_Easy_mega_20220728_mfd.factory.bin, ready to flash to offset 0x0
bin_elf_copy([".pio\build\mfd\ESP_Easy_mega_20220728_mfd.bin"], [".pio\build\mfd\ESP_Easy_mega_20220728_mfd.elf"])
 copy to:   build_output\debug\ESP_Easy_mega_20220728_mfd.elf
 file size: 1232576 > 1044464
 copy to:   build_output\reject\ESP_Easy_mega_20220728_mfd.bin
 JSON to:   build_output\json\ESP_Easy_mega_20220728_mfd.manifest.json
 file size: 1298112 > 1044464
 copy to:   build_output\reject\ESP_Easy_mega_20220728_mfd.factory.bin
 copy to:   build_output\debug\ESP_Easy_mega_20220728_mfd.env.txt
 Timestamp: 2022-07-28 18:46:03.935463

TD-er, 2022-07-29:
That's caused by how you name the build.
The size compare is done in my own script, which bases the max. size on the name of the bin file.
You need to include some hints to the partitioning in the PIO env like is done in the other envs.
e.g. 4M1M or what is used on ESP32, 4M314k (not 100% sure on that one)

fmuntean, 2022-07-31:
I see!

closed 2022-08-05 by fmuntean

Jason2866, 2022-08-06:
The framework is used now in Tasmota as core 2.0.4.1
```
platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.4.1/platform-espressif32-2.0.4.1.zip
```