letscontrolit/ESPEasy #4165: Fix plugins for ESP32 calling pinMode often

TD-er, 2022-07-29:
The Dallas plugins were reported to no longer work since ESP32 SDK 2.0.3.
Apparently the time needed to switch pin mode now takes 16 usec instead of 4 usec it was before.

Since the 1Wire protocol needs to toggle quite often between input and output mode on the same GPIO, this really makes a difference.

There are probably more plugins suffering from the same issues.

Plugins to test/check:
- [ ] Ultrasonic distance sensor using NewPing8266 package (P013 - HC-SR04/RCW-0001)
- [ ] DHT11/12/22 SONOFF2301/7021 (P005)
- [ ] RFID - PN532 (P017) the reset pin
- [ ] Ser2Net (P022) the reset pin
- [ ] P1Gateway (P044) the reset pin
- [ ] TTP229 Key Pad (P063) SDO pin
- [ ] 7DGT (P073) DIO pin
- [ ] SHT1X (P031) data pin

tonhuisman, 2022-07-29:
Tested 7DGT (P073) and that is working as intended 👍, without fixes or patches applied.

tonhuisman, 2022-07-29:
Tested ST77xx with a ST7789 320x240 display using an ESP32, and that's also working as expected.

I guess the display-related devices aren't very timing-critical.