letscontrolit/ESPEasy #4169: Possibilty to start distance measurement manual (HC-SR04)?

Ferunda, 2022-08-01:
Hello all,

I have an queasting regrading the distance measurement with the HC-SR04 modul.

Is there the possibility to start the measurement manual over an comand line input or somthing else if I need for example all 3 sec an actual measurement. 

I´m aware of the option with "interval" but I don´t want do decrease this permanent. I just need an shoter period of measurements in certain situations.

Thank you in advance for help!

tonhuisman, 2022-08-01:
You can use the `TaskRun,<TaskName>` command, either from the Tools page, using Rules, or via http (copy the url after submitting the command from Tools).

If you would need to run it for 60 seconds every 3 seconds, you could add this to Rules:
```
on ReadHCSR04 do
  LoopTimerSet,1,3,20 // start timer 1 every 3 seconds for 20 times = 60 seconds
endon

on Rules#Timer=1 do
  TaskRun,HCSR04
endon
```
Then to start a batch of readings, you just submit the command `event,ReadHCSR04` as mentioned above.

Ferunda, 2022-08-02:
Hello tonhuisman,

thank you very much for the help and the example with the rule!

It works perfect an was what I looked for.

Thank you for the support

closed 2022-08-02 by TD-er