letscontrolit/ESPEasy #4170: LoopTimerSet_ms won´t work as expected, if PN532 enabled

kuppe234, 2022-08-02:
Hello,
I have the "RFID - PN532" connected on the ESP32.
When I use "LoopTimerSet_ms" the loops are very different in its lengh. I use it to blink with the onboard LED (GPIO-2) in a rule.
The same applies to all GPIOs to which I have connected an LED.

Example:
```
On System#Boot do
  looptimerSet_ms,1,500,60
endon

On Rules#Timer=1 do
  GPIOtoggle,2
endon
```
When I switch off the plugin, "looptimerset" works as expected, the LED flashes steadily.
In a firmware I checked out and built on April 01 this year, this problem does not occur yet. What could be the cause?

**Build:** normal_ESP32_4M316k (ESP_Easy_mega_20220802_normal_ESP32_4M316k.bin)
**I2C: SDA:** GPIO-5, SCL: GPIO-4
**Plugins:** only PN532
**Wifi Status LED:** none
**GPIO boot states:** all on default

TD-er, 2022-08-02:
This is the cause: https://github.com/letscontrolit/ESPEasy/issues/4103

In short, some code does actually "wait" for a sensor to complete.
We're in the process to make this more asynchronous exactly for this reason you're describing (and others like WiFi connections getting dropped etc.)

kuppe234, 2022-08-03:
Thank you for the explanation.
But I think it can also be due to "IDF 4.4".
On 27.04.2022 there was a "Merge pull request #3845 from TD-er/build/ESP32_IDF4.4". Since then, the timers no longer work properly (tested with various git checkouts).

TD-er, 2022-08-05:
Can you give some screenshots of the timingstats page running on both builds?
And maybe even more important, can you test if it still persists in one of the last GitHub Actions test builds? (using a newer SDK version of ESP32)

kuppe234, 2022-08-10:
Here are 2 screenshots, one as of 20/04/2022 and one as of today. These are the identical configuration with the same hardware.

- Version from 20.04.2022: https://www.dsheine.de/screenshots/normal_ESP32_20220420.png
- Version from 10.08.2022: https://www.dsheine.de/screenshots/normal_ESP32_20220810.png

The RFID reading process is slow (1..5 seconds) in the current firmware, but there is no delay in the one from 20.04.

tonhuisman, 2022-08-10:
> Here are 2 screenshots,

You can drop the screenshots in the Github web editor, so we don't have to access an unknown external host to view that...

TD-er, 2022-08-10:
I just received the same sensor today, so I can see what's the difference among builds.

TD-er, 2022-08-11:
I have connected the PN532 and indeed it takes forever to get a reading.
Apparently the I2C performance has taken quite a hit in the latest ESP32 SDK version. See also: https://github.com/espressif/arduino-esp32/issues/6553

Jason2866, 2022-08-11:
The i2c driver is refactored in Arduino => 2.0.3.  It is now not amymore fault tolerant to bad (wrong) written i2c implementations. Maaany i2c libs are just garbage.

TD-er, 2022-08-11:
Thanks for the heads up.
I will have another look at the timings but it seems this PN532 is a bit of a lost cause regarding I2C compliance.

N.B. another thing I noticed is that it seems switching I2C frequencies is also not working anymore.??? (on ESP32)

Jason2866, 2022-08-11:
We gave up (i think it was 2 years ago) to get PN532 reliable running via i2c. Iirc clock stretching is a mess how it is done with the PN532. Andre Thomas burned a lot of time with... The i2c driver code was big and ugly and still no joy. So he decided to remove it completely. Luckily the device can be connected via serial too. So he wrote a driver for serial connect. Via serial it runs stable. Had a uptime for 6 month. Just ended because i did a update of Tasmota.

TD-er, 2022-08-11:
I now seem to have it running stable, but I will for sure keep it in mind when the next reported issue shows up for this plugin.

As a matter of fact, I will link this conversation in de code to remind me :)

closed 2022-08-12 by TD-er

kuppe234, 2022-08-12:
I have test the new mega branch. It worked as expected.
Here you have the actual Timings:
[image]

The repetition rate for reading is also the same as for the April-Branch:
```
703864 : Info   : PN532: New Tag: 1458541861 39
703889 : Info   : EVENT: RFID#Tag=1458541861
703907 : Info   : ACT  : publish,ESP_Easy/RFID/Uhrzeit,"2022-08-12 11:03:53"
703921 : Info   : ACT  : publish,ESP_Easy/RFID/Seriennummer,1458541861
703959 : Info   : ACT  : let,1,1
703965 : Info   : ACT  : publish,ESP_Easy/RFID/Zugang_gestattet: ,nein
703977 : Info   : ACT  : tone,3,300,600
703986 : Info   : ACT  : publish,ESP_Easy/RFID/Status,1
703998 : Info   : ACT  : TimerSet,5,10
704176 : Info   : PN532: Old Tag: 1458541861 40
704476 : Info   : PN532: Old Tag: 1458541861 41
704779 : Info   : PN532: Old Tag: 1458541861 42
705085 : Info   : PN532: Old Tag: 1458541861 43
705387 : Info   : PN532: Old Tag: 1458541861 44
705694 : Info   : PN532: Old Tag: 1458541861 45
705995 : Info   : PN532: Old Tag: 1458541861 46
706299 : Info   : PN532: Old Tag: 1458541861 47
706609 : Info   : PN532: Old Tag: 1458541861 48
706912 : Info   : PN532: Old Tag: 1458541861 49
707215 : Info   : PN532: Old Tag: 1458541861 50
707519 : Info   : PN532: Old Tag: 1458541861 51
707823 : Info   : PN532: Old Tag: 1458541861 52
708128 : Info   : PN532: Old Tag: 1458541861 53
708630 : Info   : RFID : Removed Tag
```

PR #4192 (2022-08-12): [ESP32 PN532] Fix reading PN532 on ESP32 (#4170)
Fixes: #4170

The PN532 needs some time between sending a command and reading the acknowledgement.
If this reading is started too soon, the PN532 may miss the request and things start to timeout.

Still todo:
- [x] Remove/disable logic analyzer debug code
- [x] Split sending and reading into 2 parts reducing amount of time spent in the PLUGIN_TEN_PER_SECOND call
- [x] Check if switching to low I2C speed still works on ESP32