letscontrolit/ESPEasy #4179: Incorrect render rotated text for P023_OLED

almirus, 2022-08-06:
Hi!
`Build:⋄
20116  - Mega32
System Libraries:⋄
ESP32 SDK v4.4.1
Git Build:⋄
HEAD_56d3225
Plugin Count:⋄
57 [Normal]
Build Time:⋄
Jun 16 2022 22:06:44
Binary Filename:⋄
ESP_Easy_mega_20220616_display_ESP32_4M316k
Build Platform:⋄
Linux-5.10.102.1-microsoft-standard-WSL2-x86_64-with-glibc2.29
Git HEAD:⋄
HEAD_56d3225
`
correct render
[image]
after Rotation set Rotated
[image]

my setting:
[image]

almirus, 2022-08-06:
How to render text in full screen? I want only one line text (big font size)

TD-er, 2022-08-06:
Then you should try the "Framed OLED" plugin, as that one has several scaled fonts built in.

tonhuisman, 2023-09-09:
Not sure if this is still relevant? If not it should be closed.

closed 2023-09-09 by almirus