letscontrolit/ESPEasy #4188: Display - OLED SSD1306/SH1106

BaGRoS, 2022-08-09:
Hi
Could you add functionality like
Continuous endless scroll
??
Thru all lines?
For many this should be better solution.

Many thanks
BaGRoS

TD-er, 2022-08-09:
Like those traditional LED ticker displays?

I'm affraid it might need 100% CPU load to do this continuously and that would not work on ESPEasy as it still needs to pay attention to other tasks (sensors), sending data and maintaining network connection.
So maybe I don't understand your request?

BaGRoS, 2022-08-10:
Yes, similar, but only one line
Now it's like
https://youtu.be/cShO3cM_jqM
Forward, backward, maybe only forward like for forex stocks display.
Also in this case box for choose direction.

TD-er, 2022-08-10:
But this is only 1 long line on 1 page, right?
What if you place the same "line" on several pages?

tonhuisman, 2022-08-10:
I have implemented a similar ticker-tape/horizontal scroll feature in [P073] - 7Dgt and [P131] - NeoMatrix, and used the library features of MD-Parola in [P104] - MAX7219 dot matrix displays.
Currently there is on-going work on [P036] - OLed SSD1306/SH1106 in #3997 so maybe after completing that, this horizontal scrolling can be added as an extra feature of that plugin...

BaGRoS, 2022-08-10:
> 

Not same line, all 12 lines one by one. In the attached video I used 4 lines.

BaGRoS, 2022-08-10:
> I have implemented a similar ticker-tape/horizontal scroll feature in [P073] - 7Dgt and [P131] - NeoMatrix, and used the library features of MD-Parola in [P104] - MAX7219 dot matrix displays. [...]

Fantastic information  :)

BaGRoS, 2022-08-10:
> Not same line, all 12 lines one by one. In the attached video I used 4 lines.

Not same line, all 12 lines one by one. In the attached video I used 4 lines.

BaGRoS, 2022-08-10:
> But this is only 1 long line on 1 page, right? What if you place the same "line" on several pages?

Not the same line, all 12 lines one by one. In the attached video I used 4 lines.

uwekaditz, 2023-07-07:
@tonhuisman Are there any current changes in P036 I need to know before I take a look to this issue?

tonhuisman, 2023-07-07:
> @tonhuisman Are there any current changes in P036 I need to know before I take a look to this issue?

You might want to check-out #4554 and #4722, quite a chance there will be some overlap.

uwekaditz, 2023-07-21:
I think I found a solution by modifying the P036 plug in (see attached video).

https://github.com/letscontrolit/ESPEasy/assets/55511593/3a9cac2c-8e0d-45ea-9f21-07dd42b4f057

Following restrictions and settings do apply (see aattached picture):
[image]

- the ticker function is an item of the `Scroll` setting
- only one line is displaying the ticker string
- all `line contents` are parsed and combined to this ticker string, the parsing happens only if the ticker starts (using the `Interval` setting)
- the gaps between the ticker items must be set by the line content (leading or tailing spaces)
- the starting alignment (left, center, right) depends on the setting of the first line `Alignment` (gloabal and local)
- the font is also taken from the first  line `Modify font`
- the scrolling speed depends on the length of the ticker string and the `Interval` setting (in the video it was 20seconds)
- the footer is automatically hidden
- while scrolling only a substring of the ticker string is send to the OLED controller using start and end indexes, modified in the TEN_PER_SECOND event 
- the timing of the TEN_PER_SECOND event is 15ms (avg) and 22ms (max) while scrolling, the load is approx. 30%

tonhuisman, 2023-07-21:
Looks nice 👍

tonhuisman, 2023-07-21:
For .bin size reasons you might want to wrap these changes in a `P036_FEATURE_TICKER_TAPE` (or similar) flag.
Looking forward to the PR 😉

uwekaditz, 2023-07-21:
I did it already, but I'm not sure what should be the base for this PR.
I checked out both PRs #4554 and #4722, merged them and did my changes.
Or would it be better to have the mega branch as base?

tonhuisman, 2023-07-21:
Just use the `mega` branch as your base, that makes it the easiest to merge

tonhuisman, 2023-10-01:
The PR with this feature has been recently merged, so this issue can be closed.

closed 2023-10-02 by TD-er