letscontrolit/ESPEasy #4194: ESP32&mega-20220809 AdaGFX helper issue

sobolkz, 2022-08-12:
Hi all!
Just checked latest mega-20220809 with Wemos D32 mini pro - it's look like an error in AdaGFX helper was expected.
Mega-20220809 was uploaded to clean Wemos D32 mini pro, it can work without issues due ILI9341 plugin was enabled.
After it board stopping to boot at ADAGFX helper initialization step:
```
INIT : Booting version: mega-20220809_57b0c0c (ESP32 SDK v4.4.3)
48 : Info   : INIT : Free RAM:267076
50 : Info   : INIT : Cold Boot - Restart Reason: CPU0: RTC Watch dog reset digital core and rtc module CPU1: for APP CPU, reseted by PRO CPU
51 : Info   : FS   : Mounting...
92 : Info   : FS   : Mount successful, used 151552 bytes of 8318976
175 : Info   : CRC  : SecuritySettings CRC   ...OK
198 : Info   : INIT : I2C
201 : Info   : INIT : SPI Init (without CS)
203 : Info   : ExtRTC: Read external time source: [1660235047](tel:1660235047)
204 : Info   : Time set to [1660235047](tel:1660235047).000
206 : Info   : Current Time Zone: STD time start: 2022-10-30 03:00:00 offset: 360 min
207 : Info   : Local time: 2022-08-11 22:24:07
209 : Info   : WiFi : Start network scan all channels
311 : Info   : WiFi : STA Started
6927 : Info   : WiFi : Scan finished, found: 10
6931 : Info   : WiFi : Best AP candidate: XXXXXXXXXXXXXXXXXXXX Ch:9 (-58dBm) WPA2/PSK
6933 : Info   : WIFI : Connecting XXXXXXXXXXXXXXXXXXXXX Ch:9 (-58dBm) WPA2/PSK attempt #0
6943 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
6967 : Info   : INIT : Free RAM:219656
7014 : Info   : ILI9341: Init start.
7015 : Info   : ILI9341: Init done, address: 0x3ffd3abc valid, commands: tft/tftcmd
7016 : Info   : AdaGFX_helper:
```
Web server wasn't yet enabled, therefore only full reset can help to boot board again (with disabled ILI9341). I'll try to check latest build with @tonhuisman  commit #3903

TD-er, 2022-08-12:
There is no watchdog reset?

TD-er, 2022-08-12:
Just looking through your logs and I wonder what the `tel:` does in the logs:
```
203 : Info   : ExtRTC: Read external time source: [1660235047](tel:1660235047)
204 : Info   : Time set to [1660235047](tel:1660235047).000
```

Is this some browser extention trying to create clickable links for phone nrs?

sobolkz, 2022-08-12:
> There is no watchdog reset?

Due to testing any watchdogs wasn't configured. Internal also wasn't affected - ": Info   : AdaGFX_helper:" was last message in log for a long time until board was reloaded.

> Just looking through your logs and I wonder what the `tel:` does in the logs:
>
> ```
> 203 : Info   : ExtRTC: Read external time source: [1660235047](tel:1660235047)
> 204 : Info   : Time set to [...]

As I understand it's RTC module native work. I can try to unplug it. Also - this log was copied from serial console, not wrom web log. Which extension can make it inside of serial connection?

TD-er, 2022-08-12:
The logs should only show something like this:
```
203 : Info   : ExtRTC: Read external time source: 1660235047
```
Not the brackets and (tel:....) part.

Given the fact it wasn't rebooting, that means it is stuck somewhere where it is still calling something which resets the watchdog timer.
So perhaps it is waiting for some pin to react or data to be read?
Ton told me he would have a look at it later this evening.

sobolkz, 2022-08-12:
> So perhaps it is waiting for some pin to react or data to be read? Ton told me he would have a look at it later this evening.

Yep, it's look like waiting or block code. But there is no such reactions inside of ILI9341 plugin, because there is no buttons on it's board. Touch plugin is different part of this boards, it can be used separately and it's disabled now.
Also as you can see - main ILI9341 plugin was successfully initialized before AdaGFX helper.
Therefore it can't be a waiting for pin reaction, as minimum for ILI9341.

TD-er, 2022-08-12:
If you have log level Debug enabled, then you might see a bit more, depending on where it halts.
In the initialize() function, there is some debug logs.
It is possible it might just dump enough extra into the log buffers so more logs will be flushed to give an idea of where it stops.

TD-er, 2022-08-12:
I think it may be halted here:
https://github.com/letscontrolit/ESPEasy/blob/77f7781f0b149b6b94b9693b8bbc3d080028eb6b/src/src/PluginStructs/P095_data_struct.cpp#L182

That's the first line where there's actual interaction with the display.

Do you have the backlight pin set?
If so, then it should put a PWM signal onto that pin.

I will also check on my board to see if analog write is still working as it should, as there have been quite a few issues lately with changed implementation of low-level GPIO code in the most recent SDK.

TD-er, 2022-08-12:
Looked a bit more into the code of `Adafruit_SPITFT::sendCommand`
It looks like this is a lot of calls to `pinWrite` etc. and other Arduino calls.
This is probably way too slow since the last SDK update.
So I guess Ton might need to adapt this code to use the new direct calls to GPIO pins, or maybe this library has been updated by Adafruit as it will probably cause more issues for others too.

It seems there is already some code in there wrapped in `#if defined(USE_FAST_PINIO)` checks.
So maybe it is as simple as defining this in PlatformIO envs.

sobolkz, 2022-08-12:
> Do you have the backlight pin set? If so, then it should put a PWM signal onto that pin.
> 
Nope, Backlight control isn't used by plugin (therefore PIN not configured in plugin). It's used through rules, without PWM, only On\Off functionality after booting and before sleeping.

TD-er, 2022-08-12:
You could set GPIO-2 to be as backlight pin, as that one is often connected to a LED. This may help to see what's wrong?
On the other hand, GPIO-2 is often "inverted" so if the PWM is set to "almost max" then you might not see anything.

sobolkz, 2022-08-12:
> You could set GPIO-2 to be as backlight pin, as that one is often connected to a LED. This may help to see what's wrong? On the other hand, GPIO-2 is often "inverted" so if the PWM is set to "almost max" then you might not see anything.

As I wrote before (at the forum) - it's Wemos TFT 2.4 shield. It's backlight control was linked to TFT connector at GPIO32 for Wemos D32 Pro and through pin header. Therefore I'm using rules like

> on System#Boot do
> GPIO32,1
> endon
and
> on System#Sleep do
> GPIO32,0
> GPIO2,0
> endon

GPIO2 also used in my project, but for power control of environment through MOSFET. This GPIO controlled by Hardware at boot and by rules at sleep.
As I understand it can't take impact to display, because it's controller are powered by 3.3V line separately.

TD-er, 2022-08-12:
It was merely meant as a suggestion for diagnostic reasons as we now have no clue where it halts.

sobolkz, 2022-08-12:
> It was merely meant as a suggestion for diagnostic reasons as we now have no clue where it halts.

I'll check situation with increased logging level as you proposed.

TD-er, 2022-08-12:
Which pins do you use?
Can you also show some info about  CPU revision and flash shown on the sysinfo page?

Do you have any pins set on the Hardware tab to anything other than "default" (e.g. pull-up resistors enabled)

sobolkz, 2022-08-12:
> Which pins do you use? Can you also show some info about CPU revision and flash shown on the sysinfo page?
> 
> Do you have any pins set on the Hardware tab to anything other than "default" (e.g. pull-up resistors enabled)

Hmmm it's very long story. because almost all free PINs was used. There is 2 HW Serials (one for PMSx003, another one for Senseair S8), 2 active I2C - one is BME280, another one is RTC module. There is third one I2C module planned- SCD41 (instead of S8) but it isn't used now due to P135 wasn't released yet. SPI for TFT screen and GPIO2 for power latching.
There is no external pull up or pull down resistors, only that used inside of modules.
All work's good with @tonhuisman custom release, which discussed in #4060 at 21 May. 
Now some previous builds also tested - @tonhuisman build for #4174 and same issue expected - board stopping boot at AdaGFX helper step. 
There is Debug dev serial log.
This is first boot and activation of TFT plugin afrer it:
> rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DOUT, clock div:2
> load:0x3fff0030,len:184
> load:0x40078000,len:12616
> load:0x40080400,len:2916
> entry 0x400805c4
> ▒U46 : Info   :
> 
> INIT : Booting version: mega-20220809_57b0c0c (ESP32 SDK v4.4.3)
> 47 : Info   : INIT : Free RAM:267076
> 49 : Info   : INIT : Soft Reboot #2 Last Action before Reboot: Intended Reboot: CommandReboot Last systime: 1660313699 - Restart Reason: CPU0: Software reset CPU CPU1: Software reset CPU
> 50 : Info   : FS   : Mounting...
> 90 : Info   : FS   : Mount successful, used 151552 bytes of 8318976
> 173 : Info   : CRC  : SecuritySettings CRC   ...OK
> 195 : Info   : INIT : I2C
> 198 : Info   : INIT : SPI Init (without CS)
> 210 : Info   : Time set to 1660313699.000
> 211 : Info   : Current Time Zone: STD time start: 2022-10-30 03:00:00 offset: 360 min
> 213 : Info   : Local time: 2022-08-12 20:14:59
> 226 : Info   : WiFi : Start network scan all channels
> 330 : Info   : WiFi : STA Started
> 7346 : Info   : WiFi : Scan finished, found: 10
> 7349 : Info   : WiFi : Best AP candidate: access_point8 88:DC:96:26:7F:0A Ch:9 (-58dBm) WPA2/PSK
> 7351 : Info   : WIFI : Connecting access_point8 88:DC:96:26:7F:0A Ch:9 (-58dBm) WPA2/PSK attempt #0
> 7358 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
> 7382 : Info   : INIT : Free RAM:219908
> 7963 : Info   : INFO : Plugins: 124 [Normal] [MAX ESP32] (ESP32 SDK v4.4.3)
> 7966 : Info   : EVENT: System#Wake
> 8061 : Info   : EVENT: System#BootMode=1,1,1,0
> 8077 : Info   : Webserver: start
> 8091 : Info   : OTA  : Arduino OTA enabled on port 3232
> 8103 : Info   : EVENT: System#Boot
> RuleDebug Processing:rules1.txt
>      flags CMI  parse output:
> 8128 : Info   : ACT  : timerSet,1,5
> 8133 : Info   : ACT  : GPIO,32,1
> 8138 : Info   : GPIO : port#32: set to 1
> 8243 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 10 ms
> 8259 : Error  : WiFi : Scan not allowed, unprocessed WiFi events:  disconn
> 8360 : Info   : WIFI : Set WiFi to OFF
> 8585 : Info   : WIFI : Arduino wifi status: WL_NO_SHIELD 255 ESPeasy internal wifi status: DISCONNECTED
> 8587 : Info   : WiFi : Best AP candidate: access_point8 88:DC:96:26:7F:0A Ch:9 (-58dBm) WPA2/PSK
> 8589 : Info   : WIFI : Connecting access_point8 88:DC:96:26:7F:0A Ch:9 (-58dBm) WPA2/PSK attempt #1
> 8690 : Info   : WIFI : Set WiFi to STA
> 8701 : Info   : WiFi : STA Started
> 9278 : Info   : EVENT: Time#Initialized
> 9304 : Info   : WIFI : Connected! AP: access_point8 (88:DC:96:26:7F:0A) Ch: 9 Duration: 708 ms
> 9309 : Info   : EVENT: WiFi#Disconnected
> 9324 : Info   : WIFI : DHCP IP: 192.168.3.3 (MAX-ESP32) GW: 192.168.3.1 SN: 255.255.255.0   duration: 22 ms
> 9336 : Info   : WIFI : Starting mDNS...
> 9338 : Info   : WIFI : mDNS started, with name: MAX-ESP32.local
> 9365 : Info   : EVENT: Clock#Time=Fri,20:15
> 9376 : Debug  : EVENT: Clock#Time=Fri,20:15 Processing time:11 milliSeconds
> 9381 : Info   : EVENT: WiFi#ChangedAccesspoint
> 9392 : Debug  : EVENT: WiFi#ChangedAccesspoint Processing time:11 milliSeconds
> 9395 : Info   : EVENT: WiFi#ChangedWiFichannel
> 9406 : Debug  : EVENT: WiFi#ChangedWiFichannel Processing time:11 milliSeconds
> 9410 : Info   : UDP : Start listening on port 8266
[... 157 lines cut ...]
> load:0x40080400,len:2916
> entry 0x400805c4
> ▒U47 : Info   :
> 
> INIT : Booting version: mega-20220809_57b0c0c (ESP32 SDK v4.4.3)
> 48 : Info   : INIT : Free RAM:267076
> 50 : Info   : INIT : Cold Boot - Restart Reason: CPU0: RTC Watch dog reset digital core and rtc module CPU1: for APP CPU, reseted by PRO CPU
> 51 : Info   : FS   : Mounting...
> 77 : Info   : FS   : Mount successful, used 151552 bytes of 8318976
> 139 : Info   : CRC  : SecuritySettings CRC   ...OK
> 156 : Info   : INIT : I2C
> 158 : Info   : INIT : SPI Init (without CS)
> 160 : Info   : WiFi : Start network scan all channels
> 271 : Info   : WiFi : STA Started
> 7488 : Info   : WiFi : Scan finished, found: 11
> 7491 : Info   : WiFi : Best AP candidate: access_point8 88:DC:96:26:7F:0A Ch:9 (-58dBm) WPA2/PSK
> 7493 : Info   : WIFI : Connecting access_point8 88:DC:96:26:7F:0A Ch:9 (-58dBm) WPA2/PSK attempt #0
> 7503 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
> 7527 : Info   : INIT : Free RAM:220116
> **7569 : Info   : ILI9341: Init start.**
> **7570 : Info   : ILI9341: Init done, address: 0x3ffd385c valid, commands: tft/tftcmd**
> **7571 : Info   : AdaGFX_helper**

As you can see - there is no any additional information while Debug dev was activated.
Also - RTC module wasn't connected, only NTP server was configured.

tonhuisman, 2022-08-12:
> > Which pins do you use? Can you also show some info about CPU revision and flash shown on the sysinfo page?

Can you also provide this info?
Pins are mostly on Hardware page (show entire page please, also the separate GPIO configuration settings), and for ILI9341 on its Device Configuration page, and CPU/flash stuff on the Tools/Info page (or More info button on Main page), sections 'ESP Board' and 'Storage'.

tonhuisman, 2022-08-12:
I've been testing this for a while (though I haven't got a display connected to *this* device, tried that and works just as fine with a 4MB ESP32), I got this log (at INFO level):
NB: *No code changes to ILI9341 or AdafruitGFX_helper*
```
INIT : Booting version: [**hidden to protect the innocent**] (ESP32 SDK v4.4.3)
60 : Info   : INIT : Free RAM:266208
62 : Info   : INIT : Soft Reboot #7 Last Action before Reboot: Background Task Last systime: 1660316244 - Restart Reason: CPU0: Software reset CPU CPU1: Software reset CPU
63 : Info   : FS   : Mounting...
86 : Info   : FS   : Mount successful, used 151552 bytes of 8318976
129 : Info   : CRC  : SecuritySettings CRC   ...OK
141 : Info   : INIT : I2C
143 : Info   : INIT : SPI Init (without CS)
155 : Info   : Time set to 1660316244.000
156 : Info   : Current Time Zone:  DST time start: 2022-03-27 02:00:00 offset: 120 min STD time start: 2022-10-30 03:00:00 offset: 60 min
159 : Info   : Local time: 2022-08-12 16:57:24
171 : Info   : WiFi : Start network scan all channels
288 : Info   : WiFi : STA Started
6904 : Info   : WiFi : Scan finished, found: 45
6910 : Info   : WiFi : Best AP candidate: [**ssid**] 6E:FF:7B:68:06:44 Ch:8 (-44dBm) WPA2/PSK
6912 : Info   : WIFI : Connecting [**ssid**] 6E:FF:7B:68:06:44 Ch:8 (-44dBm) WPA2/PSK attempt #0
6919 : Info   : IP   : Static IP : 192.168.1.103 GW: 192.168.1.1 SN: 255.255.255.0 DNS: 192.168.1.1
6929 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
6952 : Info   : INIT : Free RAM:214988
7004 : Info   : ILI9341: Init start.
7708 : Info   : ILI9341: Init done, address: 0x3ffd4d78 valid, commands: tft/tftcmd
7709 : Info   : AdaGFX_helper: TFT Init.
7770 : Info   : P095 Splash start
8327 : Info   : INFO : Plugins: 126 [Normal] [MAX ESP32] (ESP32 SDK v4.4.3)
8329 : Info   : EVENT: System#Wake
8363 : Info   : EVENT: System#BootMode=1,1,1,0
8368 : Info   : Webserver: start
8372 : Info   : OTA  : Arduino OTA enabled on port 3232
8393 : Info   : EVENT: System#Boot
8497 : Info   : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 18 ms
8517 : Error  : WiFi : Scan not allowed, unprocessed WiFi events:  disconn
8618 : Info   : WIFI : Set WiFi to OFF
8858 : Info   : WIFI : Arduino wifi status: WL_NO_SHIELD 255 ESPeasy internal wifi status: DISCONNECTED
8861 : Info   : WiFi : Best AP candidate: [**ssid**] 6E:FF:7B:68:06:44 Ch:8 (-44dBm) WPA2/PSK
8863 : Info   : WIFI : Connecting [**ssid**] 6E:FF:7B:68:06:44 Ch:8 (-44dBm) WPA2/PSK attempt #1
8964 : Info   : WIFI : Set WiFi to STA
8986 : Info   : WiFi : STA Started
9094 : Info   : IP   : Static IP : 192.168.1.103 GW: 192.168.1.1 SN: 255.255.255.0 DNS: 192.168.1.1
9115 : Info   : EVENT: Time#Initialized
9621 : Info   : EVENT: TaskInit#ILI9341=8,1
9625 : Info   : WIFI : Connected! AP: [**ssid**] (6E:FF:7B:68:06:44) Ch: 8 Duration: 754 ms
9627 : Info   : WIFI : Static IP: 192.168.1.103 (ESP-32-103) GW: 192.168.1.1 SN: 255.255.255.0   duration: 8 ms
9639 : Info   : WIFI : Starting mDNS...
9641 : Info   : WIFI : mDNS started, with name: ESP-32-103.local
9643 : Info   : EVENT: WiFi#Disconnected
9660 : Info   : EVENT: Clock#Time=Fri,16:57
9666 : Info   : EVENT: WiFi#ChangedAccesspoint
9668 : Info   : EVENT: WiFi#ChangedWiFichannel
9669 : Info   : EVENT: WiFi#Connected
9720 : Info   : UDP : Start listening on port 65500
9721 : Info   : firstLoopConnectionsEstablished
9730 : Info   : WD   : Uptime 0 ConnectFailures 0 FreeMem 187868 WiFiStatus WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init
9793 : Error  : MQTT : Intentional reconnect
13801 : Error  : MQTT : Broker C002 connection failed (1/0)
13867 : Info   : NTP  : NTP replied: delay 21 mSec Accuracy increased by 0.782 seconds
13870 : Info   : Time set to 1660316280.782 Time adjusted by 23067.76 msec. Wander: 0.000 msec/second Source: NTP
13872 : Info   : Local time: 2022-08-12 16:58:00
13875 : Info   : EVENT: Clock#Time=Fri,16:58
13881 : Info   : EVENT: Time#Set
16706 : Info   : P095 Splash finished.
37873 : Info   :  Webserver 2 Arguments: 0: 'index' length: 1 1: 'page' length: 1
39729 : Info   : WD   : Uptime 1 ConnectFailures 1 FreeMem 183148 WiFiStatus WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init
```

sobolkz, 2022-08-12:
Ok, there is information for @tonhuisman release (as fully working one):
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

sobolkz, 2022-08-12:
> Edit: Bonus question: How long are the wires to the display, and are they all of the same length (less than 2 cm. diff)?
It's native Wemos TFT cable used.
Whole device with you release (as you can see it's working :-)
here:
[image]

It's working with you release but isn't working with newest builds.
Also - you test release for SCD4X module was checked with another board, with Wemos D1 pro board. This one for Wemos D32 pro can't be used with same display issue.

sobolkz, 2022-08-12:
Bingo!
Some progress with investigation - If all devices was added without TFT screen (and, also without rules ) board crashes was reproduced at PMS7003 initialization:
 
> Rebooting...
> ets Jun  8 2016 00:22:57
> 
> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DOUT, clock div:2
> load:0x3fff0030,len:184
> load:0x40078000,len:12616
> load:0x40080400,len:2916
> entry 0x400805c4
> ▒U47 : Info   :
> 
> INIT : Booting version: mega-20220809_57b0c0c (ESP32 SDK v4.4.3)
> 48 : Info   : INIT : Free RAM:267076
> 50 : Info   : INIT : Soft Reboot #7 Last Action before Reboot: Const Interval: TIMER_MQTT Last systime: 1660318992 - Restart Reason: CPU0: Software reset CPU CPU1: Software reset CPU
> 51 : Info   : FS   : Mounting...
> 109 : Info   : FS   : Mount successful, used 151552 bytes of 8318976
> 209 : Info   : CRC  : SecuritySettings CRC   ...OK
> 236 : Info   : INIT : I2C
> 238 : Info   : INIT : SPI not enabled
> 239 : Info   : ExtRTC: Read external time source: 1660319031
> 241 : Info   : Time set to 1660319031.000
> 242 : Info   : Current Time Zone: STD time start: 2022-10-30 03:00:00 offset: 360 min
> 244 : Info   : Local time: 2022-08-12 21:43:51
> 246 : Info   : WiFi : Start network scan all channels
> 349 : Info   : WiFi : STA Started
> 7566 : Info   : WiFi : Scan finished, found: 11
> 7569 : Info   : WiFi : Best AP candidate: access_point8 88:DC:96:26:7F:0A Ch:9 (-56dBm) WPA2/PSK
> 7570 : Info   : WIFI : Connecting access_point8 88:DC:96:26:7F:0A Ch:9 (-56dBm) WPA2/PSK attempt #0
> 7577 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
> 7601 : Info   : INIT : Free RAM:219884
> 7654 : Info   : PMSx003 : using hardware serialGuru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
> 
> Core  1 register dump:
> PC      : 0x40093bbe  PS      : 0x00060f30  A0      : 0x80094212  A1      : 0x3ffb22e0
> A2      : 0xffffffff  A3      : 0x00000017  A4      : 0x3ffc7b64  A5      : 0x0000abab
> A6      : 0x00060f23  A7      : 0x00060f20  A8      : 0x00000002  A9      : 0xffffffff
> A10     : 0x00000002  A11     : 0x00000110  A12     : 0x00000017  A13     : 0x0000abab
> A14     : 0x00060f23  A15     : 0x00060f23  SAR     : 0x00000003  EXCCAUSE: 0x0000001c
> EXCVADDR: 0x0000001b  LBEG    : 0x4008c43d  LEND    : 0x4008c44d  LCOUNT  : 0xffffffff
> 
> 
> Backtrace:0x40093bbb:0x3ffb22e00x4009420f:0x3ffb2300 0x4009447c:0x3ffb2320 0x4008499a:0x3ffb2340 0x400849ad:0x3ffb2360 0x40084bc8:0x3ffb2380 0x401eb96d:0x3ffb23a0 0x401e5b9e:0x3ffb23e0 0x401e1d7a:0x3ffb2440 0x400e258a:0x3ffb2480 0x401b0453:0x3ffb24b0 0x40135df7:0x3ffb2510 0x4018b9f3:0x3ffb2620 0x4018bd0a:0x3ffb2670 0x40199096:0x3ffb2760 0x40189d39:0x3ffb27a0 0x40123f7f:0x3ffb2800 0x401e5ea6:0x3ffb2820
> 
> 
> 
> 
> ELF file SHA256: 0000000000000000
> 
> Rebooting...
> ets Jun  8 2016 00:22:57

Board was going to cycle reboot with same serial messages for some time and then going back with disabled PMS7003 plugin:
[image]
After it board was working good until display was added and board was rebooted after it:

> rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DOUT, clock div:2
> load:0x3fff0030,len:184
> load:0x40078000,len:12616
> load:0x40080400,len:2916
> entry 0x400805c4
> ▒U47 : Info   :
> 
> INIT : Booting version: mega-20220809_57b0c0c (ESP32 SDK v4.4.3)
> 48 : Info   : INIT : Free RAM:267076
> 50 : Info   : INIT : Soft Reboot #13 Last Action before Reboot: Intended Reboot: CommandReboot Last systime: 1660320187 - Restart Reason: CPU0: Software reset CPU CPU1: Software reset CPU
> 51 : Info   : FS   : Mounting...
> 67 : Info   : FS   : Mount successful, used 151552 bytes of 8318976
> 94 : Info   : CRC  : SecuritySettings CRC   ...OK
> 104 : Info   : INIT : I2C
> 107 : Info   : INIT : SPI Init (without CS)
> 109 : Info   : ExtRTC: Read external time source: 1660320187
> 110 : Info   : Time set to 1660320187.000
> 111 : Info   : Current Time Zone: STD time start: 2022-10-30 03:00:00 offset: 360 min
> 112 : Info   : Local time: 2022-08-12 22:03:07
> 114 : Info   : WiFi : Start network scan all channels
> 216 : Info   : WiFi : STA Started
> 7432 : Info   : WiFi : Scan finished, found: 9
> 7435 : Info   : WiFi : Best AP candidate: access_point8 88:DC:96:26:7F:0A Ch:9 (-58dBm) WPA2/PSK
> 7437 : Info   : WIFI : Connecting access_point8 88:DC:96:26:7F:0A Ch:9 (-58dBm) WPA2/PSK attempt #0
> 7444 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
> 7468 : Info   : INIT : Free RAM:219972
> 7501 : Info   : ILI9341: Init start.
> 7502 : Info   : ILI9341: Init done, address: 0x3ffd391c valid, commands: tft/tftcmd
> 7503 : Info   : AdaGFX_h
>

sobolkz, 2022-08-12:
P.S. also - WiFi isn't connected to AP, there is no such device in WiFi client's list at the router.

tonhuisman, 2022-08-12:
Thinking about this for a while, but I'm inclined to think that this is related to recent IDF changes, regarding pin mode settings etc.
Also, using pins 13 and 14 for CS and a serial connection pin might interfere with the regular HSPI pin designation (that's not initialized as you have selected VSPI), so it shouldn't matter, but if it does that *could* be a bug introduced by the latest IDF update, but that's just a wild guess.

TD-er, 2022-08-12:
I don't have the exact same display here (or at least I can't find it...)
But I tested with an ST7789 which does seem to be using the same Adafruit libraries.
I also used the VSPI bus and it is working just fine.
So I don't think it is related to the timing issues we saw for other plugins when accessing GPIO pins on recent ESP32 SDK builds.
[image]

[image]

Also looked at the signals using the Logic Analyzer, but there doesn't seem to be any strange jitter in the timings 
[image]
It is a very high frequency SPI clock by the way (roughly 26 MHz).
Not sure if this frequency has changed since older builds.

sobolkz, 2022-08-13:
OK, I'll try to check situation with new same board. It isn't soldered, therefore it can't take HW affect, only SW.

sobolkz, 2022-08-13:
So, same bootloop with new board, wasn't previously used yet. Was flashed with mega-20220809 and minimally configured - same Advanced settings, same Hardware and VSPI. GPIO2&32 was manually changed to 1 by "Commands" through web server. After Display was added it send Core panic, rebooted and stucked at AdaGFX step:

> ets Jul 29 2019 12:21:46
> 
> rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DOUT, clock div:2
> load:0x3fff0030,len:184
> load:0x40078000,len:12616
> load:0x40080400,len:2916
> entry 0x400805c4
> ▒U42 : Info   :
> 
> INIT : Booting version: mega-20220809_57b0c0c (ESP32 SDK v4.4.3)
> 43 : Info   : INIT : Free RAM:267396
> 45 : Info   : INIT : Cold Boot - Restart Reason: CP./components/esp_littlefs/src                                                            /littlefs/lfs.c:1229:error: Corrupted dir pair at {0x0, 0x1}
> U0: Vbat power on reset CPU1: for APP CPU, reseted by PRO CPU
> 46 : Info   : FS   : Mounting...
> FS   : Mount failed
> 54 : Error  : FS   : Mount failed
> RESET./components/esp_littlefs/src/littlefs/lfs.c:1229:error: Corrupted dir pair                                                             at {0x0, 0x1}
> : Resetting factory defaults... using default settings
> RESET: Warm boot, reset count: 0
> RESET: formatting...
> RESET: formatting done...
> 31170 : Info   : SaveToFile: free stack: 7416
> 33202 : Info   : FILE : Saved config.dat offset: 8192 size: 820
> 33203 : Info   : SaveToFile: free stack after: 7416
> 33204 : Info   : SaveToFile: free stack: 7448
> 35374 : Info   : FILE : Saved config.dat offset: 0 size: 3004
> 35375 : Info   : SaveToFile: free stack after: 7448
> 35377 : Info   : SaveToFile: free stack: 7256
> 35488 : Info   : FILE : Saved security.dat offset: 0 size: 593
> 35489 : Info   : SaveToFile: free stack after: 7256
> 35501 : Info   : SaveToFile: free stack: 7000
> 35587 : Info   : FILE : Saved security.dat offset: 1024 size: 6
> 35588 : Info   : SaveToFile: free stack after: 7000
> RESET: Successful, rebooting. (you might need to press the reset button if you've just flashed the firmware)
> 36771 : Info   : WIFI : Set WiFi to OFFets Jul 29 2019 12:21:46
> 
> rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DOUT, clock div:2
> load:0x3fff0030,len:184
> load:0x40078000,len:12616
> load:0x40080400,len:2916
> entry 0x400805c4
> ▒U47 : Info   :
> 
> INIT : Booting version: mega-20220809_57b0c0c (ESP32 SDK v4.4.3)
> 48 : Info   : INIT : Free RAM:267076
> 50 : Info   : INIT : Soft Reboot #1 Last Action before Reboot: Intended Reboot: ResetFactory Last systime: 36 - Restart Reason: CPU0: Software reset CPU CPU1: Software reset CPU
> 51 : Info   : FS   : Mounting...
> 89 : Info   : FS   : Mount successful, used 151552 bytes of 8318976
> 175 : Info   : CRC  : SecuritySettings CRC   ...OK
> 197 : Info   : INIT : I2C
> 199 : Info   : INIT : SPI not enabled
> 200 : Info   : Time set to 36.000 Time adjusted by 35800.00 msec. Wander: 0.000 msec/second Source: RTC at boot
> 202 : Info   : Current Time Zone: STD time start: 1970-10-25 03:00:00 offset: 0 min
> 203 : Info   : Local time: 1970-01-01 00:00:36
> 205 : Info   : WiFi : Start network scan all channels
> 305 : Info   : WiFi : STA Started
> 6720 : Info   : WiFi : Scan finished, found: 9
> 6723 : Error  : WIFI : No valid wifi settings
> 6825 : Info   : WIFI : Set WiFi to AP+STA
> 6828 : Info   : WiFi : AP Started
> 7390 : Info   : WiFi : AP Stopped
> 7391 : Info   : WiFi : AP Started
[... 159 lines cut ...]
> INIT : Booting version: mega-20220809_57b0c0c (ESP32 SDK v4.4.3)
> 48 : Info   : INIT : Free RAM:267076
> 50 : Info   : INIT : Soft Reboot #2 Last Action before Reboot: Background Task Last systime: 1660364444 - Restart Reason: CPU0: Software reset CPU CPU1: Software reset CPU
> 51 : Info   : FS   : Mounting...
> 114 : Info   : FS   : Mount successful, used 151552 bytes of 8318976
> 222 : Info   : CRC  : SecuritySettings CRC   ...OK
> 251 : Info   : INIT : I2C
> 254 : Info   : INIT : SPI Init (without CS)
> 256 : Info   : ExtRTC: Read external time source: 1660364449
> 257 : Info   : Time set to 1660364449.000
> 258 : Info   : Current Time Zone: STD time start: 2022-10-30 03:00:00 offset: 360 min
> 260 : Info   : Local time: 2022-08-13 10:20:49
> 262 : Info   : WiFi : Start network scan all channels
> 363 : Info   : WiFi : STA Started
> 6980 : Info   : WiFi : Scan finished, found: 13
> 6983 : Info   : WiFi : Best AP candidate: access_point8 88:DC:96:26:7F:0A Ch:9 (-56dBm) WPA2/PSK
> 6985 : Info   : WIFI : Connecting access_point8 88:DC:96:26:7F:0A Ch:9 (-56dBm) WPA2/PSK attempt #0
> 6995 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
> 7019 : Info   : INIT : Free RAM:219540
> 7072 : Info   : ILI9341: Init start.
> 7073 : Info   : ILI9341: Init done, address: 0x3ffd3ab4 valid, commands: tft/tftcmd
> 7074 : Info   : AdaGFX_helper

Test scheme is:
[image]

sobolkz, 2022-08-13:
By the way - if it can help - I can't use @tonhuisman 's May's config from test release at new build (by save&load), look's like a difference in filesystem and new build can't successfully load it - board going to cyclic bootloop after loading old config.

tonhuisman, 2022-08-13:
> It is a very high frequency SPI clock by the way (roughly 26 MHz).
> Not sure if this frequency has changed since older builds.

That 26 MHz is what I've seen in these display-related sources ever since I started working with them, so most likely that hasn't changed.

tonhuisman, 2022-08-13:
OK, so I finally re-played your *exact* use-case: Download the zip file from GH, extract ESP_Easy_mega_20220809_max_ESP32_16M8M_LittleFS.bin, and flash that on my unit.
After reboot I got this log (with ILI9341 disabled):
```
INIT : Booting version: mega-20220809_57b0c0c (ESP32 SDK v4.4.3)
60 : Info   : INIT : Free RAM:266500
62 : Info   : INIT : Soft Reboot #1 Last Action before Reboot: Background Task Last systime: 1660374874 - Restart Reason: CPU0: Software reset CPU CPU1: Software reset CPU
63 : Info   : FS   : Mounting...
102 : Info   : FS   : Mount successful, used 151552 bytes of 8318976
174 : Info   : CRC  : SecuritySettings CRC   ...OK
193 : Info   : INIT : I2C
195 : Info   : INIT : SPI Init (without CS)
207 : Info   : Time set to 1660374874.000
208 : Info   : Current Time Zone:  DST time start: 2022-03-27 02:00:00 offset: 120 min STD time start: 2022-10-30 03:00:00 offset: 60 min
211 : Info   : Local time: 2022-08-13 09:14:34
223 : Info   : WiFi : Start network scan all channels
340 : Info   : WiFi : STA StartedGuru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40092588  PS      : 0x00060933  A0      : 0x800941e0  A1      : 0x3ffb5d60
A2      : 0x05000000  A3      : 0xb33fffff  A4      : 0x0000cdcd  A5      : 0x0000abab  
A6      : 0x00060923  A7      : 0x00060920  A8      : 0x0000abab  A9      : 0xffffffff
A10     : 0x00001040  A11     : 0x00060923  A12     : 0x00060920  A13     : 0x3ffb6000  
A14     : 0x003fffff  A15     : 0x00000000  SAR     : 0x0000001b  EXCCAUSE: 0x0000001d
EXCVADDR: 0x05000000  LBEG    : 0x4008b570  LEND    : 0x4008b586  LCOUNT  : 0xffffffff  

Backtrace:0x40092585:0x3ffb5d600x400941dd:0x3ffb5da0 0x40094475:0x3ffb5dc0 0x4008499a:0x3ffb5de0 0x400849f0:0x3ffb5e00 0x40094861:0x3ffb5e20 0x40257126:0x3ffb5e40 0x40257111:0x3ffb5e60 0x400d744a:0x3ffb5e80 0x400d6866:0x3ffb5ea0 0x400d6b2c:0x3ffb5ff0

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0030,len:184
load:0x40078000,len:12616
load:0x40080400,len:2916
entry 0x400805c4
�U59 : Info   : 

INIT : Booting version: mega-20220809_57b0c0c (ESP32 SDK v4.4.3)
60 : Info   : INIT : Free RAM:266500
```
😱 Core panic!
I think to deduce from this that at least the max build used here isn't that well-baked as it should be...
That could also be an explanation for the crashes when loading another config file.

After that initial crash the unit *seems* to run as expected, and I was able to start the ILI9341 task without errors, (display still not connected on this hardware...) though I haven't tested much with it, and re-installed a newer local build (see below), to avoid any possible mishap in settings etc.

@sobolkz Can you please test this fresh build (flash the factory.bin file via Serial please to have the latest bootloader) to see if that sticks better on your board?
[ESP_Easy_mega_20220812_max_ESP32_16M8M_LittleFS.zip](https://github.com/letscontrolit/ESPEasy/files/9331024/ESP_Easy_mega_20220812_max_ESP32_16M8M_LittleFS.zip)

sobolkz, 2022-08-13:
> Can you please test this fresh build (flash the factory.bin file via Serial please to have the latest bootloader) to see if that sticks better on your board?

No result, same Core panic after PMS7003 activation:

> ets Jun  8 2016 00:22:57
> 
> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DOUT, clock div:2
> load:0x3fff0030,len:184
> load:0x40078000,len:12616
> load:0x40080400,len:2916
> entry 0x400805c4
> ▒U42 : Info   :
> 
> INIT : Booting version: ESP_Easy_mega_20220813_max_ESP32_16M8M_LittleFS, (Self built) feature/several-small-improvements-2022-08_5bd455d (ESP32 SDK v4.4../components/esp_littlefs/src/littlefs/lfs.c:1071:error: Corrupted dir pair at {0x0, 0x1}
> 3)
> 44 : Info   : INIT : Free RAM:266928
> 45 : Info   : INIT : Cold Boot - Restart Reason: CPU0: Vbat power on reset CPU1: for APP CPU, reseted by PRO CPU
> 46./components/esp_littlefs/src/littlefs/lfs.c:1071:error: Corrupted dir pair at {0x0, 0x1}
>  : Info   : FS   : Mounting...
> FS   : Mount failed
> 54 : Error  : FS   : Mount failed
> RESET: Resetting factory defaults... using default settings
> RESET: Warm boot, reset count: 0
> RESET: formatting...
> RESET: formatting done...
> 8015 : Info   : SaveToFile: free stack: 7416
> 9483 : Info   : FILE : Saved config.dat offset: 8192 size: 820
> 9484 : Info   : SaveToFile: free stack after: 7416
> 9486 : Info   : SaveToFile: free stack: 7448
> 11060 : Info   : FILE : Saved config.dat offset: 0 size: 3004
> 11061 : Info   : SaveToFile: free stack after: 7448
> 11063 : Info   : SaveToFile: free stack: 7256
> 11156 : Info   : FILE : Saved security.dat offset: 0 size: 593
> 11157 : Info   : SaveToFile: free stack after: 7256
> 11169 : Info   : SaveToFile: free stack: 7000
> 11237 : Info   : FILE : Saved security.dat offset: 1024 size: 6
> 11238 : Info   : SaveToFile: free stack after: 7000
> RESET: Successful, rebooting. (you might need to press the reset button if you've just flashed the firmware)
> 12419 : Info   : WIFI : Set WiFi to OFFets Jun  8 2016 00:22:57
> 
> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DOUT, clock div:2
> load:0x3fff0030,len:184
> load:0x40078000,len:12616
> load:0x40080400,len:2916
> entry 0x400805c4
> ▒U46 : Info   :
> 
> INIT : Booting version: ESP_Easy_mega_20220813_max_ESP32_16M8M_LittleFS, (Self built) feature/several-small-improvements-2022-08_5bd455d (ESP32 SDK v4.4.3)
> 48 : Info   : INIT : Free RAM:266608
> 49 : Info   : INIT : Soft Reboot #1 Last Action before Reboot: Intended Reboot: ResetFactory Last systime: 12 - Restart Reason: CPU0: Software reset CPU CPU1: Software reset CPU
> 50 : Info   : FS   : Mounting...
> 88 : Info   : FS   : Mount successful, used 151552 bytes of 8318976
> 174 : Info   : CRC  : SecuritySettings CRC   ...OK
> 195 : Info   : INIT : I2C
> 198 : Info   : INIT : SPI not enabled
> 199 : Info   : Time set to 12.000 Time adjusted by 11801.00 msec. Wander: 0.000 msec/second Source: RTC at boot
> 201 : Info   : Current Time Zone: STD time start: 1970-10-25 03:00:00 offset: 0 min
> 202 : Info   : Local time: 1970-01-01 00:00:12
> 204 : Info   : WiFi : Start network scan all channels
> 306 : Info   : WiFi : STA Started
> 6921 : Info   : WiFi : Scan finished, found: 11
> 6925 : Error  : WIFI : No valid wifi settings
> 7026 : Info   : WIFI : Set WiFi to AP+STA
> 7030 : Info   : WiFi : AP Started
[... 470 lines cut ...]
> 6956 : Info   : WiFi : Scan finished, found: 11
> 6960 : Info   : WiFi : Best AP candidate: access_point8 88:DC:96:26:7F:0A Ch:9 (-58dBm) WPA2/PSK
> 6961 : Info   : WIFI : Connecting access_point8 88:DC:96:26:7F:0A Ch:9 (-58dBm) WPA2/PSK attempt #0
> 6971 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
> 6995 : Info   : INIT : Free RAM:219256
> 7048 : Info   : PMSx003 : using hardware serialGuru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
> 
> Core  1 register dump:
> PC      : 0x40093c1a  PS      : 0x00060f30  A0      : 0x8009426e  A1      : 0x3ffb22e0
> A2      : 0xffffffff  A3      : 0x00000017  A4      : 0x3ffc7bbc  A5      : 0x0000abab
> A6      : 0x00060f23  A7      : 0x00060f20  A8      : 0x00000002  A9      : 0xffffffff
> A10     : 0x00000002  A11     : 0x00000110  A12     : 0x00000017  A13     : 0x0000abab
> A14     : 0x00060f23  A15     : 0x00060f23  SAR     : 0x00000003  EXCCAUSE: 0x0000001c
> EXCVADDR: 0x0000001b  LBEG    : 0x4008c43d  LEND    : 0x4008c44d  LCOUNT  : 0xffffffff
> 
> 
> Backtrace:0x40093c17:0x3ffb22e00x4009426b:0x3ffb2300 0x400944d8:0x3ffb2320 0x4008499a:0x3ffb2340 0x400849ad:0x3ffb2360 0x40084bc8:0x3ffb2380 0x401f2af1:0x3ffb23a0 0x401ec9be:0x3ffb23e0 0x401e8b0e:0x3ffb2440 0x400e2786:0x3ffb2480 0x401bd933:0x3ffb24b0 0x401384f7:0x3ffb2510 0x4019426f:0x3ffb2620 0x40194586:0x3ffb2670 0x401a5872:0x3ffb2760 0x4018e869:0x3ffb27a0 0x401265eb:0x3ffb2800 0x401eccc6:0x3ffb2820
> 
> 
> 
> 
> ELF file SHA256: 0000000000000000
> 
> Rebooting...
>

tonhuisman, 2022-08-13:
Do you still have the Senseair connected and configured as shown before? Because using HWSerial0 combined with Serial logging can't work as intended. Serial logging is always bound to HWSerial0 on ESP32 when enabled.

sobolkz, 2022-08-13:
Ah, OK. I'll change settings. Now I can't exclude second serial because P135 plugin doesn't included in release. But then it'll be added I'll change S8 to SCD41, which use I2C.

tonhuisman, 2022-08-13:
[P135] PR was last updated with the latest `mega` branch a few hours ago, download from this [Github Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/2851458721)

sobolkz, 2022-08-13:
> Because using HWSerial0 combined with Serial logging can't work as intended. Serial logging is always bound to HWSerial0 on ESP32 when enabled.

Trying to use HWSerial1 or HWSerial2 (HWSerial0 isn't used anymore) for PMS7003 without adding S8 cause same issue with Core panic:

> 06328 : Debug dev : sendcontent free: 184644 chunk size:1360
> 306339 : Debug dev : sendcontent free: 181280 chunk size:1360
> 306351 : Debug dev : sendcontent free: 184636 chunk size:1360
> 306362 : Debug dev : sendcontent free: 182904 chunk size:1360
> 306379 : Debug dev : sendcontent free: 184584 chunk size:1360
> 306392 : Info   :  Webserver 29 Arguments (First 20): 0: 'index' length: 1 1: 'p                                                                                               age' length: 1 2: 'TDNUM' length: 2 3: 'TDN' length: 7 4: 'TDE' length: 2 5: 'se                                                                                               rPort' length: 1 6: 'i2cuart_addr' length: 2 7: 'i2cuart_ch' length: 1 8: 'taskd                                                                                               evicepin1' length: 2 9: 'taskdevicepin2' length: 1 10: 'rstpin' length: 2 11: 'p                                                                                               wrpin' length: 2 12: 'p053_model' length: 1 13: 'p053_output' length: 1 14: 'p05                                                                                               3_events' length: 1 15: 'p053_delay_wake' length: 2 16: 'p053_oversample' length                                                                                               : 2 17: 'TDT' length: 2 18: 'TDVN1' length: 5 19: 'TDF1' length: 0
> 306399 : Debug dev : Write settings: TaskSettings index: 3
> 306401 : Info   : SaveToFile: free stack: 6792
> 307542 : Info   : FILE : Saved config.dat offset: 38912 size: 536
> 307544 : Info   : SaveToFile: free stack after: 6792
> 307545 : Debug dev : Read settings: TaskSettings index: 0
> 307590 : Debug dev : Read settings: TaskSettings index: 1
> 307620 : Debug dev : Read settings: TaskSettings index: 2
> 307649 : Debug dev : Read settings: TaskSettings index: 3
> 307693 : Debug dev : Write settings: TaskSettings index: 3
> 307695 : Info   : SaveToFile: free stack: 7096
> 308987 : Info   : FILE : Saved config.dat offset: 38912 size: 536
> 308989 : Info   : SaveToFile: free stack after: 7096
> 308990 : Debug dev : Read settings: TaskSettings index: 0
> 309003 : Debug dev : Read settings: TaskSettings index: 1
> 309013 : Debug dev : Read settings: TaskSettings index: 2
> 309024 : Debug dev : Read settings: TaskSettings index: 3
> 309034 : Info   : SaveToFile: free stack: 7176
> 310912 : Info   : FILE : Saved config.dat offset: 0 size: 3004
> 310913 : Info   : SaveToFile: free stack after: 7176
> 310914 : Debug dev : Write settings: ExtendedControllerCredentials index: 0
> 310920 : Info   : SaveToFile: free stack: 6728
> 311004 : Info   : FILE : Saved security.dat offset: 1024 size: 6
> 311005 : Info   : SaveToFile: free stack after: 6728
> 311007 : Debug dev : Read settings: TaskSettings index: 3
> 311028 : Debug  : PMSx003 : config 13 4 -1 -1
> 311030 : Info   : PMSx003 : using hardware serialGuru Meditation Error: Core  1                                                                                                panic'ed (LoadProhibited). Exception was unhandled.
> 
> Core  1 register dump:
> PC      : 0x40093c1a  PS      : 0x00060930  A0      : 0x8009426e  A1      : 0x3f                                                                                               fb2180
> A2      : 0xffffffff  A3      : 0x00000017  A4      : 0x3ffc7bbc  A5      : 0x00                                                                                               00abab
> A6      : 0x00060923  A7      : 0x00060920  A8      : 0x00000002  A9      : 0xff                                                                                               ffffff
> A10     : 0x00000002  A11     : 0x00000110  A12     : 0x00000017  A13     : 0x00                                                                                               00abab
> A14     : 0x00060923  A15     : 0x00060923  SAR     : 0x00000003  EXCCAUSE: 0x00                                                                                               00001c
> EXCVADDR: 0x0000001b  LBEG    : 0x4008c43d  LEND    : 0x4008c44d  LCOUNT  : 0xff                                                                                               ffffff
> 
> 
> Backtrace:0x40093c17:0x3ffb21800x4009426b:0x3ffb21a0 0x400944d8:0x3ffb21c0 0x400                                                                                               8499a:0x3ffb21e0 0x400849ad:0x3ffb2200 0x40084bc8:0x3ffb2220 0x401f2af1:0x3ffb22                                                                                               40 0x401ec9be:0x3ffb2280 0x401e8b0e:0x3ffb22e0 0x400e2786:0x3ffb2320 0x401bd933:                                                                                               0x3ffb2350 0x401384f7:0x3ffb23b0 0x40194784:0x3ffb24c0 0x401e45ff:0x3ffb25b0 0x4                                                                                               02758d5:0x3ffb2660 0x400d80a6:0x3ffb2680 0x400dcc7d:0x3ffb26a0 0x400dcd2d:0x3ffb                                                                                               26e0 0x400dce35:0x3ffb2740 0x4018e0ce:0x3ffb2790 0x4018e385:0x3ffb [... cut ...]                                                                                 0x3ffb24b0 0x401384f7:0x3ffb2510 0x4019426f:0x3ffb2620 0x40194586:0x3ffb2670 0x4                                                                                               01a5872:0x3ffb2760 0x4018e869:0x3ffb27a0 0x401265eb:0x3ffb2800 0x401eccc6:0x3ffb                                                                                               2820
> 
> 
> 
> 
> ELF file SHA256: 0000000000000000
> 
> Rebooting...

tonhuisman, 2022-08-13:
You have tried a hardware reset, or better still, a power cycle (disconnect power/wait 30 sec./reconnect)?

tonhuisman, 2022-08-13:
Oh, and to avoid having to wipe and reflash the unit, it's much easier to use the `TaskEnable,<taskname>|<tasknr>` command 😉 after a crash the device will be disabled again...

tonhuisman, 2022-08-13:
Are both your boards equipped with the same CPU model & revision, `Model: ESP32-D0WDQ5` `Revision: 1` ?
Because I've been testing on a `Model: ESP32-D0WDQ5-V3` `Revision: 3` (16MB flash), and a `Model: ESP32-D0WDQ6` `Revision: 1` (4MB flash + ILI9341 connected).
So it can still be a matter of silicon instead of software... even though that's what the SDK/IDF should take care of.

sobolkz, 2022-08-13:
> Are both your boards equipped with the same CPU model & revision, `Model: ESP32-D0WDQ5` `Revision: 1` ? Because I've been testing on a `Model: ESP32-D0WDQ5-V3` `Revision: 3` (16MB flash), and a [...]

I'm testing now with only old one, because it's look like board HW independent issue.
So, latest results - you build 3227 seems to work with 4M316k and 16M1M versions - I can upload this versions and use with backup. Version 16M8M LittleFS still can't be used - I can't restore backup. Board going to cyclic reboot after restoring, hardware power disconnection doesn't help. Also board going to reboot after PMS7003 was added at HWSerial1 or HWSerial2.
I'll upload 16M1M version and will test it more detail.

TD-er, 2022-08-13:
So looks like there might be some timing issue related to larger file systems, like on 8MB LittleFS.
```
7739 : Info : PMSx003 : using hardware serialGuru Meditation Error: Core 1 pa nic'ed (LoadProhibited). Exception was unhandled.
```

This is caused by one of the following:
- Memory not allocated (or no longer allocated), but dereferenced
- Delete of non-initialized pointer
- Reading of flash string with function not compatible with flash strings. (should be using functions with `_P` in the name)
- Calling `delay()` in ISR functions. (but that one may be a different error now)

sobolkz, 2022-08-13:
So, 16M8M flash was used in limited boards. All of this boards should use LittleFS because basic filesystem  has problems with throughput in large spaces. Low throughput is equal with large timings and delays.  Therefore LittleFS versions should work more efficiently. But really this version can't work, It's look like some mistakes was merged between May and presented releases.
As I see - there is some applicable commits. We can check any new addition from newest to oldest and try to find latest, where LittleFS works. And understand whiere this bug was added.
It isn't so difficult, because can be expected only while any active  serial device added in devices list.

TD-er, 2022-08-13:
I think the update of the ESP32 SDK may be the most likely candidate as we've seen quite a lot of issues lately with the latest changes in the SDK.

One of the other changes I recently made, mainly for the ESP32 with larger LittleFS filesystems is that I added some caching so we need far less calls to the filesystem.
Not saying there can't be any bug in there, but on the other hand, it should lessen the load on the filesystem.

tonhuisman, 2022-08-13:
These is reported an issue elsewhere related to the flash_mode used for flash-access.
@sobolkz This is a build using `qio` flash mode, can you test this one please?
[ESP_Easy_mega_20220813_max_ESP32_16M8M_LittleFS.zip](https://github.com/letscontrolit/ESPEasy/files/9332108/ESP_Easy_mega_20220813_max_ESP32_16M8M_LittleFS.zip)

An OTA update should be enough, no need to wipe and serial-flash. (as far as currently known)

TD-er, 2022-08-13:
@Jason2866 Can you make a new release of the ESP32 SDK, including your latest commits, as suggested [here](https://github.com/esphome/esp-web-tools/issues/278#issuecomment-1210669894) ?
I have a very strong feeling we're running into issues here related to LittleFS and DOUT/DIO mode.

Also you seem to have tackled the flashing issues where the ESP would not be set into flash mode without pulling GPIO-0 down.

sobolkz, 2022-08-14:
> @sobolkz This is a build using `qio` flash mode, can you test this one please?
> [ESP_Easy_mega_20220813_max_ESP32_16M8M_LittleFS.zip](https://github.com/letscontrolit/ESPEasy/files/9332194/ESP_Easy_mega_20220813_max_ESP32_16M8M_LittleFS.zip)

No success, bootloop after uploading, neither with erasing nor with flashing without erase. there is serial output:

- rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
- configsip: 0, SPIWP:0xee
- clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
- mode:QIO, clock div:2
- load:0x3fff0030,len:344
- load:0xffffffff,len:-1
- ets Jul 29 2019 12:21:46

tonhuisman, 2022-08-14:
And you have flashed the `factory.bin` file, at address 0x0000, via serial again, I assume?

sobolkz, 2022-08-14:
Yes, sure. Download tool v. 3.9.2. As all before. Also, 16m1m was successfully flashed after it.

TD-er, 2022-08-14:
What state has the checkbox to change the bin? (not sure about the exact text)
Does it allow to change the bin during flash or not?
What does the flash tool show about the detected module? (flash frequency, flash mode, etc.)

Jason2866, 2022-08-14:
> @Jason2866 Can you make a new release of the ESP32 SDK, including your latest commits, as suggested [here](https://github.com/esphome/esp-web-tools/issues/278#issuecomment-1210669894) ? I have a [...]

I dont think it will solve your issues. The SDK (Arduino) is the same. Changed is only that esptool v4.1 and the USB reset command can be used.
You can try already with
```
platform = https://github.com/Jason2866/platform-espressif32.git
```

TD-er, 2022-08-14:
OK, will try.

Do you have any idea whether issues like these are really related to the slower speed of LittleFS on larger file systems?
I did try to flash nodes using QIO, but esptool kept changing it to DIO mode (also on configs where I set it to DOUT).
So on those nodes I cannot even switch to QIO even if I want to.

Jason2866, 2022-08-14:
Dont think LittleFS issues are speed related, we use heavily for Tasmota and our standard flash mode is `dout`. But when there is a mismatch in flashed bootloader and firmware i always had strange issues with LittleFS.
Are the issues when flashed with platformio and/or with factory firmware?

Jason2866, 2022-08-14:
Added [https://github.com/arendst/Tasmota/blob/development/pio-tools/post_esp32.py#L134-L142](https://github.com/arendst/Tasmota/blob/development/pio-tools/post_esp32.py#L134-L142) in factory firmware build script since you can not flash bootloaders in flash mode `qio` and `qout`.

Jason2866, 2022-08-14:
Tasmota USB serial code changes starting [here](https://github.com/arendst/Tasmota/blob/development/tasmota/tasmota.ino#L188)

TD-er, 2022-08-14:
Thanks, will look into these links.

Jason2866, 2022-08-14:
> Yes, sure. Download tool v. 3.9.2. As all before. Also, 16m1m was successfully flashed after it.

@TD-er  Since how the bootloader are built for core 2.0.4 idk if using old tools is still working. There are NO defaults anymore in magic bytes firmware header.

TD-er, 2022-08-14:
Oh and just curious, why don't you have the option to not append the SHA digest when generating the factory bin?

Jason2866, 2022-08-14:
esptool v4.1 does not patch the header if there is the SHA diggest. So the need to be stripped before to make it possible to do the needed header patching.
In IDF this is not needed since the bootloaders are built together with the firmware with the settings made for size, speed and mode.

TD-er, 2022-08-14:
Yep, I know why it is needed when patching the speed/mode flags.
But if it isn't possible to do so on "factory" images, then flashing using flash tools which may patch these on the fly will cause bootloops, right?

Jason2866, 2022-08-14:
See annswer above. The firmware.img does not have anymore the diggest. So header the needed patching is now possible with esptool v4.1 too.

sobolkz, 2022-08-14:
> What state has the checkbox to change the bin? (not sure about the exact text) Does it allow to change the bin during flash or not? What does the flash tool show about the detected module? (flash frequency, flash mode, etc.)

As I use natuve Win tool Esspessif Download tool and remember option like "Do mot change the bin" always checked. Also it doesn't matter which type of flashing mode was used - it oleays use propriate type DIO or QIO. 
Also, as I wrote before @tonhuisman's old test release 16M8M can be successfully used with same tool and same flashing options.

Jason2866, 2022-08-14:
Exactly old test release, stuff changed. Old may not working anymore!

Jason2866, 2022-08-14:
Even Platformio guys seems sometimes a bit confused about. Did a PR [https://github.com/platformio/platform-espressif32/pull/880](https://github.com/platformio/platform-espressif32/pull/880)

[11 more replies]

PR #4199 (2022-08-15): Fix boot flash mode for factory firmware
and set default to `dout` if none is set in boards manifest or in platformio.ini

Maybe fixes #4194
closed 2022-08-19 by sobolkz