letscontrolit/ESPEasy #4198: eventvalue parsing

chemmex, 2022-08-14:
I'm not sure it's because of #4146, but zero-valued eventvalues are not parsed anymore in the latest release.

Tested on ESP_Easy_mega_20220809_normal_ESP8266_4M1M build

The rule triggers log record upon arrival of **setctl** command

```
on setctl do
	logentry,"All: %eventvalue0%, #1: %eventvalue1%, #2: %eventvalue2% #3: %eventvalue3%, #4: %eventvalue4%"
endon
```

If I send to the command topic the following payload:
`event,setctl=1,2,3`

the log shows 
```
689929: EVENT: setctl=1,2,3
690049: ACT : logentry,'All: 1,2,3, #1: 1, #2: 2 #3: 3, #4: 0'
690051: All: 1,2,3, #1: 1, #2: 2 #3: 3, #4: 0
```
If I change the payload to 

`event,setctl=0,2,3`

logentry is not triggered:

```
724681: EVENT: setctl=0,2,3
```

TD-er, 2022-08-14:
OK, this may indeed be related to this strip leading zeroes.
When parsing events, the first eventvalue must be a numerical and I think the stripping part may indeed mess things up here.

I was already a bit afraid this may become an issue and was tempted to add another checkbox to enable/disable this optimization.

ghtester, 2022-08-14:
Please don't add the checkbox, the optimization (or a value translation) works fine except this particular case, where the first zero looks to be eaten by wrong test condition. Hopefully this could be fixed without complicating the settings page.

PR #4204 (2022-08-16): [Rules] Fix matching events with first eventvalue being zero (#4198)
Fixes: #4198

@chemmex Can you test this PR to see if it is now fixed?
@ghtester: Can you test to see if this PR does not break #4146 ?
commit 69632bfb (2022-08-16): [Rules] Fix matching events with first eventvalue being zero (#4198)

Fixes: #4198
closed 2022-08-16 by TD-er