letscontrolit/ESPEasy #4201: missing file

IW2FAG-11, 2022-08-15:
I tried to compile mega-20220809  and I find the errors:
PluginStats.h: 10: 29: fatal error: CircularBuffer.h: No such file or directory # include <CircularBuffer.h>
And also I fixed an incorrect link in file .... src \ PluginStructs \ P002_data_struct.h from:
# include "src / Helpers / Hardware.h"
to:
# includes "../Helpers/Hardware.h".

I am missing the CircularBuffer.h file for a custom compilation esp8266_POWER_1M

TD-er, 2022-08-15:
Hmm PlatformIO has the library marked as a dependency.

Are you maybe compiling using Arduino IDE?

TD-er, 2022-08-15:
I added a PR which should fix the things you reported.

IW2FAG-11, 2022-08-15:
Yes dear, i'm also using Arduino ti male mine compilation

PR #4202 (2022-08-15): [Build] Fix include path + add CircularBuffer to libs folder (#4201)
Fixes: #4201
commit 04813c93 (2022-08-15): [Build] Fix include path + add CircularBuffer to libs folder (#4201)

Fixes: #4201
closed 2022-08-15 by TD-er