letscontrolit/ESPEasy #4203: Latest Release ESP_Easy_mega_20220809_normal_ESP8266_4M1M don't send HTTP in rules.

fly74, 2022-08-15:
Latest Release ESP_Easy_mega_20220809_normal_ESP8266_4M1M don't send HTTP in rules.

Log:

`321046 : Info   : EVENT: MS#State=0
321055 : Debug  : timeStringToSeconds: "0" --> invalid
321056 : Debug  : timeStringToSeconds: "1" --> invalid
321059 : Debug  : conditionMatch: "0=1" --> "0" = "1" --> false (0 = 1)
321060 : Debug  : conditionMatchExtended: false "0=1"
321060 : Debug  : Lev.1: [if false]=false
321064 : Debug  : Lev.1: [else]=true
321066 : Info   : ACT  : SendToHTTP,esphost2,80,control?cmd=event,poff
321068 : Debug  : Command: SendToHTTP
321069 : Debug  : SendToHTTP,esphost2,80,control?cmd=event,poff
321070 : Debug  : Par1: 846230272 Par2: 80 Par3: 1073710592 Par4: 1717989120 Par5: 0
321072 : Error  : Too many arguments: cmd=SendToHTTP Arg1=esphost2 Arg2=80 Arg3=control?cmd=event ExtraArg4=poff lineLength=44
321072 : Error  : Line: _SendToHTTP,esphost2,80,control?cmd=event,poff_
321073 : Error  : Command not executed! See: https://github.com/letscontrolit/ESPEasy/issues/2724
321075 : Debug  : SendToHTTP: Host: esphost2 port: 80 path: control?cmd=event,poff
321107 : Info   : HTTP : Start Digest Authorization for esphost2
321161 : Error  : HTTP : SendToHTTP esphost2 GET... failed HTTP code: 401
321167 : Info   : Command unknown: SendToHTTP,esphost2,80,control?cmd=event,poff
321170 : Info   : ACT  : SendToHTTP,esphost1,80,control?cmd=event,poff
321172 : Debug  : Command: SendToHTTP
321173 : Debug  : SendToHTTP,esphost1,80,control?cmd=event,poff
321174 : Debug  : Par1: 829453056 Par2: 80 Par3: 1073710592 Par4: 1717989120 Par5: 0
321176 : Error  : Too many arguments: cmd=SendToHTTP Arg1=esphost1 Arg2=80 Arg3=control?cmd=event ExtraArg4=poff lineLength=44
321177 : Error  : Line: _SendToHTTP,esphost1,80,control?cmd=event,poff_
321177 : Error  : Command not executed! See: https://github.com/letscontrolit/ESPEasy/issues/2724
321180 : Debug  : SendToHTTP: Host: esphost1 port: 80 path: control?cmd=event,poff
325408 : Error  : HTTP : SendToHTTP esphost1 GET... failed HTTP code: -1 connection failed
325412 : Info   : Command unknown: SendToHTTP,esphost1,80,control?cmd=event,poff
325417 : Debug  : EVENT: MS#State=0 Processing time:4371 milliSeconds
325425 : Info   : EVENT: http#esphost2=401
325428 : Debug  : EVENT: http#esphost2=401 Processing time:3 milliSeconds
325436 : Info   : EVENT: http#esphost1=-1`

Curl with url from a linux cli works.

tonhuisman, 2022-08-15:
> Error : Too many arguments: cmd=SendToHTTP

Is the `Tolerant last parameter` setting in Tools/Advanced enabled? It has to because you are not using quotes around the last argument for SendToHTTP command. But you can also wrap that in quotes, to solve this:
`SendToHTTP,esphost2,80,"control?cmd=event,poff"`

fly74, 2022-08-15:
looks different but don't work

24892 : Info   : EVENT: MS#State=1
24900 : Debug  : timeStringToSeconds: "1" --> invalid
24901 : Debug  : timeStringToSeconds: "1" --> invalid
24903 : Debug  : conditionMatch: "1=1" --> "1" = "1" --> true (1 = 1)
24904 : Debug  : conditionMatchExtended: true "1=1"
24904 : Debug  : Lev.1: [if true]=true
24907 : Info   : ACT  : SendToHTTP,esphost1,80,"control?cmd=event,pon"
24909 : Debug  : Command: SendToHTTP
24910 : Debug  : SendToHTTP,esphost1,80,"control?cmd=event,pon"
24911 : Debug  : Par1: 846230272 Par2: 80 Par3: 1073710848 Par4: 0 Par5: 0
24913 : Debug  : SendToHTTP: Host: esphost1 port: 80 path: control?cmd=event,pon
24945 : Info   : HTTP : Start Digest Authorization for esphost1
24991 : Error  : HTTP : SendToHTTP esphost1 GET... failed HTTP code: 401
24997 : Info   : ACT  : SendToHTTP,esphost2,80,"control?cmd=event,pon"
25001 : Debug  : Command: SendToHTTP
25002 : Debug  : SendToHTTP,esphost2,80,"control?cmd=event,pon"
25003 : Debug  : Par1: 829453056 Par2: 80 Par3: 1073712384 Par4: 0 Par5: 0
25006 : Debug  : SendToHTTP: Host: esphost2 port: 80 path: control?cmd=event,pon
25049 : Info   : HTTP : Start Digest Authorization for esphost2
25101 : Error  : HTTP : SendToHTTP esphost2 GET... failed HTTP code: 401
25107 : Debug  : Lev.1: [else]=false
25114 : Debug  : EVENT: MS#State=1 Processing time:221 milliSeconds
25117 : Info   : EVENT: http#esphost1=401
25119 : Debug  : EVENT: http#esphost1=401 Processing time:2 milliSeconds
25124 : Info   : EVENT: http#esphost2=401
25126 : Debug  : EVENT: http#esphost2=401 Processing time:2 milliSeconds
27286 : Info   : SW  : GPIO=14 State=0 Output value=0

fly74, 2022-08-15:
but curl http://esphost1/control?cmd=event,pon works

fly74, 2022-08-15:
Downgrade to ESP_Easy_mega_20220616_normal_ESP8266_4M1M works again.

TD-er, 2022-08-15:
This is related to this issue: https://github.com/letscontrolit/ESPEasy/issues/4198
The 1st parameter is a `0` which is now not parsed correctly.

TD-er, 2022-08-23:
Ah I think I found the issue here.
I first tested the event command via the tools page command line field.
This generated a command for me to test and it uses the `tools?cmd=...` url, but that should not really matter.
It also replaced the command by `%2C` but adding quotes like what has been suggested also does not make it work.

Not working:
```
sendtohttp,192.168.10.114,80,tools?cmd=event%2Cpoff
```

Working:
```
sendtohttp,192.168.10.114,80,/tools?cmd=event%2Cpoff
sendtohttp,192.168.10.114,80,"/tools?cmd=event,poff"
sendtohttp,192.168.10.114,80,"/control?cmd=event,poff"
```

So as you can see, I need to prefix a / in the URI part, which I accidentally removed.
I will add the / again, but at least you can already try if it is working by changing your commands.

fly74, 2022-08-23:
I think the 401 can not be a reason of the removed / because it means unauthorized.

TD-er, 2022-08-23:
> I think the 401 can not be a reason of the removed / because it means unauthorized.

That's for sure (found another issue)
Can you try the latest GH Actions test build from [here](https://github.com/letscontrolit/ESPEasy/actions/runs/2913169845) ?

fly74, 2022-08-23:
ESP_Easy_mega_20220823_normal_ESP8266_4M1M tested positive with no slash and no quotes in rule. :)

TD-er, 2022-08-23:
> ESP_Easy_mega_20220823_normal_ESP8266_4M1M tested positive with no slash and no quotes in rule. :)

Aiiii!
Thanks for reporting, I will put the build in 7 days isolation and order some tests. ;)

PR #4205 (2022-08-25): [SendToHttp] Do not set HTTP authentication when no user/pass set
A number of setups may report HTP error 401 since the 20220809 build.
This fix should not add an auth header if no user/pass is set.

- [x] Fix wait for acknowledgement flag / timeout (#4208)
- [x] Fix HTTP 401 error code where old builds worked fine (#4203)

Fixes: #4203
Fixes: #4208
closed 2022-08-25 by TD-er