letscontrolit/ESPEasy #4207: Mesh with the ESP8266

BaGRoS, 2022-08-18:
Is it possible to build a mesh with the ESP8266 so that one of the devices can access the internet and the rest can transmit data as well?

Many thanks,
BaGRoS

tonhuisman, 2022-08-18:
That is being worked on, based on ESP-NOW, called ESPEasy-now in our context: #3301

BaGRoS, 2022-08-18:
So it's already working? Will it work in the future?
I am only interested in sending MQTT.

TD-er, 2022-08-18:
> So it's already working? Will it work in the future? I am only interested in sending MQTT.

It is not yet merged due to compatibility issues with ESP8266 and ESP32 and also it now takes too much time to "discover" other nodes in the mesh. Also there is still a bug in the code which may cause messages to hop among nodes without reaching the broker.
But the whole idea is that MQTT messages will be forwarded to a "gateway node" which is a node which has a connection to the MQTT broker.

BaGRoS, 2022-08-18:
Super information and at the same time a great pity that it doesn't work. It would have helped a lot of people.