letscontrolit/ESPEasy #4208: Generic and Advanced HTTP controllers timeout despite Ignore Acknowledgement option is set

ghtester, 2022-08-18:
It looks the Generic HTTP & Generic HTTP Advanced controllers on recent ESP firmware builds needs a response from the HTTP host which are connecting to despite the **Check Reply:** option set to **Ignore Acknowledgement**

Tested with the **Port Listener** application running on a MS Windows powered target host. This app is just a listener so it accepts the connection only and does not respond anything to client.

Despite the connection is made successfully , ESP node is waiting for response and reports connection fail after Timeout expiration.

I believe it's a bug as I wouldn't expect this error when the **Ignore Acknowledgement** option is set and when the connection to host is successful. AFAIK in past this worked without error. So from my perspective the response should be checked only when the **Check Acknowledgement** option is active in the **Check Reply:**  configuration listbox. 
When the **Ignore Acknowledgement** option is active, the Timeout error should occur only when the target host is inaccessible / can't be connected to at the configured port number.

The ESP node log:
5503126: HTTP : C008 192.168.44.34 GET... failed HTTP code: -11 read Timeout
85503127: HTTP : C008 connection failed (521/0)
85503129: EVENT: http#192.168.44.34=-11
85503531: HTTP : C008 192.168.44.34 GET... failed HTTP code: -11 read Timeout
85503532: HTTP : C008 connection failed (522/0)
85503536: EVENT: http#192.168.44.34=-11
85503813: HTTP : C008 192.168.44.34 GET... failed HTTP code: -11 read Timeout
85503814: HTTP : C008 connection failed (523/0)
85503817: EVENT: http#192.168.44.34=-11

The target accepted the connection repeatedly (as it is just listener, without any reply to initiator):
Client connected
GET /demo.php?name=ESP_04&task=Voltage&valuename=Analog&value=39 HTTP/1.1
Host: 192.168.44.34
User-Agent: ESP Easy/20116/Aug 16 2022 14:06:46
Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0
Authorization: Basic Og==
Connection: keep-alive
Content-Length: 0

Client disconnected
Client connected
GET /demo.php?name=ESP_04&task=Voltage&valuename=Analog&value=39 HTTP/1.1
Host: 192.168.44.34
User-Agent: ESP Easy/20116/Aug 16 2022 14:06:46
Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0
Authorization: Basic Og==
Connection: keep-alive
Content-Length: 0

Client disconnected
...
...

ghtester, 2022-08-21:
Tried to recompile with PR 4205, so far the Timeout issue still persists.

TD-er, 2022-08-23:
I'm not 100% this timeout is a bug.
Before it was also having a timeout, but less noticable.
And if it didn't timeout, then it was probably taking longer than the timeout to make the initial connection.
There were several problems with the old HTTP client code:
- Lots of versions of the same code, not only code duplication, but also different behavior depending on what function you used to make a HTTP call.
- On ESP32 the timeout appeared to be set in seconds instead of msec. Thus "200" was effectively 200 seconds. Since the timeout is defined as an int, it may still be 1 sec even though it is set to something less than 1000 msec.
- ignoring the server reply may lead to undefined behavior on the server side as it may lead to stale connections which eventually timeout.

In a pending PR (#4205) I now made some changes to fix other HTTP call issues too.
However, there will still be some waiting on a server reply, since I am now using Arduino HTTP client code which cannot skip waiting for some reply.
What I did change today is that with the ignore acknowledgement option set, it will not try to use a digest auth retry if an initial Basic auth attempt led to a HTTP 401 error. (unauthorized)

ghtester, 2022-08-23:
> * ignoring the server reply may lead to undefined behavior on the server side as it may lead to stale connections which eventually timeout.

I agree but my (maybe a bit wrong) understanding of the **Ignore Acknowledgement** option was that no reply from the HTTP host is required as soon as the connection is successful.
I know this option should not be widely used with ordinary HTTP hosts but it's useful at least for testing and the Timeout messages in this case are annoying. So thanks a lot for your work on it.

TD-er, 2022-08-23:
> I agree but my (maybe a bit wrong) understanding of the **Ignore Acknowledgement** option was that no reply from the HTTP host is required as soon as the connection is successful.

It was more the other way around...
Initially the code was such that it just made some connection (without proper timeout settings applied, due to a bug in how timeout of the client is implemented, sometimes resulting in a crash of the ESP) and then sent the HTTP request and closed the client object.

Since this caused issues on some servers, I added this option.
The option was to wait for some reply of the host.
Now that I'm using the Arduino code, I cannot quickly ignore the reply and close the connection before receiving a reply.

Problem is that the WiFiClient object not always respects the set timeout, which in some stages of making the connection could lead to watchdog reboots.
Hopefully those will now occur less, if any.

ghtester, 2022-08-24:
OK, thanks for the details. Unfortunately now it seems to be a bit worse because the results are more random. See the pairs of log from ESP and from listener below.
The controller is Generic HTTP with the following queue settings:
[image]

------------------------------------------------
ESP log:

825713: WD   : Uptime 14 ConnectFailures 83 FreeMem 14920 WiFiStatus WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init
833157: EVENT: Clock#Time=Wed,03:49
854274: ADC  : Analog value: 42 = 42
854303: EVENT: Voltage#Analog=42
854692: HTTP : C008 192.168.44.34 GET... failed HTTP code: -11 read Timeout
854693: HTTP  : C008 connection failed (84/0)
854697: EVENT: http#192.168.44.34=-11
855074: HTTP : C008 192.168.44.34 GET... failed HTTP code: -11 read Timeout
855075: HTTP  : C008 connection failed (85/0)
855078: EVENT: http#192.168.44.34=-11
855417: HTTP : C008 192.168.44.34 GET... failed HTTP code: -11 read Timeout
855418: HTTP  : C008 connection failed (86/0)
855422: EVENT: http#192.168.44.34=-11
855919: HTTP : C008 192.168.44.34 GET... failed HTTP code: -11 read Timeout
855920: HTTP  : C008 connection failed (87/0)
855958: EVENT: http#192.168.44.34=-11
855965: WD   : Uptime 14 ConnectFailures 87 FreeMem 14712 WiFiStatus WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init
856352: HTTP : C008 192.168.44.34 GET... failed HTTP code: -11 read Timeout
856353: HTTP  : C008 connection failed (88/0)
856357: EVENT: http#192.168.44.34=-11
856728: HTTP : C008 192.168.44.34 GET... failed HTTP code: -11 read Timeout
856729: HTTP  : C008 connection failed (89/0)
856733: EVENT: http#192.168.44.34=-11

Listener log:

Client connected
  GET /demo.php?name=ESP_04&task=Voltage&valuename=Analog&value=42 HTTP/1.1
  Host: 192.168.44.34
  User-Agent: ESP Easy/20205/Aug 23 2022 22:42:56
  Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0
  Connection: close
  Content-Length: 0
  
Client disconnected
Client connected
  GET /demo.php?name=ESP_04&task=Voltage&valuename=Analog&value=42 HTTP/1.1
  Host: 192.168.44.34
  User-Agent: ESP Easy/20205/Aug 23 2022 22:42:56
  Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0
  Connection: close
  Content-Length: 0
  
Client disconnected
Client connected
Client connected
Client disconnected
  GET /demo.php?name=ESP_04&task=Voltage&valuename=Analog&value=42 HTTP/1.1
  Host: 192.168.44.34
  User-Agent: ESP Easy/20205/Aug 23 2022 22:42:56
  Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0
  Connection: close
  Content-Length: 0
  
Client disconnected
Client connected
  GET /demo.php?name=ESP_04&task=Voltage&valuename=Analog&value=42 HTTP/1.1
  Host: 192.168.44.34
  User-Agent: ESP Easy/20205/Aug 23 2022 22:42:56
  Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0
  Connection: close
  Content-Length: 0
  
[... 673 lines cut ...]
  Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0
  Connection: close
  Content-Length: 0
  
Client disconnected
Client connected
  GET /demo.php?name=ESP_04&task=Voltage&valuename=Analog&value=42 HTTP/1.1
  Host: 192.168.44.34
  User-Agent: ESP Easy/20205/Aug 23 2022 22:42:56
  Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0
  Connection: close
  Content-Length: 0
  
Client disconnected
Client connected
Client connected
Client disconnected
  GET /demo.php?name=ESP_04&task=Voltage&valuename=Analog&value=42 HTTP/1.1
  Host: 192.168.44.34
  User-Agent: ESP Easy/20205/Aug 23 2022 22:42:56
  Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0
  Connection: close
  Content-Length: 0
  
Client disconnected

ghtester, 2022-08-24:
BTW. I thought and mentioned earlier that the Listener does not respond anything but just tested with telnet and it sends 'Hello !' to the client as soon as the connection is made.

TD-er, 2022-08-24:
OK, so essentially you don't want to see such errors when "ignore" is set?
The errors like `-1` are probably good as they can't even make a connection, but timeout errors should not be shown when it is known that the host will not ack anyway, right?

TD-er, 2022-08-24:
If the "Ignore Acknowledgement" is set, I now turn `HTTPC_ERROR_READ_TIMEOUT` into error code 200.
This way the controller will not retry to send it as it is considered to be sent successfully.

TD-er, 2022-08-24:
For the controllers, I made the timeout dynamic (unless it is set to ignore ack).
The timeout will now be 3x the last actual duration, with the set timeout as minimum.
If it is set to ignore the server ack. this will obviously not apply.

Still thinking whether I should make this dynamic behavior optional.

ghtester, 2022-08-25:
> OK, so essentially you don't want to see such errors when "ignore" is set? The errors like `-1` are probably good as they can't even make a connection, but timeout errors should not be shown when it is known that the host will not ack anyway, right?

Yes and yes. :-)
Thank you for the code update, I'll give it a try.

ghtester, 2022-08-25:
Quickly tested and now everything looks to be working fine and as expected with both Check and Ignore options selected. Thanks again for your great work! :-)

TD-er, 2022-08-25:
Thanks for testing, now I can merge it :)

PR #4205 (2022-08-25): [SendToHttp] Do not set HTTP authentication when no user/pass set
A number of setups may report HTP error 401 since the 20220809 build.
This fix should not add an auth header if no user/pass is set.

- [x] Fix wait for acknowledgement flag / timeout (#4208)
- [x] Fix HTTP 401 error code where old builds worked fine (#4203)

Fixes: #4203
Fixes: #4208
closed 2022-08-25 by TD-er