letscontrolit/ESPEasy #4212: Can't compile the fresh custom Vagrant build

ghtester, 2022-08-21:
Just tried to make a fresh custom build using Vagrant environment, unfortunately it ends with the following error:  

....
....
default: Compiling .pio/build/custom_IR_ESP8266_4M1M/FrameworkArduino/Tone.cpp.o
    default: Compiling .pio/build/custom_IR_ESP8266_4M1M/FrameworkArduino/TypeConversion.cpp.o
    default: Compiling .pio/build/custom_IR_ESP8266_4M1M/FrameworkArduino/Updater.cpp.o
    default: Compiling .pio/build/custom_IR_ESP8266_4M1M/FrameworkArduino/WMath.cpp.o
    default: /home/vagrant/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/WMath.cpp: In function 'unsigned int makeWord(unsigned char, unsigned char)':
    default: /home/vagrant/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/WMath.cpp:84:55: error: new declaration 'unsigned int makeWord(unsigned char, unsigned char)'
    default:  unsigned int makeWord(unsigned char h, unsigned char l) {
    default:                                                        ^
    default: In file included from include/../src/Custom.h:7:0,
    default:                  from include/ESPEasy_config.h:94,
    default:                  from <command-line>:0:
    default: /home/vagrant/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Arduino.h:259:10: error: ambiguates old declaration 'uint16_t makeWord(byte, byte)'
    default:  uint16_t makeWord(byte h, byte l);
    default:           ^
    default: *** [.pio/build/custom_IR_ESP8266_4M1M/FrameworkArduino/WMath.cpp.o] Error 1
    default: ========================= [FAILED] Took 652.57 seconds =========================
    default: Environment             Status    Duration
    default: ----------------------  --------  ------------
    default: custom_IR_ESP8266_4M1M  FAILED    00:10:52.570
    default: ==================== 1 failed, 0 succeeded in 00:10:52.570 ====================
    default: Error: Option '-e' requires an argument.
    default: Error: Option '-e' requires an argument.
    default: Error: Option '-e' requires an argument.
    default: Error: Option '-e' requires an argument.
    default: Error: Option '-e' requires an argument.
    default: Created /tmp/tmp.me02qYkyTl
    default: Created /tmp/tmp.me02qYkyTl/ESPEasy_collect_dist
    default: ### Creating zip archives
    default:
    default: zip error: Nothing to do! (/home/vagrant/GitHub/letscontrolit/ESPEasy/ESPEasy_ELF_files_20220821_vagrant.zip)
    default: ### Created ESPEasy_ELF_files_20220821_vagrant.zip
    default: cp:
    default: cannot stat 'build_output/bin/*'
    default: : No such file or directory
    default: ### Created ESPEasy_ESP82xx_20220821_vagrant.zip
    default: cp:
    default: cannot stat 'build_output/bin/*ESP32*'
    default: : No such file or directory

TD-er, 2022-08-21:
Can you show me the Custom.h file (stripped from passwords etc.)

ghtester, 2022-08-21:
Hopefully I did not forget to remove all secrets... :-)

#ifndef ESPEASY_CUSTOM_H
#define ESPEASY_CUSTOM_H

// #define LIMIT_BUILD_SIZE
// #define DCORE_POST_3_0_0

#include <Arduino.h>

/*
    To modify the stock configuration without changing the EspEasy.ino file :
    1) rename this file to "Custom.h" (It is ignored by Git)
    2) define your own settings below
    3) define USE_CUSTOM_H as a build flags. ie : export PLATFORMIO_BUILD_FLAGS="'-DUSE_CUSTOM_H'"
 */

/*
 #######################################################################################################
   Your Own Default Settings
 #######################################################################################################
    You can basically ovveride ALL macro defined in ESPEasy.ino.
    Don't forget to first #undef each existing #define that you add below.
    But since this Custom.h is included before other defines are made, you don't have to undef a lot of defines.
    Here are some examples:
 */

// --- Feature Flagging ---------------------------------------------------------

#define FEATURE_ESPEASY_P2P       1     // (1/0) enables the ESP Easy P2P protocol
#define FEATURE_ARDUINO_OTA       1     //enables the Arduino OTA capabilities
// #define FEATURE_SD                1     // Enable SD card support
// #define FEATURE_DOWNLOAD          1     // Enable downloading a file from an url

#ifdef BUILD_GIT
# undef BUILD_GIT
#endif // ifdef BUILD_GIT

#define BUILD_GIT           "Build: "  __DATE__ " "  __TIME__

#define DEFAULT_NAME        "ESP_"                                 // Enter your device friendly name
#define UNIT                0                                        // Unit Number
#define DEFAULT_DELAY       60                                       // Sleep Delay in seconds

// --- Wifi AP Mode (when your Wifi Network is not reachable) ----------------------------------------
#define DEFAULT_AP_IP       192, 168, 4, 1                           // Enter IP address (comma separated) for AP (config) mode
#define DEFAULT_AP_SUBNET   255, 255, 255, 0                         // Enter IP address (comma separated) for AP (config) mode
#define DEFAULT_AP_KEY      "passwordkey"                                  // Enter network WPA key for AP (config) mode

// --- Wifi Client Mode -----------------------------------------------------------------------------
#define DEFAULT_SSID                         "defaultssid"            // Enter your network SSID
#define DEFAULT_KEY                          "defaultpasswordkey" // Enter your network WPA key
#define DEFAULT_SSID2                        ""                      // Enter your fallback network SSID
#define DEFAULT_KEY2                         ""                      // Enter your fallback network WPA key
#define DEFAULT_WIFI_INCLUDE_HIDDEN_SSID     true                   // Allow to connect to hidden SSID APs
#define DEFAULT_USE_STATIC_IP                false                   // (true|false) enabled or disabled static IP
#define DEFAULT_IP                           "192.168.5.50"          // Enter your IP address
#define DEFAULT_DNS                          "192.168.5.1"           // Enter your DNS
#define DEFAULT_GW                           "192.168.5.1"           // Enter your Gateway
#define DEFAULT_SUBNET                       "255.255.255.0"         // Enter your Subnet
#define DEFAULT_IPRANGE_LOW                  "0.0.0.0"               // Allowed IP range to access webserver
#define DEFAULT_IPRANGE_HIGH                 "255.255.255.255"       // Allowed IP range to access webserver
#define DEFAULT_IP_BLOCK_LEVEL               1                       // 0: ALL_ALLOWED  1: LOCAL_SUBNET_ALLOWED  2:
// ONLY_IP_RANGE_ALLOWED
#define DEFAULT_ADMIN_USERNAME               "adminusername"
#define DEFAULT_ADMIN_PASS                   "adminpasswordkey"

#define DEFAULT_WIFI_CONNECTION_TIMEOUT      10000 // minimum timeout in ms for WiFi to be connected.
#define DEFAULT_WIFI_FORCE_BG_MODE           false // when set, only allow to connect in 802.11B or G mode (not N)
#define DEFAULT_WIFI_RESTART_WIFI_CONN_LOST  false // Perform wifi off and on when connection was lost.
[... 407 lines cut ...]
// #define USES_C009   // FHEM HTTP
// #define USES_C010   // Generic UDP
 #define USES_C011   // Generic HTTP Advanced
// #define USES_C012   // Blynk HTTP
 #define USES_C013   // ESPEasy P2P network
// #define USES_C014   // homie 3 & 4dev MQTT
// #define USES_C015   // Blynk
// #define USES_C016   // Cache controller
// #define USES_C017   // Zabbix
// #define USES_C018   // TTN/RN2483

/*
 #######################################################################################################
 ###########     Notifiers
 #######################################################################################################
 */

 #define USES_N001   // Email
 #define USES_N002   // Buzzer

//#define LIMIT_BUILD_SIZE

#define FEATURE_RTTTL 1  // to support RTTTL command in Custom build !!!

#endif // ESPEASY_CUSTOM_H

ghtester, 2022-08-21:
Hopefully I did not forget to remove all secrets... :-)
[CustomH.txt](https://github.com/letscontrolit/ESPEasy/files/9388973/CustomH.txt)

TD-er, 2022-08-21:
Apparently `makeWord` is defined in Arduino.h as well as in WMapth.cpp.
At least for ESP8266.

This is for sure a bug in the Arduino code. but I will remove calls to this in the ESPEasy code.

TD-er, 2022-08-21:
Ah, jumped to the wrong conclusions...
You have `#include <Arduino.h>` in your Custom.h file.
It should not be there.

TD-er, 2022-08-21:
I also removed the include from the `Custom_sample.h` file.
See: https://github.com/letscontrolit/ESPEasy/pull/4214

ghtester, 2022-08-21:
Thanks a lot for your quick help, I'll update my Custom.h  as adviced and as soon as the PR 4214 is merged, I'll try to compile again.

PR #4214 (2022-08-21): [Build] Fix warning when P009 or P019 is not included in build
Fixes: #4212
closed 2022-08-21 by TD-er