letscontrolit/ESPEasy #4216: [ArduinoIDE] ESPEasy_config.h outside src directory breaks build

clumsy-stefan, 2022-08-22:
Having `ESPEasy_config.h` outside the `src` directory breaks the ArduinoIDE build which uses `ESPEasy/src` as root build dir.

I suggest to keep all compile relevant files (all source and include files) within the `ESPEasy/src` directory to clearly differentiate between needed files for compiling and surrounding utilities for scripting/building, etc.

As always IMHO...

TD-er, 2022-08-22:
Can you test the PR #4217 ?

clumsy-stefan, 2022-08-22:
Sure I'll try. But it seems there are a number of other changes/issues which makes it not compile anymore...

I try to give feedback this week (before I'm gone with limited access for 3mths...)

TD-er, 2022-08-22:
I can also merge it, since it should not break "PlatformIO builds" and I don't want to complicate other PRs.

clumsy-stefan, 2022-08-22:
I think that would also make it easier for me to do a PR with the other changes and not mix it with this PR..

TD-er, 2022-08-22:
OK, will merge it. (that will close this issue, but I'm sure you will let me know if it is still a problem ;) )

commit f7d8142a (2022-08-22): [Build] Move include/ESPEasy_config.h to src/include (#4216)

Fixes: #4216