letscontrolit/ESPEasy #4220: Hide sensitive/sucurity related info from unprotected pages

tonhuisman, 2022-08-23:
Currently, general info pages like `/json` are available without providing the admin password, when that is set. (from [this discussion](https://github.com/letscontrolit/ESPEasy/pull/4187#issuecomment-1224790736))

It can be hidden based on a setting in Tools/Advanced, or when the Admin password is set (less settings is better).

We should hide these fields:
- All `SSID` and `BSSID` fields
- All `MAC` fields
- ...

(suggestions welcome)

fly74, 2022-08-24:
The good combination is, when a password is set, then no sensitive in json.

TD-er, 2022-08-25:
> The default should be hiding and an override option with warning text to disable.

Don't get me wrong, we do agree on this.

However, only using a MAC filter is a bit like security-through-obscurity.
Better to have these units on a separate VLAN in your network.
Granted that knowing the MAC, you can still try to impose it as an attacker to claim an IP from the DHCP server.
But having access to a host in your network which can change its MAC is already past a point of no return regarding hacker access.
There are lots of other ways to get to know the MAC address of any host in your network, by simply doing an ARP request. Or performing a ping and looking at the ARP table of the computer. This doesn't need administrator rights on Windows or Linux as far as I know.

fly74, 2022-08-25:
It was only a weak example.

TD-er, 2023-10-16:
Well I now had to break into my own units as there is a bug in ESP-IDF 5.1 I'm working on where the remote IP is not detected reliable.

And I now see where the security holes are as I was eventually able to access the tools page via some text-based browser trickery on Linux and uploading a new firmware file while I was being blocked according to the access control code.

So there is for sure room for improvement here.

But it was a nice exercise to try and break in your own code :)  (took me way longer than expected)