letscontrolit/ESPEasy #4222: MQTT Controller in custom Build doesn´t work

BangerTech, 2022-08-25:
Hey guys,

i´´ve compiled a custom version of espeasy with PlatformIO. 
Everything seems to work so far. The only Problem i have right now is the MQTT Controller. 
I´ve used the Custom.h method of defining the Plugins and everything. Under the Controllers section i #define the HomeAssistant (openHAB) MQTT. But now if i go to Controllers in the WebUI and click on Edit i can select the right MQTT Controller.
Now when i type a Controller IP and Port 1883 click on enable and submit it doesn´t save my settings. 
Basically nothing i´ve typed in in the Controller Settings is saved. 
What could this be? Did i miss something in the Custom.h? Sorry i´m a NOOB when it comes to combiling ESPEASY. 
Help would be appreciated. 

Thanks

TD-er, 2022-08-25:
Which core version did you use?
In other words, which "custom" PlatformIO env did you use? 
It is probably shown in the filename and version string at the bottom right corner.

BangerTech, 2022-08-25:
so i´ve used custom_ESP32_4M316k if that helps. i don´t know where to finde the core version, basically i´ve just renamed the custom file to Custom.h and defined the plugins and the mqtt controller
the only problem i get from vscode is: c_cpp_properties.json ESPEasy-mega/include wasn´t found. 
but after flashing the esp everthing seems to work besides the controller

BangerTech, 2022-08-25:
i´ve just testet it again and it seem like espeasy doesn´t save anything so for example if i try to enable a new device and type a name in enable it and click on submit it says "Warning: Task Device Name is empty. It is adviced to give tasks an unique name" that´s weird. 
anybody experienced that before?

TD-er, 2022-08-25:
What browser do you use?
If you power cycle the unit, are the WiFi credentials still present?

tonhuisman, 2022-08-25:
Are you sure you need a Custom build? Can you first try with the latest 'official' [Release](https://github.com/letscontrolit/ESPEasy/releases) (20220809) to see if that works as expected?

TD-er, 2022-08-25:
Or use one of the latest [GitHub Actions builds](https://github.com/letscontrolit/ESPEasy/actions/runs/2925712474)

BangerTech, 2022-08-25:
@TD-er i use chrom and no it´s the same with WiFi credentials, they´re lost after a power cycle.
@tonhuisman yeah i kinda need a custom firmware because i need 2 plugins from the normal build, 1 from collectionA and one from the display build. So i thought it would be easier to build my own as i do with tasmota. 
i got these 3 builds running on 3 different esp´s now ;-)

TD-er, 2022-08-25:
OK, understand why you may need a custom build.
But please try the "official" build and/or the "actions" build I linked first.
Right now we need to see if the flash is being addressed in the right way.

If trying the Actions build, please try to do an OTA update first and if that's not fixing it, try to flash a "factory" bin file to the ESP using serial.

The reason for performing these steps (first OTA and then serial) is that I've been working on some board definitions in the past week regarding how the flash is being addressed.
By performing an OTA update, you keep the bootloader and performing a flash via serial using the factory image, you update the bootloader.

Please use the Espressif download tool, with "DoNotChangeBin" option checked so it will not try to patch the bin file during flashing.

BangerTech, 2022-08-25:
@TD-er i´ve just flashed the esp32 with the normal build i compiled in vscode and its the same no WiFi is saved after powercycle and of course mqtt connector doesn´t save settings too. 
so it has to be something in vscode right?

TD-er, 2022-08-25:
Please check the pre-built binaries!
This way we can rule out whether it is a hardware issue, or a build environment issue.

tonhuisman, 2022-08-25:
We have [this page](https://espeasy.readthedocs.io/en/latest/Participate/PlatformIO.html) on how to correctly set up VSCode with PlatformIO, that includes how to develop for/with/on ESPEasy. That is also very applicable when building your own Custom builds.

BangerTech, 2022-08-25:
ok i think i found the issue i had somehow some wrong config in my vscode (maybe from another build) now it´s working like charm. 
sometimes you just need a rubber duck to overthink things ;-) 
Thanks guys!!!

closed 2022-08-25 by BangerTech

undee123, 2022-10-06:
> ok i think i found the issue i had somehow some wrong config in my vscode (maybe from another build) now it´s working like charm. sometimes you just need a rubber duck to overthink things ;-) Thanks guys!!!

Would be interesting which setting in VSCode has been fault as I have your same problem now and can't find out.

tonhuisman, 2022-10-06:
> as I have your same problem

Did you follow the guide I linked to, above?

> We have [this page](https://espeasy.readthedocs.io/en/latest/Participate/PlatformIO.html) on how to correctly set up VSCode with PlatformIO

undee123, 2022-10-06:
> Values set in `Custom.h` are applied at either a fresh install (new/wiped unit) of when running a Factory Reset on the unit, see: https://www.letscontrolit.com/forum/viewtopic.php?t=9283

Ok, so when I upload my build and afterwards do a (hardware) factory reset my settings should be set, right?