letscontrolit/ESPEasy #4227: P067 Only one channel operative at a time.

8lazo8, 2022-08-27:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### Summarize of the problem/feature request

I have a setup NodeMCU ESP8266 with two HX711 connected.
One HX711 has only channel A enabled and it working fine, channel B is off.
The other HX711 has a load cell on each channel. The problem is that dual channel mode is not working. If channel A is enabled, channel B displays 0. If I disable channel A then channelb B starts to work fine. When I enable channel A again channel B shows 0 again.

I am using official builds. The actual build I am using is ESP_Easy_mega_20220809_collection_A_ESP8266_4M1M.bin
I tried it with a clean install with the build ESP_Easy_mega_20220616_test_A_ESP8266_4M1M.bin with the same result.

I tried to interchange the loadcells, tried them one by one on every channel. Every each works fine until only one channel is enabled.

### Expected behavior
 To work in dual mode (interleaved mode)

### Actual behavior
Only one channel can be used at a time, when both are enabled ony A is producing data.

### Steps to reproduce
1. Add new device HX711
2. Configure both channels
3. Check data. Only channel A gives data, B is 0.
4. Turna channel A off.
5. Channel B starts to show data.

Hardware:
NodeMCU ESP8266
HX711 (green)

ESP Easy version: 

Build:⋄
20116  - Mega
System Libraries:⋄
ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄
mega-20220809_57b0c0c
Plugin Count:⋄
79 [Normal] [Collection] [Collection_A]
Build Time:⋄
Aug  9 2022 09:42:55
Binary Filename:⋄
ESP_Easy_mega_20220809_collection_A_ESP8266_4M1M
Build Platform:⋄
Linux-5.10.102.1-microsoft-standard-WSL2-x86_64-with-glibc2.29
Git HEAD:⋄
mega-20220809_57b0c0c

ESP Easy settings/screenshots: 

When both channels enabled:

[image]
[image]
[image]

When channel A off:

[image]
[image]
[image]

### Rules or log data

```

```

tonhuisman, 2022-12-27:
The problem here is that this plugin doesn't yet support multiple instances of the plugin being active at the same time, with the result that they are stepping on each others feet (data) 🤕.

I'll see if that can be fixed.

8lazo8, 2022-12-28:
> The problem here is that this plugin doesn't yet support multiple instances of the plugin being active at the same time, with the result that they are stepping on each others feet (data) 🤕.
> 
> I'll see if that can be fixed.

Not sure about 'only one plugin instance can be active', how do you mean that?
I am using now three devices from the same plugin type. I have to use three HX711 for three devices (only chanel one is enabled for each device) and in this way its working stable. The downside is wasting channels of HX711 and GPIOs from the MCU.
Does this mean that there are three instances from the same plugin parallel running?

The problem appears when trying to use both channels of one HX711.
By the way thank you for looking up this issue.

[image]

tonhuisman, 2022-12-28:
On further investigation, this plugin _does_ indeed support multiple instances. I have started refactoring it though, as it uses both global and static variables, that we want to avoid as much as possible, as that occupies memory all the time, even when the plugin is not used.
I have seen, while working on the code, that switching channels _might_ be somewhat sketchy, but I'll have to do real tests once the hardware I've ordered has arrived (few days).

tonhuisman, 2022-12-29:
@8lazo8 How is your Load-cell hardware connected to both channels of the HX711, as I've found a couple of 1, 2 and 4 cell configurations, but they all seem to be on the A channel, and not using the B channel. (NB: Hardware arrived today, so preparing to start testing 😉)

tonhuisman, 2022-12-30:
> @8lazo8 How is your Load-cell hardware connected to both channels of the HX711, 

I have connected 1 load-cell to channel A and another to channel B, and for me it is all working as intended, Also connected a second HX711 but as I have only 2 load-cells currently, it is only measuring some noise. I'll keep it running for some time to see if it is stable.

8lazo8, 2022-12-30:
> I have connected 1 load-cell to channel A and another to channel B, and for me it is all working as intended, Also connected a second HX711 but as I have only 2 load-cells currently, it is only [...]

I dont have the exact picture of my setup, but I can put in a picture of a similar one.

[image]

The first cell of this type was connected to channel A. A second one from the same type connected to channel B.

The second one's white wire went to B- pin, the green wire to B+ pin.

The red and black was connected in parallel with the first cell's wires.

I tried this with multiply HX711 modules with the same result.
The details are in the original post.
Data was only accurate and stable if only one channel was enabled at a time, doesnt matter which one. (if I recall correctly if any chanel was enabled was working fine individually, if both channels were enabled, only channel A was providing usuable data.  )
 I also replaced the cells on the inputs, it didnt change the end result.

tonhuisman, 2022-12-30:
I'll install an older build, that doesn't have my refactored code, to compare how that behaves 😊

8lazo8, 2022-12-30:
> I'll install an older build, that doesn't have my refactored code, to compare how that behaves 😊

Try the build I used.

tonhuisman, 2022-12-30:
@8lazo8 Once [this Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/3808773395) is completed you can download the `Binaries.zip` file and update your unit for testing. Please report back your findings.

8lazo8, 2022-12-30:
> @8lazo8 Once [this Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/3808773395) is completed you can download the `Binaries.zip` file and update your unit for testing. Please report back your findings.
> 
> NB: It's all forward and backward compatible, just improvements on the code.

Thank you, will do a test.

8lazo8, 2022-12-31:
I updated the binary on the device and it seems to work. :)

Unfortunatelly the update got rid of my tare data.

tonhuisman, 2023-01-01:
> I could not set the offset or reset the tare at all, the readings didnt change despite I set 0 offset or something other number.

There was a typo (copy/paste error) in my code that broke the Offset/Tare value for channel A. Fix is available after [this Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/3817073690) is completed.

8lazo8, 2023-01-01:
> > I could not set the offset or reset the tare at all, the readings didnt change despite I set 0 offset or something other number.
> 
> There was a typo (copy/paste error) in my code that broke _applying_ the Offset/Tare value for channel A. Fix is available after [this Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/3817073690) is completed.

This run seems has failed. Is the collection A binary usuable from it? (if its included in the binary.zip in this case)

tonhuisman, 2023-01-01:
Yep, most collection A builds have succeeded, AFAICS, so they are usable 👍

8lazo8, 2023-01-01:
> Yep, most collection A builds have succeeded, AFAICS, so they are usable 👍

Updated to this build and now every function I tried seems to work. Thank you and btw happy new year! :)

tonhuisman, 2023-01-01:
> and btw happy new year! :)

Thanks, happy new year to you too 😉

PR #4438 (2023-01-26): [P067] Refactoring and other improvements
Resolves #4227 

Features:
- [x] Refactoring to reduce the use of global and static variables
- [x] Fix possible issue with reading both A and B channels ~(_waiting for hardware to arrive_)~
- [x] Investigate GPIO initialization causing reading/is_ready faults (as reported in https://github.com/bogde/HX711/issues/222) -> (works fine with `INPUT` instead of `INPUT_PULLUP`)
- [x] Testing ([confirmed](https://github.com/letscontrolit/ESPEasy/issues/4227#issuecomment-1368505049))
- [ ] Documentation
closed 2023-01-26 by TD-er